Xyce/Xyce #2: Simbus backplane

umarcor, 2019-12-01:
After reading the following articles, I tried to search the codebase of the 'Simbus backplane' with no success. I'm actually interested on investigating whether it would allow mixed signal simulation with GHDL and Xyce. @tvrusso, can you provide any hint?

https://ieeexplore.ieee.org/document/1393986
https://ieeexplore.ieee.org/document/1612863

tvrusso, 2019-12-01:
I'm sorry to say that you're out of luck with SIMBUS.

The SIMBUS backplane was part of a proof-of-principle project with Clifton Labs dating back to around 2001-2004.  When that project ended, it appears that SIMBUS died with it.

It provided a mechanism to couple Xyce with the VHDL simulator Savant/TYVIS/Warped.  The interface in Xyce that was used to do this is no longer functional, as Xyce has changed significantly in the 15 years since that work concluded.  Further, even if you could find a copy of the old SIMBUS source code (which was never really published), it is likely that it wouldn't compile with today's compilers.  It was a cool idea at the time, but lack of funding and lack of a real development driver led it to whither on the vine.

A new Xyce mixed-signal interface is under development, but it is still embryonic and is not yet documented.  With it, one could couple Icarus Verilog to Xyce (once it is fully functional). But as yet, there is no fully functional mechanism for performing coupled, mixed-signal simulation.   You should contact the Xyce team via the google group, and the person who is working on that interface may be able to offer some pointers to you if you wanted to try it out.

tvrusso, 2019-12-01:
I misspoke.  The document describing the new mixed signal interface was in fact released, it just isn't on our documentation&tutorials web page.   You can find it at

https://www.osti.gov/biblio/1483152-application-note-mixed-signal-simulation-xyce

closed 2019-12-01 by tvrusso

umarcor, 2019-12-02:
Thanks a lot for the quick and insightful reply.

Although it is unfortunate that SIMBUS source code was never published, I believe that Xyce's mixed-signal interface is really interesting. Precisely, [GHDL](https://github.com/ghdl/ghdl) supports VPI (https://ghdl.readthedocs.io/en/latest/using/InvokingGHDL.html#vpi-build-commands) and a subset of VHPIDIRECT (https://ghdl.readthedocs.io/en/latest/using/Foreign.html). 

During the last months, I wrote a prototype to co-execute VHDL a Python web server and a JavaScript frontend: VUnit/vunit/pull/568. I am also using Python `ctypes` to interact with the API implemented in VHDL through VHPIPIDIRECT. This setup seems to be quite similar to the Python-based interface of Xyce described in the reference you provided. Unfortunately, I cannot spend any time actively developing the prototype, but I will read it carefully.

Moreover, I found that the same authors published https://www.osti.gov/biblio/1488489-digital-analog-cosimulation-using-cocotb-xyce a few months later, which already contains a section about GHDL and Xyce co-simulation with cocotb (which uses VPI). Thanks again!