Xyce/Xyce #4: Install difficulties

aolofsson, 2020-01-22:
I tried to follow the build procedure verbatim, but clearly I am missing something.
 
https://xyce.sandia.gov/documentation/BuildingGuide.html

See install file and config log attached. Has this been built on Ubuntu 18.04? Exact procedure below.

```bash
#!/bin/bash

###########################################################################
# References:
#
# https://xyce.sandia.gov/documentation/BuildingGuide.html
#
###########################################################################

#Dependancies
sudo apt-get install gfortran bison flex libfl-dev libfftw3-dev libsuitesparse-dev libsuitesparse-doc libblas-dev liblapack-dev libtool libopenmpi-dev openmpi-bin libtrilinos-teuchos-dev

###########################################################################
#Install Trilinos from source
wget https://github.com/trilinos/Trilinos/archive/trilinos-release-12-12-1.tar.gz 
tar zxvf trilinos-release-12-12-1.tar.gz
mkdir -p XyceLibs/Serial
SRCDIR=$PWD/Trilinos-trilinos-release-12-12-1
ARCHDIR=$PWD/XyceLibs/Serial
FLAGS="-O3 -fPIC"
cmake \
-G "Unix Makefiles" \
-DCMAKE_C_COMPILER=gcc \
-DCMAKE_CXX_COMPILER=g++ \
-DCMAKE_Fortran_COMPILER=gfortran \
-DCMAKE_CXX_FLAGS="$FLAGS" \
-DCMAKE_C_FLAGS="$FLAGS" \
-DCMAKE_Fortran_FLAGS="$FLAGS" \
-DCMAKE_INSTALL_PREFIX=$ARCHDIR \
-DCMAKE_MAKE_PROGRAM="make" \
-DTrilinos_ENABLE_NOX=ON \
-DNOX_ENABLE_LOCA=ON \
-DTrilinos_ENABLE_EpetraExt=ON \
-DEpetraExt_BUILD_BTF=ON \
-DEpetraExt_BUILD_EXPERIMENTAL=ON \
-DEpetraExt_BUILD_GRAPH_REORDERINGS=ON \
-DTrilinos_ENABLE_TrilinosCouplings=ON \
-DTrilinos_ENABLE_Ifpack=ON \
-DTrilinos_ENABLE_Isorropia=ON \
-DTrilinos_ENABLE_AztecOO=ON \
-DTrilinos_ENABLE_Belos=ON \
-DTrilinos_ENABLE_Teuchos=ON \
-DTeuchos_ENABLE_COMPLEX=ON \
-DTrilinos_ENABLE_Amesos=ON \
-DAmesos_ENABLE_KLU=ON \
-DTrilinos_ENABLE_Sacado=ON \
-DTrilinos_ENABLE_Kokkos=OFF \
-DTrilinos_ENABLE_ALL_OPTIONAL_PACKAGES=OFF \
-DTrilinos_ENABLE_CXX11=ON \
-DTPL_ENABLE_AMD=ON \
-DAMD_LIBRARY_DIRS="/usr/lib" \
-DTPL_AMD_INCLUDE_DIRS="/usr/include/suitesparse" \
-DTPL_ENABLE_BLAS=ON \
-DTPL_ENABLE_LAPACK=ON \
$SRCDIR

make

sudo make install

##############################################################################

#Clone Xyce
git clone git@github.com:Xyce/Xyce.git

#Build Xyce
cd Xyce
./bootstrap
cd ../
./configure CXXFLAGS="-O3 -std=c++11" ARCHDIR="/home/aolofsson/work/darpa/POSH/sandia-spice/Serial" CPPFLAGS="-I/usr/include/suitesparse"

```
 
[config.log](https://github.com/Xyce/Xyce/files/4098777/config.log)

tvrusso, 2020-01-22:
Please review the building guide's instructions about ARCHDIR.

You are installing your Trilinos libraries into $HOME/XyceLibs/Serial, but are giving a completely different ARCHDIR to Xyce's configure script.  They should be the same.

From the building guide:

> 
> We recommend installing Trilinos, built especially according to the directions in this guide, into a separate directory.  In the earlier section on building Trilinos, we instructed you to use a variable called "ARCHDIR" in your CMake invocation for Trilinos.  Xyce's configure script also has a variable called "ARCHDIR" that can be used not only to find these libraries and headers, but also some extra files that help determine interdependencies between Trilinos packages and link in all required Trilinos libraries.  Simply use the same install directory you gave to the Trilinos CMake script (ARCHDIR) as the value of Xyce's configure ARCHDIR:
> 
> /path/to/Xyce/configure ARCHDIR=$HOME/XyceLibs/Serial [other configure options]

Since you are providing inconsistent ARCHDIRs to the two build processes, Xyce's configure is being told to look for Trilinos in some other place than where you actually installed it.

Further, the building guide discourages you from attempting to use any version of Trilinos provided in your system's package management system, because no package managers ever build Trilinos with the same options that Xyce needs.  From the "Install prerequisite packages" section:

> Note that at this time no system provides packages for versions of Trilinos that are compatible with Xyce, so it will always need to be built from source. While some systems do provide packages for Trilinos in package management systems, these versions are either too old to be used or are not configured to have all the essential package options required by Xyce, and we do not recommend using them even if you happen to find one for your system.

  You have "libtrilinos-teuchos-dev" in your apt-get command.  You should remove this package, which may cause conflicts with the version you built from source .

tvrusso, 2020-01-22:
A few more points: you are installing Trilinos to $HOME/XyceLibs/Serial, but are using "sudo make install" to do so.  It is unnecessary to use sudo to install into a directory that you own, and doing so will simply make the files in your own home directory be owned by root instead of you.

tvrusso, 2020-01-22:
One last batch of comments.  It looks like you want Xyce itself to be installed in /home/aolofsson/work/darpa/POSH/sandia-spice/Serial, which you tried to do by specifying that directory as ARCHDIR.  That's probably because we used that variable name to set the "CMAKE_INSTALL_PREFIX"  in the instructions for building Trilinos.  We did it that way to try to emphasize that Trilinos was getting installed into ARCHDIR, and we are later telling Xyce to look for it thereby using a variable of the same name.

ARCHDIR is used to tell configure where to look for libraries specific to your architecture (not necessarily just trilinos --- you could install other system-specific things there if you had to build them from source for some reason).  

If you want configure to set you up so that "make install" will put Xyce somewhere other than its default /usr/local (which we definitely recommend you do), you use the "--prefix" option to configure.  From the building guide:

> 
> **Specifying alternate installation directory**
> 
> By default, configure will set up the Xyce build process so that "make install" installs Xyce components into subdirectories /usr/local: the executable will go in /usr/local/bin, associated libraries into /usr/local/lib, and header files into /usr/local/include.  This behavior can be changed by adding the "--prefix" option to configure:
> 
> path/to/Xyce/configure \
> --prefix=/opt/Xyce \
> ...
> 
> The example above will cause "make install" to install into /opt/Xyce/bin, /opt/Xyce/lib, and /opt/Xyce/include.
> 
> If you do not have root privileges on your system, this procedure can allow you to install Xyce into a directory that you own instead.  Otherwise, use "sudo" with "make install" to install Xyce into system directories.
> 
> **The Xyce Team STRONGLY recommends that you always specify a "--prefix" for Xyce builds, and not use the default "/usr/local."  Installing both serial and parallel versions of Xyce directly into /usr/local will cause the two versions to overwrite some components and leave one or the other unusable.**
> 

So what you really want in your configure line is:
./configure CXXFLAGS="-O3 -std=c++11" ARCHDIR=$HOME/XyceLibs/Serial --prefix="/home/aolofsson/work/darpa/POSH/sandia-spice/Serial" CPPFLAGS="-I/usr/include/suitesparse"

We also strongly discourage building the code inside its source directory, because that pollutes the directory with build artifacts, and can make it difficult to build multiple variants of Xyce (e.g. both serial and parallel builds) because droppings from a previous build might interfere with the next one.  Instead of running "./configure" inside the source tree you should create an empty build directory somewhere and build there.  This isolates the build artifacts from the source tree and keeps the source tree pristine.  Please do the following instead of "./configure":

```
cd Xyce
git clean -Xf     # this will get rid of any droppings that your previous configure attempt created
./bootstrap
cd ../
mkdir build_Xyce_Serial
cd build_Xyce_Serial
../Xyce/configure CXXFLAGS="-O3 -std=c++11" ARCHDIR="/home/aolofsson/work/darpa/POSH/sandia-spice/Serial" CPPFLAGS="-I/usr/include/suitesparse"
make
make install

```

Your source tree will remain clean and your build directory will contain only build artifacts.  Should you want to build Xyce for parallel runs as well, you would create a second build directory and issue your parallel configure line there.

This technique is also useful when the source tree resides on a remotely mounted filesystem shared by multiple computers of different architectures --- the same source tree can be used to build for all the different systems without each system stepping on the other.

The building guide's instructions all presuppose that you are following this guidance.  Per the building guide, again:

> **Build Directories**
> 
> It is important to note that our recommendations for building Trilinos and Xyce make use of "out-of-source" builds. This means that the executables and/or libraries are built in a directory completely separate from the directory containing the source code. This is accomplished by creating an empty directory anywhere in the file system of your computer (the "build directory"), and using appropriate arguments in "configure" scripts to tell the build system where to find the source code (the "source directory").
> 
> Doing builds in directories separate from the source code keeps the original source tree pristine, and allows you to build many different versions (specifically, both serial and parallel versions) without changing the source tree. It also permits sharing a single source tree over a networked filesystem, and building the code for multiple operating systems without copying the source to every machine. This process is described in detail in the generic "INSTALL" file included at the top level of the Xyce source tree; this file is part of the GNU Automake system, and is not specific to Xyce.
> 

This assumption of out-of-source builds is also evident in the Linux-specific example configure line we provide:

> **Linux**
> 
> If all packages have been installed as recommended, configuring on Linux is fairly simple. This is because the package manager installs all of its packages in /usr/lib and /usr/include, and these are searched by configure and the compilers with no extra options required.
> 
> The example configure invocation below should be appropriate for any distribution of Linux, though it has only been tested specifically on Red Hat, Ubuntu, and Linux Mint variants.  Minor modifications may be necessary on other distros.

```
/path/to/Xyce/configure \
/path/to/Xyce/configure \
CXXFLAGS="-O3 -std=c++11" \
ARCHDIR="$HOME/XyceLibs/Serial" \
CPPFLAGS="-I/usr/include/suitesparse"
```

aolofsson, 2020-01-22:
Yes, I succeeded in the end. My apologies for turning an RTFM problem on my part into an issue.  I do question why one would leave it up to stupid users (like me) to mess up the paths, if there is only one correct way of doing it. Attaching the script I used.
```bash

###########################################################################
# References:
#
# https://xyce.sandia.gov/documentation/BuildingGuide.html
# https://github.com/Xyce/Xyce
#
###########################################################################

###########################################################################
#Install Dependancies
###########################################################################

sudo apt-get install gfortran bison flex libfl-dev libtool
sudo apt-get install libfftw3-dev libsuitesparse-dev libblas-dev liblapack-dev
sudo apt-get install libopenmpi-dev openmpi-bin

###########################################################################
#Install Trilinos from source
###########################################################################

wget https://github.com/trilinos/Trilinos/archive/trilinos-release-12-12-1.tar.gz 
tar zxvf trilinos-release-12-12-1.tar.gz

SRCDIR=$PWD/Trilinos-trilinos-release-12-12-1
LIBDIR=/opt/xyce/xyce_lib
INSTALLDIR=/opt/xyce/xyce_serial
FLAGS="-O3 -fPIC"

cmake \
-G "Unix Makefiles" \
-DCMAKE_C_COMPILER=gcc \
-DCMAKE_CXX_COMPILER=g++ \
-DCMAKE_Fortran_COMPILER=gfortran \
-DCMAKE_CXX_FLAGS="$FLAGS" \
-DCMAKE_C_FLAGS="$FLAGS" \
-DCMAKE_Fortran_FLAGS="$FLAGS" \
-DCMAKE_INSTALL_PREFIX=$LIBDIR \
-DCMAKE_MAKE_PROGRAM="make" \
-DTrilinos_ENABLE_NOX=ON \
-DNOX_ENABLE_LOCA=ON \
-DTrilinos_ENABLE_EpetraExt=ON \
-DEpetraExt_BUILD_BTF=ON \
-DEpetraExt_BUILD_EXPERIMENTAL=ON \
-DEpetraExt_BUILD_GRAPH_REORDERINGS=ON \
-DTrilinos_ENABLE_TrilinosCouplings=ON \
-DTrilinos_ENABLE_Ifpack=ON \
-DTrilinos_ENABLE_Isorropia=ON \
-DTrilinos_ENABLE_AztecOO=ON \
-DTrilinos_ENABLE_Belos=ON \
-DTrilinos_ENABLE_Teuchos=ON \
-DTeuchos_ENABLE_COMPLEX=ON \
-DTrilinos_ENABLE_Amesos=ON \
-DAmesos_ENABLE_KLU=ON \
-DTrilinos_ENABLE_Sacado=ON \
-DTrilinos_ENABLE_Kokkos=OFF \
-DTrilinos_ENABLE_ALL_OPTIONAL_PACKAGES=OFF \
-DTrilinos_ENABLE_CXX11=ON \
-DTPL_ENABLE_AMD=ON \
-DAMD_LIBRARY_DIRS="/usr/lib" \
-DTPL_AMD_INCLUDE_DIRS="/usr/include/suitesparse" \
-DTPL_ENABLE_BLAS=ON \
-DTPL_ENABLE_LAPACK=ON \
$SRCDIR

make

sudo mkdir -p $LIBDIR
sudo make install

###########################################################################
#Install Xyce from Source
###########################################################################

#Clone Xyce
git clone git@github.com:Xyce/Xyce.git

#Build Xyce
cd Xyce
./bootstrap
./configure CXXFLAGS="-O3 -std=c++11" ARCHDIR=$LIBDIR --prefix=$INSTALLDIR CPPFLAGS="-I/usr/include/suitesparse"
make
sudo mkdir -p $INSTALLDIR
sudo make install

#Test installation
$INSTALLDIR/bin/Xyce
```

closed 2020-01-22 by aolofsson

tvrusso, 2020-01-23:
The reason is that there is NOT "one correct way to do it."

The configure script is extraordinarily flexible and allows for all sorts of ridiculous permutations of system configuration.  We provide guidance on a recommended way to do it that keeps everything consistent, and if that guidance is followed then things are easy.  If the user knows better and wants to do something else, the capability is there.  

A big problem here is that there are so many ways to compile Trilinos, and their default build is not suited to use with Xyce.  It is difficult for autoconf to pick up how Trilinos is compiled (although we try pretty hard to detect it and give a minimum of dials that have to be set just right).  But if one doesn't tell Xyce's configure where to look for Trilinos's libraries and headers, then it has no way of finding them.  Once it finds them, it does as best it can to detect what components of the Trilinos install are needed and whether they will work.