Xyce/Xyce #6: Segmentation fault on linux

jmgc, 2020-04-04:
I have a segmentation fault on linux. I am using Xyce-7.0. It happens in two machines, one running CentOS Linux release 7.7.1908, and the second Ubuntu 18.04.4 LTS, while things run smoothly on MacOS. I am using the following circuit:

```
.title HSOP77case

.param prb = 0
.param vin = 2.5
.param r_1 = 10000.0
.param r_2 = 10000.0
.param r_3 = 10000.0
.param r_4 = 10000.0
.param r_o = 200
.param c_2 = 2e-11
.param c_4 = 2e-11
.param v_1offset = 0
.param v_2offset = 0
.model 2n2222 NPN (IS=1e-14 VAF=100 BF=200 IKF=0.3 XTB=1.5 BR=3 CJC=8e-12 CJE=25e-12 TR=100e-9 TF=400e-12 ITF=1 VTF=2 XTF=3 RB=10 RC=.3 RE=.2)
.model yellow D (IS=93.1p RS=42m N=4.61 BV=2 IBV=10u CJO=2.97p VJ=.75 M=.333 TT=4.32u)

.subckt op77 1 2 99 50 39
.model dx D (IS=1e-15)
.model den D (IS=1e-12 RS=12.08k KF=1e-17 AF=1)
.model qx NPN (BF=417e6)
.model dy D (IS=1e-15 BV=50)
.model din D (IS=1e-12 RS=7.55e-6 KF=1.55e-15 AF=1)

r1 2 3 5e11
r2 1 3 5e11
r3 5 97 0.0606
r4 6 97 0.0606
cin 1 2 4e-12
c2 5 6 218.9e-9
i1 4 51 1
ios 1 2 0.15e-9
bos 9 10 v={10E-6 + V(30, 33)}
q1 5 2 7 qx
q2 6 9 8 qx
r5 7 4 0.009
r6 8 4 0.009
d1 2 1 dx
d2 1 2 dx
en 10 1 12 0 1
gn1 0 2 15 0 1
gn2 0 1 18 0 1
eref 98 0 33 0 1
eplus 97 0 99 0 1
eneg 51 0 50 0 1
dn1 11 12 den
dn2 12 13 den
vn1 11 0 dc 2
vn2 0 13 dc 2
dn3 14 15 din
dn4 15 16 din
vn3 14 0 dc 2
vn4 0 16 dc 2
dn5 17 18 din
dn6 18 19 din
vn5 17 0 dc 2
vn6 0 19 dc 2
r7 20 98 1
g1 98 20 5 6 59.91
d3 20 21 dx
d4 22 20 dx
b1 97 21 v={-2.4 + V(97, 33)}
b2 22 51 v={-2.4 + V(33, 51)}
r8 23 98 6.01e9
c3 23 98 500e-12
g2 98 23 20 33 33.3e-6
v1 97 24 1.3
v2 25 51 1.3
d5 23 24 dx
d6 25 23 dx
r9 26 27 1
c4 26 27 -39.75e-9
r10 27 98 1e-6
e3 26 98 23 33 1e6
r13 30 31 1
l2 31 98 7.96e-3
g4 98 30 3 33 1.0e-7
d7 30 97 dx
d8 51 30 dx
r14 32 98 1
c5 32 98 79.5e-9
g5 98 32 27 33 1
r15 33 97 1
r16 33 51 1
bsy 99 50 i={0.325E-3 + 0.0425E-3*V(99, 50)}
f1 34 0 v3 1
f2 0 34 v4 1
r17 34 99 400
r18 34 50 400
l3 34 39 2e-7
g6 37 50 32 34 2.5e-3
g7 38 50 34 32 2.5e-3
g8 34 99 99 32 2.5e-3
g9 50 34 32 50 2.5e-3
v3 35 34 6.8
v4 34 36 4.4
d9 32 35 dx
d10 36 32 dx
d11 99 37 dx
d12 99 38 dx
d13 50 37 dy
d14 50 38 dy
.ends op77

iinj 0 probe 0 ac {0.5*prb*(prb+1)}
vinj probe ninplp 0 ac {0.5*prb*(prb-1)}
vprobe probe noutlp 0
xu1 ninp ninn np15 nm15 ninplp op77
xu2 no nre np15 nm15 nrep op77
r1 ninnm 0 {r_1}
r2 ne ninnm {r_2}
c2 ne ninnm {c_2}
vu1offset ninn ninnm {v_1offset}
r3 nin ninp {r_3}
vu2offset nre nrep {v_2offset}
r4 nrep ninp {r_4}
c4 nrep ninp {c_4}
r5 ne no {r_o}
vi nin 0 {vin} ac {1-prb*prb}
v2p15 np15 0 15
vm15 nm15 0 -15
q1 nc nb ne 2n2222
r6 np15 nc 50
r7 noutlp nb 10
d1 ngl no yellow
vl ngl 0 0
.options TEMP = 27
.options TNOM = 27
.ic 
.dc vi 0.1 5.0 0.1
.end
```

The reconfigure script for Xyce is:

```
../Xyce-7.0/configure \
CXXFLAGS="-O3 -g -std=c++11" \
CFLAGS="-O3 -g" \
ARCHDIR=/opt/Xyce \
CPPFLAGS=-I/usr/include/suitesparse \
--prefix=/opt/Xyce
```

I have debugged it a little, and it fails at:

```
(gdb) frame 2
#2  check_IC_NODESET_OptionBlock (this=0x2837610, this@entry=0x2835578)
    at ../../../Xyce-7.0/src/IOInterfacePKG/N_IO_DistToolBase.C:347
347             if (it->begin()->tag() == "SUBCKT")
(gdb) print it->begin()->tag()
Attempt to take address of value not located in memory.
(gdb) print it
$1 = {_vptr.OptionBlock = <vtable for Xyce::Util::OptionBlock+16>,
  name_ = "IC",
  expressionsAllowed_ = Xyce::Util::OptionBlock::ALLOW_EXPRESSIONS,
  netlistLocation_ = {fileNumber_ = 1, lineNumber_ = 128},
  params_ = empty std::list}
```

Thanks in advance

tvrusso, 2020-04-04:
Thank you for your report.  Xyce should not segfault on an issue like this and we will have to fix it.  However, the source of the problem is that your ".ic" line is ill-formed.  What's happening is that the input parser is assuming it can read a second field on the .ic line, and you have none.  

Please review the syntax of the .IC line in the reference guide.  If you need to set an initial condition, specify what that initial condition is on the .ic line.  Otherwise, just remove that line altogether.

jmgc, 2020-04-04:
Thanks for your answer. I will correct it. However, why does it work smoothly in MacOS?

tvrusso, 2020-04-04:
It is entirely operating system dependent as to what happens when you try to dereference the iterator returned by "begin()" on an empty list.  

It doesn't crash on FreeBSD, either.  Using a tool like valgrind, however, does report that the illegal dereference is happening, even though it doesn't crash the program there.

It is still a mistake to be doing this dereferencing without checking, and we'll need to correct the code to report the ill-formed .IC line earlier, and not try to press blindly ahead with working on that line when it is ill-formed.

jmgc, 2020-04-04:
Thanks another time.

tvrusso, 2020-04-19:
Fixed in commit 7122423.

closed 2020-04-19 by tvrusso