Xyce/Xyce #7: how to install Xyce

aashishgethub, 2020-05-23:
Please tell me how to install Xyce, I cloned it by git clone --recursive https://github.com/Xyce/Xyce.git
then in the Xyce directory there is no 'configure' file present. When I run autoconf it gives error'
`configure.ac:29: error: possibly undefined macro: AM_SILENT_RULES
      If this token and others are legitimate, please use m4_pattern_allow.
      See the Autoconf documentation.
configure.ac:38: error: possibly undefined macro: AM_INIT_AUTOMAKE
configure.ac:99: error: possibly undefined macro: AM_CONDITIONAL
configure.ac:531: error: possibly undefined macro: AC_CXX_HAVE_STL
configure.ac:557: error: possibly undefined macro: AC_FINITE_NUMBER_CHECK
`
Please help me. Thanks

tvrusso, 2020-05-24:
Please read the building guide on the Xyce web site.

https://xyce.sandia.gov/documentation/BuildingGuide.html

closed 2020-05-24 by tvrusso

aashishgethub, 2020-05-25:
OK thanks