Xyce/Xyce #8: .LIB cannot find model libraries at a given path.

eskiyerli, 2020-09-06:
The .LIB statement does not find the library location correctly in Linux. For example:
.LIB "/home/eskiyerli/OneDrive/Documents/Glade/xxxx/models/xyce/xxxxx.l" TT

I changed the process and library name to keep confidentiality.

This is the output of Xyce:
*****
***** Welcome to the Xyce(TM) Parallel Electronic Simulator
*****
***** This is version Xyce DEVELOPMENT-202007051804-g-cc01fe5-opensource
***** Date: Sun Sep 06 10:20:34 CEST 2020

***** Executing netlist /home/eskiyerli/Documents/Glade/xxxx/Simulations/design_xxxx/rcFilter/rcFilter_schematic_TRAN.sp

***** Reading and parsing netlist...
Netlist error: Could not find include file xxxx.l
Netlist error: Could not find include file xxxx.l

Again, I changed the relevant names. The paths are correct. If I remove the paths and place the model library as in the same directory as simulation deck, Xyce runs successfully.

tvrusso, 2020-09-06:
Could you be more specific about what doesn't work for you?  This could be a bug or could be a reasonable feature request, but as it stands I can't tell.

As a general statement that .lib can't find model libraries given an absolute path, I am unable to reproduce this in any simple manner.  

.lib certainly does work with absolute paths specified.  But specifying such an absolute path in a library file name  in the top level netlist does not in any way change the search path it will use for any hierarchically included libraries.  Perhaps that's the issue here.  That is, if you include a library using an absolute path that tries to include other libraries in the same directory with relative paths, that will fail.

I have tried this (a simple modification of the test suite's "BUG_907_SON" certification test):
In the file `/my/directory/issue8_netlists/test2.cir`:
```
% Test of hierarhical libraries

vd d 0 dc 1.8
vg g 0 dc 0

r1 g d 1k

.dc vg 0 1.8 0.01

.print dc i(vd) i(vf) 

.param x2  = '(x)/3.0'

.lib /my/directory/issue8_libs/lib3.lib lib4_top

.end
```
Then in the files `/my/directory/issue8_libs/lib3.lib` and `/my/directory/issue8_libs/lib4.lib`:
lib3.lib:
```
.lib lib4_B
.param x = 1
vf f 0 dc 1.5
.endl lib4_B

.lib lib4_top
.lib /my/directory/issue8_libs/lib4.lib lib4_A
.lib /my/directory/issue8_libs/lib3.lib lib4_B
.endl lib4_top
```

lib4.lib:
```
.lib lib4_A
r2 g f 2k
.endl lib4_A
```

This works fine so long as lib3.lib's internal ".lib" statements *also* use absolute paths.  Changing these to relative paths does NOT work.  Relative paths names are always relative to the directory in which the _netlist_ lives.

Is this the specific problem you're having?

eskiyerli, 2020-09-17:
I only have a single lib file divided to lib sections.

tvrusso, 2020-09-17:
> I only have a single lib file divided to lib sections.

Strange, then.  Because I am able to use a single .lib file with an absolute path with no problem, and am unable to reproduce this as a general rule.

Cut and paste this .lib file into "lib2.lib" and save it into one directory:

-----
.lib lib4_top
r2 g f 2k
.param x = 1
vf f 0 dc 1.5
.endl lib4_top
-----

then cut and paste (with obvious edit of the path in the .lib line) this netlist into a different directory.

----
*Test library path issue
vd d 0 dc 1.8
vg g 0 dc 0
r1 g d 1k
.dc vg 0 1.8 0.01
.print dc i(vd) i(vf)
.param x2  = '(x)/3.0'
.lib /explicit_path_to/lib2.lib lib4_top
.end
-----

When I run Xyce on that netlist, it finds the .lib just fine.

Since I can't reproduce with a simple test (other than the one I brought up earlier, where hierarchical libs do NOT work unless all paths are explicit), I'll have to ask you to come up with a complete example of a failing case that is free from distribution restrictions and post it here.

tvrusso, 2020-09-17:
Just one possibility that dawned on me:  are there by any chance any spaces in the "xxxx" part of the path you've redacted?  That might explain the problem.

eskiyerli, 2021-01-01:
Sorry for the late reply. This is the error I get from Xyce:
Netlist error: Could not find include file cr018gpii_v1d1.l
Netlist error: Could not find include file cr018gpii_v1d1.l

These are the lib statements:
.LIB "/home/eskiyerli/Documents/Glade/xxxxxx/models/xyce/cr018gpii_v1d1.l" TT
.LIB "/home/eskiyerli/Documents/Glade/xxxxxx/models/xyce/cr018gpii_v1d1.l" stat_noise
What is worrying is that it cannot seem to be able to traverse path hierarchy.

On the other hand, Xyce can find the include file without problems:
.INC "/home/eskiyerli/Documents/Glade/xxxxxx/Simulations/design_xxxxx/tb_diffPair/tb_diffPair_schematic.cir"

There are no spaces anywhere in the paths.

eskiyerli, 2021-01-01:
I think the problem is with the model library converted from HSPICE format using XDM. 
Xyce expects 
.lib TT
models
.endl TT

However HSPICE accepts
.lib cr018gpii_v1d1.l TT
.endl TT

This is where Xyce chokes.

eskiyerli, 2021-01-01:
Adding the full path after .lib statements solves it. You were right in your earlier comments about the need for the full path. Closing the issue. Of course, it might be wise to mimic HSPICE behaviour as this makes it easier to use without needing the full path.

closed 2021-01-01 by eskiyerli