Xyce/Xyce #19: Xyce terminates API user

pepijndevos, 2021-03-04:
I am using the `GenCouplingSimulator` API to build a long-running wrapper around Xyce.
However, I found that when there is a parse error in the arguments or the spice netlist, the `initialize` call just terminates the entire program.
Is there any way to avoid this?

I found that I can use `Xyce::set_report_handler(report_handler);`, but if I do not exit inside it I get `std::exception: Failed to exit on fatal error`, and if I throw my own exception inside the handler it just terminates the program anyway. I could catch the exception that Xyce throws, but this seems a bit improper.

Is there a good way to terminate the Xyce instance without terminating the whole program?

tvrusso, 2021-03-04:
Why do you think that catching the exception that Xyce throws would be improper?    This is the first thing I would have suggested to get around this for the moment, and it seems the ideal way to work around the problem.

We'll have to look into why initialize simply terminates the program on a parse error and whether there's a good way to handle it properly.   The error handling of Xyce parsing is not really designed for this use case, and may not play well with it.  The external coupling interface tries to leverage existing Xyce capabilities at a very high level, and it could be that parts of the parser are forcing a fatal error at a place that makes sense when Xyce is used normally and no sense at all when used with this coupling interface.  Investigating and fixing that is worth doing, but I wouldn't expect it to be a quick fix.

Using try/catch blocks and exceptions (even those that you throw yourself) to handle this would probably be the safest and quickest way to get past it while we look into it.

tvrusso, 2021-03-04:
I see the issue already --- the "safeBarrier" function in the error handling package is called periodically throughout the run at "safe" places to exit (in parallel, this means when all processors are at the same barrier).  If any fatal errors have happened up to that point, the code emits a message using the UserFatal0.die() method.  This method (very indirectly) calls the function you pass to set_report_handler, as you figured out.

The default report_handler that Xyce sets on its own handles exiting the code in what is supposed to be a safe way when run under MPI.

The throw of an exception you're seeing is there precisely to handle the case where somehow the report handler has *not* exited when there is a fatal error condition that Xyce cannot process further.  It was put there (by a developer who is no longer here) pretty much precisely to handle the case where someone has overridden the default behavior and to allow that person to handle the "fatal" error some other way than exiting.

The entire "report_handler" thing is poorly documented and not at all easy to follow in the code.  To date nobody has ever tried to override the default, even the group for which the general coupling interface was first developed.  It's impressive that you worked out that you could get past things this way.

But rather than saying that it would be improper to catch the exception, or that it is a mere work around for broken behavior, I'd say it was very much exactly what the person who set up the system had in mind for your exact use case.

Write your own report handler function as you have done, and don't make it call exit().  Then catch the exception that is telling you that Xyce has given up instead.

pepijndevos, 2021-03-04:
Thanks for clarifying! It was mostly the exception message that made me think it is a mistake for the error handler not to exit. And throwing a custom exception terminates anyway. But glad to learn catching the generated exception is a valid approach.

tvrusso, 2021-03-04:
Glad to have helped.  It was a great question and one that has never come up before.  Answering it required digging into a part of the code that is rarely revisited.

I think there was very little thought behind the message in the exception other than to explain why it was thrown.  It says "Failed to exit on fatal error" but really all it means is "Fatal error detected and hasn't been handled otherwise yet."

In almost all cases ever run to date, execution has never reached that throw (it is only emitted when the report handler doesn't exit, and the only handler we ever use DOES exit), so nobody cared what it said.  

Its intent, however, is clearly to provide an alternate means of detecting fatal errors other than an abrupt exit so that users like you can change that behavior by catching the exception and doing something else.  If you catch that exception you can be sure that there is absolutely no point in continuing to process, and it should be safe to destroy the simulator object.

tvrusso, 2021-03-04:
If you are happy with where things stand, please feel free to close this issue.

pepijndevos, 2021-03-05:
Actually I'm trying the approach of just ignoring the error and it appears that after Xyce "completes" it leaves the API in a non-functional state. If there is an error, subsequent runs don't produce any output and also error. If Xyce initializes correctly, the object is freed, and another instance is created, a segfault occurs.

```c++
        xyce = kj::heap<Xyce::Circuit::GenCouplingSimulator>();
        Xyce::set_report_handler(report_handler);
        try {
            xyce->initialize(tmpargs.size(), const_cast<char**>(tmpargs.data()));
        } catch (std::runtime_error &e) {
            xyce = nullptr;
            // e = std::runtime_error("Fatal error in parsing netlist, check logs");
            throw;
        }
```
(code uses https://github.com/capnproto/capnproto/blob/master/c++/src/kj/memory.h )

```
libxyce.so.0!Xyce::Stats::Stat::getStatSet() const (Unknown Source:0)
libxyce.so.0!Xyce::Stats::Stat::Stat(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, Xyce::Stats::Stat) (Unknown Source:0)
libxyce.so.0!Xyce::Stats::StatTop::StatTop(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (Unknown Source:0)
libxyce.so.0!Xyce::Circuit::Simulator::initializeEarly(int, char**) (Unknown Source:0)
libxyce.so.0!Xyce::Circuit::Simulator::initialize(int, char**) (Unknown Source:0)
```

It seems Xyce is really not designed to be invoked repeatedly from long-running processes. At this point I'm wondering maybe I should either just shell out to Xyce and parse the raw file, or spawn a subprocess that runs the simulation and exits. (or would a thread suffice? It'd probably share the same global data that causes the current problems)

tvrusso, 2021-03-05:
If the simulation fails, then I would expect the GenCouplingSimulator object to be in a state that can't be reused.  If you want to try a different simulation, it should be necessary to destroy the object and allocate a new one.  In fact, a new one should be allocated for each simulation --- as you've noticed, the object does have state and trying to reuse it is not likely to work.  But the package as a whole should not have persistent state and reallocating a new simulator object *should* reset everything (and if it doesn't there's likely a bug to fix).

You're right, Xyce is not currently designed to be part of a long running processThe current API really just exposes most of what the monolithic Xyce binary does itself --- so it's meant to run one simulation and then exit.  There is a goal of making a better API that has more of an interactive mode, but we're not there yet and can't promise when it might exist at this time.

Have you tried deleting the object and creating a new one for each simulation you need to perform?  That *should* work, and if it doesn't there's a real bug that might be easier to fix than it would be to implement the desired interactive-mode API.

tvrusso, 2021-03-05:
Could we ask what sort of "long running process" you're trying to create?  Is it some sort of user interface that allows repeated runs of Xyce as a sort of back-end simulator?  Are you trying to use a persistent Xyce object almost as a simulation server?

The use of the persistent object to perform multiple simulations is not a happening deal with the current design, but ideally (if there are not real bugs) allocating separate simulator objects for each simulation run *should* work (each would have to go through the whole "initializeEarly/initializeLate/runSimulation" process from the beginning).

pepijndevos, 2021-03-05:
I am in fact allocating a new xyce object for each run as shown by my code snippet. Maybe these "safe points" persist somehow? I can try to make a small reproducer tomorrow.

I'm indeed making a simulation server with capnproto. But creating a new instance is fine.

tvrusso, 2021-03-05:
Hmmmm.

A simple, reproducible example that we can use to debug would help.

The best we could do right now would be to make sure that independently allocated simulator objects are truly independent and don't keep state from previous allocations --- that is what we *expect* and *intend* to be the case, and it sounds like there may be some static allocations or singletons that snuck into the code despite our best efforts to eradicate them.

pepijndevos, 2021-03-06:
Here is a testcase:

```c++
#include <iostream>
#include <Xyce_config.h>
#include <N_CIR_GenCouplingSimulator.h>
#include <N_ERH_Message.h>

void report_handler(const char *message, unsigned type) {
    std::cout << message;
}

int main(int argc, char *argv[])
{
        Xyce::Circuit::GenCouplingSimulator xyce1;
        Xyce::set_report_handler(report_handler);
        try {
            xyce1.initialize(argc, argv);
        } catch (std::runtime_error &e) {
            std::cout << std::endl << "first instance crashed" << std::endl;
        }
        Xyce::Circuit::GenCouplingSimulator xyce2;
        Xyce::set_report_handler(report_handler);
        try {
            xyce2.initialize(argc, argv);
        } catch (std::runtime_error &e) {
            std::cout << std::endl << "second instance crashed" << std::endl;
        }
}
```
On my machine, this produces the following output, where `foo.txt` is an intentionally invalid netlist.
```
XyceSimServer foo.txt 

*****
***** Welcome to the Xyce(TM) Parallel Electronic Simulator
*****
***** This is version Xyce Release 6.12-opensource
***** Date: Sat Mar 06 08:53:39 UTC 2021

***** Executing netlist foo.txt

***** Reading and parsing netlist...
Netlist error: No analysis specified.Simulation aborted due to error.  There are 0 MSG_FATAL errors and 1 MSG_ERROR errors
first instance crashed
Simulation aborted due to error.  There are 1 MSG_FATAL errors and 1 MSG_ERROR errors
second instance crashed
```
The first time it prints the normal intro message and then errors. The second time it immediately says there is a fatal error.

tvrusso, 2021-03-06:
Wow, it doesn't get simpler than that example.  That example is exactly what I would have expected to work correctly, and it should not be very hard to track down why it doesn't.

tvrusso, 2021-03-08:
This issue has been assigned internally as "gitlab-ex#150"

tvrusso, 2021-03-14:
Yes, as it turns out there is a singleton map of error counts kept by the error manager that is never reset --- obviously, since almost all uses of Xyce have been allocating simulator objects do so once, run the simulation, and exit, this has never been an issue.  There is no separate set of counts kept that is local to a single simulator object --- if you allocate more than one it may be possible to run both of them simultaneously, but their error message counts will be shared between them, and it will probably be a mess.

I am experimenting with making the Simulator base class clear the message counts upon instantiation.  It involves creating just one extra function in N_ERH_Messenger.C and calling that function in the Simulator constructor right after the call to set_report_handler.  This will at least fix your use case (it does) and will likely not break any of our existing use cases (I'm checking this now).

The weekly push of Xyce development code to Github has already happened this week, so even if I get this nailed today it won't appear on Github until at least March 22nd (and could be even later).

tvrusso, 2021-03-14:
While this will not be available on github for a week or so, you can apply the attached patch to your git master branch checkout and get the simplest fix I could make.    All it does is reset the error manager's message counts upon creation of a simulator object.  Apply this patch and reinstall Xyce, and your code should stop showing the behavior you saw above.

This will only fix the sequential declaration/use of Simulator objects like in your example code (though I personally wrapped those blocks so that the GenCoupSimulator object went out of scope and got deleted after each use).  It will NOT be safe to run multiple objects simultaneously in separate threads --- the use of static data would allow each object to interfere with all the others so that if any one of them exited with error they all would (though it is *possible* that they would run fine if they did not produce any errors, but I can't promise).  Fixing the error manager so it doesn't use static data like this is likely a much, much bigger effort and may not happen anytime soon.

This patch will be in the next batch of commits that make it to github.

[issue19_patch.diff.txt](https://github.com/Xyce/Xyce/files/6137503/issue19_patch.diff.txt)

pepijndevos, 2021-03-15:
Thanks! For now I'm not too worried about parallel simulations.

tvrusso, 2021-03-20:
The problem of this issue should be fixed by the commits above.  Pull the lastest Xyce master branch and a second allocated Simulator (or GenCouplingSimulator) object should no longer terminate abnormally just because a prior one had.

You still have to allocate a new Simulator object for each netlist, and it is best if you destroy the old one first.

I would not advise trying to make a single binary run more than one netlist simultaneously (e.g. in separate threads, each with its own Simulator object) because the singleton error handling manager will almost certainly not cope well with that.  For that, you'd want subprocesses running completely independent programs (that could have "output" code that actually communicates with the controlling process rather than writing to files somehow).  But you said that's not really your concern right now, so the current state of affairs should fix the problem you were seeing.

closed 2021-03-20 by tvrusso