Xyce/Xyce #21: Xyce does not prepend TIME vector

pepijndevos, 2021-03-11:
As per https://xyce.sandia.gov/documentation/AppNote-GenExt.pdf
> If an output interface is intended to be used in transient or frequency domain simulation(as opposed to DC steady state), it is advisable to include the independent variable “TIME”or “FREQUENCY” in the list of requested outputs. **If this is omitted, Xyce will prepend the appropriate variable to the list.**  No independent variable is prepended for DC simulation, or if the appropriate variable already appears anywhere in the requested list.

However, as of Xyce 6.12 this appears to be not the case. With the following code, I only get the voltage and current vectors, and no TIME vector.

```
    void requestedOutputs(std::vector<std::string> & outputVars) {
        outputVars.push_back("V(*)");
        outputVars.push_back("I(*)");
    };
```

tvrusso, 2021-03-11:
Hmmmm.  It's supposed to.  I have just verified that it does not in fact do so.  Thank you for your report.  This shouldn't be hard to fix.

tvrusso, 2021-03-11:
Thank you for reporting this.  Turns out that while I had gone through some efforts to write the functions needed to scan the list of requested outputs and prepend TIME or FREQUENCY if they weren't in the list, I never actually *called* those functions anywhere, and never tested the feature before writing the app note claiming that it worked that way.  In fact, it has never worked as advertised :roll_eyes: .

Fortunately, the code is all there to make it work and all that was missing was one line to actually *call* that code.  This fix will be in the next batch of commits posted to github (most likely on Monday) and in Xyce 7.3.  But if you want the fix now, it is a one line fix.

Edit src/IOInterfacePKG/N_IO_ExtOutWrapper.C and find the constructor, which currently reads:
```
ExternalOutputWrapper::ExternalOutputWrapper ( ExternalOutputInterface * extIntP
tr)
  : extIntPtr_(extIntPtr)
{
  checkVars_();
}
```

Change this to:
```
ExternalOutputWrapper::ExternalOutputWrapper ( ExternalOutputInterface * extIntP
tr)
  : extIntPtr_(extIntPtr)
{
  checkVars_();
  normalizeVarList_();
}
```

The normalizeVarList_() function is what prepends TIME or FREQUENCY to the list if it isn't there already.  It was simply not being called anywhere.  This makes it be called upon instantiation of the external output wrapper.

tvrusso, 2021-03-13:
The commits referenced above fix this issue.

Thank you for reporting it.  I'm glad it was so easy to fix this one.  Your other issue on the error handling is deeper and will take more work.

closed 2021-03-13 by tvrusso