Xyce/Xyce #23: add file search path option relative to calling file

jeras, 2021-04-05:
Skywater PDK is using `.include` and `.library` paths relative to the calling file, but Xyce only supports file paths relative to the execution folder.

tvrusso, 2021-04-05:
This has already been fixed in the last week or two and should be present in the master branch of Xyce right now.

closed 2021-04-05 by tvrusso

Karlsefni2012, 2021-04-05:
This has already been partially fixed on the Xyce development branch, and is the subject (along with other Skywater issues) of much discussion here:

https://github.com/google/skywater-pdk/issues/186

jeras, 2021-04-05:
Thanks for the link, I will go through the discussion now.
Just in case more issues were solved, is there a commit I could safely use?

Karlsefni2012, 2021-04-05:
To get the recent fixes you'll have to build Xyce from source.   If you clone the master branch from github you'll get all the recent updates.

We're intending to support Skywater, but there are still a few known things we haven't addressed.  You mentioned (under another github issue) the comment delimeter "$" character.  Xyce does not recognize that, for example.

jeras, 2021-04-05:
Thanks, I already compiled Xyce on Ubuntu 20.04, so I should be able to compile master, I just was not sure, how stable maser was. I could mention here, the "OpenMPI" test did not return any "BAD", so OpenMPI might have been built correctly in at least in version 20.04 of Ubuntu.

The `$` issue is rather easy to fix with a search/replace.
I would assume, there is no spice standard (like Verilog LRM) which could be used to decide how to best handle such issues.

Karlsefni2012, 2021-04-05:
We only push to master if all the tests pass.  So, master should be very stable.

Yes, unfortunately the SPICE language has never been standardized.  There is an effort to come up with a standard within the Si2 organization, but not sure how far along that is.

We have a tool, XDM, which is for translating netlists from various formats (Pspice, Hspice and Spectre) to Xyce.    The Skywater files are all ngspice format however, so that isn't covered by that tool yet.

tvrusso, 2021-04-05:
> Thanks, I already compiled Xyce on Ubuntu 20.04, so I should be able to compile master, I just was not sure, how stable maser was. I could mention here, the "OpenMPI" test did not return any "BAD", so OpenMPI might have been built correctly in at least in version 20.04 of Ubuntu.

OpenMPI was fixed for Ubuntu 20.04.  It remained broken in every release since 17.10 up until then.  It is still broken in 18.04 LTS, even though that is an LTS system and has several more years of life.

jeras, 2021-04-06:
I tested a build from master and include paths used in Skywater 130nm PDK are working properly.