Xyce/Xyce #24: `libADMS` and `libNeuronModels` depend on symbols defined in `libxyce`

staticfloat, 2021-04-26:
Hello there, we are trying to wrap `libxyce` in [Julia](https://julialang.org/), a dynamic language that usually loads external dependencies via `dlopen()`.  When attempting to load `libADMS.so`, we get undefined symbol errors such as the following:

```
ERROR: could not load library "libADMS.so.0.0.0"
libADMS.so.0.0.0: undefined symbol: _ZTIN4Xyce6Device14DeviceInstanceE
```
(Passing that through `c++filt` shows that the missing symbol is `Xyce::Device::DeviceInstance`)

This is because `libADMS.so` is using a symbol that is defined within `libxyce`, but `libADMS.so` has no `-lxyce` on its link line.  This is not a problem when linking with the C linker, but when using the dynamic linker (e.g. using `dlopen()`) even though we load `libxyce.so` first, some platforms (such as Linux) do not have `RTLD_GLOBAL` semantics by default, which causes `libADMS` to fail in its lookup because the symbols loaded previously by `dlopen("libxyce.so")` are namespaced and only available to modules that express a dependency on `libxyce`.  We tend to discourage use of using `RTLD_GLOBAL`since it pollutes the global namespace, and in a dynamic language it is difficult to predict what symbols will be loaded by different packages, so it is very easy to accidentally load two different libraries that eventually conflict with eachother in that global namespace.

I see that the build system seems to treat `libADMS` as dependency of `libxyce`, even though it is not necessarily linked in, so it may be challenging to figure out the right way to set up the dependency graph such that `libADMS` can properly express a dependency on `libxyce`, but perhaps that is just because I am unfamiliar with the build system here. :)

tvrusso, 2021-04-26:
You are absolutely correct, the libADMS and libNeuronModels libraries are not set up to be built with the understanding that they have dependencies on libxyce.  We are painfully aware of it and it is on our agenda to address it, but the build system is in flux --- so while it should be possible to fix it in the autoconf system by appropriate tweaking, there is a big push to change to a CMake build system which has other challenges.  As a result, there is some inertia to doing any big tweaks to the autoconf system at the moment.

The primary target systems for Xyce internally are Linux where we use our own main program linked to all those libraries, and this has not yet become a serious problem for us given that libtool is setting up an RPATH in the binaries to resolve these internal inconsistencies.

We do want to address this specific issue, though, as it causes problems for other reasons, notably in the generation of shared library plugins (which also need libxyce).  It just hasn't bubbled up to the top of the priority list yet.  It may be solved in the CMake system when that becomes default, and I will attempt to work out the issues in autoconf.  Unfortunately, this issue will linger past the 7.3 release, which is already frozen.

it really should just be a matter of adding a few lines to the Makefile.am for the ADMS and NeuronModel directories, but the question is "which few lines?"  Answering that question takes a little more thought and time than has been directed at it so far.

tvrusso, 2021-04-26:
For what it's worth, this is issue 204 in our internal issue tracker.

I'm leaving it open here.  I *hope* to get a chance to work on it in the coming months, but can promise nothing.

Keno, 2021-04-26:
Could you clarify what the intended behavior is? It seems to me like we should just be linking libADMS and libNeuronModels statically into libXyce rather than having them be separate dynamic libraries. Is that correct?

tvrusso, 2021-04-26:
Well, yes, it *could* be done that way, making libADMS and libNeuronModels be treated just as "convenience" libraries the way other subdirectories are.

Separating them out was a part of an unfinished refactor that was going to allow different model subsets to be "plugins" in some way.  That effort is long abandoned, and perhaps the simplest fix is just to include them in libxyce like all the other convenience libraries in all the other subdirectories.

Keno, 2021-04-26:
> That effort is long abandoned, and perhaps the simplest fix is just to include them in libxyce like all the other convenience libraries in all the other subdirectories.

Yes, that sounds good. Plugins + autotools is a complicated topic, so if there are no plans to push forward on either, probably best to just go back to the simplest possible setup.

tvrusso, 2021-04-26:
I'm working on how this might be fixable, and the path of least resistance is as we said, to bundle the libADMS and libNeuronModels stuff into libxyce and not make them separate.  This is probably what we'll do, but it won't be done for the next release 7.3.  You might want to hack in a bandaid solution in the meantime.   The rest of this comment suggests something to try on your end if you want to get past this problem sooner than we can address it on our end.

If you look in src/Makefile.am, you'll see that a variable XYCELINK is defined that includes the ".la" files for each subdirectory (these are the "convenience libraries" that are built for each directory but never installed), and then libxyce uses that variable to determine what to add in using a "LIBADD" variable later in the file.  Everything that is in XYCELINK gets bundled together into the monolithic libxyce, and this does  not currently include any of the optional model subsets.

You'll also see that there are a bunch of conditionals that set up extra libraries to link in to the main Xyce executables depending on what was built (things like libADMS, libNeuronModels, and a few that are not part of the open source code).  These are each then individually linked into the monolithic Xyce binary later on where there's a "Xyce_LDADD" line.

For example:
```# conditionally link in ADMS device models.
if ADMSMODELS
    ADMSLD = $(top_builddir)/src/DeviceModelPKG/ADMS/libADMS.la 
else
    ADMSLD =
endif

# conditionally link in Neuron device models.
if NEURONMODELS
    NEURONLD = $(top_builddir)/src/DeviceModelPKG/NeuronModels/libNeuronModels.l
a 
else
    NEURONLD =
endif

```

adds libADMS.la to the link line for Xyce later on, where $(ADMSLD) and $(NEURONLD) are used in Xyce_LDADD.

If you look at the Makefile.am files for all those other directories, they define `noinst_LTLIBRARIES` to state that a convenience library should be built but not installed.  But in the Makefile.am for the ADMS and NeuronModels directory, there's a `lib_LTLIBRARIES` instead, indicating the library should be built and installed to ${prefix}/lib.

If one changed the conditional in src/Makefile.am to:
```
# conditionally link in ADMS device models.
if ADMSMODELS
    XYCELINK = ${XYCELINK} $(top_builddir)/src/DeviceModelPKG/ADMS/libADMS.la 
endif

ADMSLD =

# conditionally link in Neuron device models.
if NEURONMODELS
    XYCELINK = ${XYCELINK} $(top_builddir)/src/DeviceModelPKG/NeuronModels/libNeuronModels.la 
endif
NEURONLD =
```
and the `lib_LTLIBRARIES` to `noinst_LTLIBRARIES` in src/DeviceModelPKG/ADMS/Makefile.am and src/DeviceModelPKG/NeuronModels/Makefile.am, then I believe the issue will be solved for you and those two libraries will no longer be separate and libxyce will contain both of them.

Because this will not be in place for 7.3 (it's already too late in the release process for that), I suggest you try that in the short term.  I will probably follow a similar path to fix it on our end, but it may not be for a few weeks, after the team has gotten 7.3 out the door.

tvrusso, 2021-04-26:
I see that in your https://github.com/JuliaPackaging/Yggdrasil/pull/2899 link, you are trying to change "lib_LTLIBRARIES" to "noinst_LTLIBRARIES" to try to solve this.  That is only half the issue.  You'll also have to change XYCELINK in the main src/Makefile.am.

Keno, 2021-04-26:
Appreciate the detailed reply, we'll be sure to try it that way!

tvrusso, 2021-04-26:
> Appreciate the detailed reply, we'll be sure to try it that way!

Please let us know if this fixes the problem for you.  It is very likely what we'll do on this end.

I went back and looked at a much older version of Xyce (around 2012, long before we released to the public) and it used to be this way.  It was turned into the way it is now as part of that unfinished attempt to turn optional model sets into plugins.  So this really just undoes that unnecessary complication.

I suspect this will solve your interdependency problem.

tvrusso, 2021-04-26:
Note that I have a syntax error in my earlier suggestion, where I use things like "${XYCELINK}" --- that's shell syntax, not Makefile syntax.  It should be "$(XYCELINK)".

tvrusso, 2021-04-27:
As it turns out, my suggestion is a little too simplistic, largely due to automake arcana.

I am working on this properly today, and may have a patch available later today that you could simply apply to Xyce to make libxyce a complete library instead of a weird thing that has circular dependencies with libADMS and libNeuronModels.  It isn't very much more complex than what I said above, but there are some strange things about automake function that make the exact thing I said above throw automake errors.

When I have something that works I'll attach it here as a patch.

After the 7.3 release, assuming I really do get it to work properly, this fix will appear on the github master branch (but not until then).

tvrusso, 2021-04-27:
The attatched patch should fix the linking problem entirely: 
[github_24_patch.txt](https://github.com/Xyce/Xyce/files/6386831/github_24_patch.txt)

With this patch (which will be part of Xyce master branch in a few weeks when we push to github next), libADMS and libNeuronModels are no longer created --- what used to go into those libraries is now just part of libxyce.  You can remove all references to those two libraries from your Julia builds.

Linking to Xyce libraries means linking to "libxyce.a" or "libxyce.so" now, and all that unnecessary complication of interlibrary dependence can be forgotten (it was worse than we talked about, as there were circular dependencies --- libADMS depended on functions in libxyce, but libxyce also depended on functions in libADMS).

tvrusso, 2021-04-28:
@Keno:  Just a quick, somewhat frivolous, comment, as I see you all have pulled that patch into [your pull request](https://github.com/JuliaPackaging/Yggdrasil/pull/2899).  You've given it the name "detach.patch" because it "detaches" libADMS and libNeuronModels.  This is a misnomer.  That patch detaches nothing --- it *incorporates* libADMS and libNeuronModels into libxyce so that libxyce is now complete instead of fractured.

tvrusso, 2021-04-29:
This will go away in the next master merge, and it appears that the issue has been handled via the attached patch for Yggdrasil.

This fix will still be a manual patch on the Yggdrasil side for Xyce 7.3, which is too far along in the release process to get this patch and will be released in the next few weeks.

When building off of the Xyce master branch *after* the release of 7.3,  you will NOT need the patch anymore.

I'm leaving this issue open until the master merge, but will close it promptly after that unless something about the patch turns out to be inadequate for your needs.

tvrusso, 2021-05-17:
This issue is fixed in the development branch and was merged into github with the commit above (unfortunately, the specific commit that fixed the problem did not have the text that would have automatically gotten it referenced in this issue report, but that commit was commit Xyce@8ff0f127).

Since it is fixed in the development branch, it will be part of the next release, Xyce 7.4.

If you are building Xyce 7.3 or earlier you will still need the manual patch discussed so far.  Building off of this repo's "master" branch after today will no longer require that.

closed 2021-05-17 by tvrusso