Xyce/Xyce #26: HB simulations in Xyce 7.3

eskiyerli, 2021-05-22:
I compiled Xyce in Linux Manjaro without issues. The same setup was used to simulate in Xyce 7.2. When I try to use HB simulation, it dies with the following message:
***** Beginning Transient Calculation...

***** Problem read in and set up time: 0.935061 seconds
 ***** Interpolating transient solution for IC calculation....

/usr/include/c++/10.2.0/bits/stl_vector.h:1045: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = int; _Alloc = std::allocator<int>; std::vector<_Tp, _Alloc>::reference = int&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__builtin_expect(__n < this->size(), true)' failed.

Is it because an error in my compilation setup or due to something in the code, I can't say?  I used the following make statements in compiling Xyce source code from Github:
    cmake \
    	-B "${pkgname}-Release-${pkgver}/build" \
    	-S "${pkgname}-Release-${pkgver}" \
        -DCMAKE_BUILD_TYPE='None' \
        -D Xyce_PLUGIN_SUPPORT=ON \
        -DCMAKE_INSTALL_PREFIX='/opt/Xyce' \
        -Wno-dev
        make -C "${pkgname}-Release-${pkgver}/build" all
make -j 2 -C "${pkgname}-Release-${pkgver}/build" DESTDIR="$pkgdir" install

It is slightly different from the statements in Xyce documentation as I wanted to create a Arch Linux package to keep things tidy. I also installed adms separately.

Karlsefni2012, 2021-05-22:
Hi @eskiyerli , it is hard to say what is going on from this.  Would be possible for you to share your input file with us?

eskiyerli, 2021-05-22:
I copied the simulation deck below. I am using GCC 11 to compile Xyce. The same circuit works without problems in Xyce 7.2. However it was compiled a while back.

* Xyce Simulation Deck for /home/eskiyerli/Documents/Glade/tsmc18/Simulations/design_tsmc18/tb_diffPair_2tones/tb_diffPair_2tones_schematic.cir
.PREPROCESS REPLACEGROUND TRUE
.OPTIONS OUTPUT PHASE_OUTPUT_RADIANS=true 
.OPTIONS OUTPUT PRINTFOOTER=false 
.OPTIONS DEVICE TEMP = 25
.OPTIONS DEVICE TNOM = 25
.OPTIONS DEVICE GMIN = 1e-12
.OPTIONS DEVICE VOLTLIM = 1
.OPTIONS DEVICE LAMBERTW = 0
.OPTIONS TIMEINT METHOD = 7
.OPTIONS TIMEINT NEWLTE = 2
.OPTIONS TIMEINT ABSTOL = 1e-12
.OPTIONS TIMEINT RELTOL = 1e-6
.OPTIONS PARSER MODEL_BINNING = 1
.OPTIONS HBINT SAVEICDATA = 0
.OPTIONS HBINT VOLTLIM = 1
.OPTIONS HBINT TAHB= 1
.OPTIONS HBINT INTMODMAX = 3
.OPTIONS HBINT STARTUPPERIODS = 2
.OPTIONS HBINT NUMTPTS = 101
.OPTIONS HBINT NUMFREQ = 10,10
.INC "/home/eskiyerli/Documents/Glade/tsmc18/Simulations/design_tsmc18/tb_diffPair_2tones/tb_diffPair_2tones_schematic.cir"
.LIB '/home/eskiyerli/OneDrive/Documents/Glade/tsmc18/models/xyce/cr018gpii_v1d1.lib' TT
.LIB '/home/eskiyerli/OneDrive/Documents/Glade/tsmc18/models/xyce/cr018gpii_v1d1.lib' stat_noise
.GLOBAL_PARAM pdd=1.8
.GLOBAL_PARAM va=10m
.STEP va 10m 110m 20m
.GLOBAL_PARAM ibias=200u
.STEP ibias LIST 200u 300u 400u
.GLOBAL_PARAM vo=0.7
.STEP vo 0.7 1.1 0.2
.OP
.HB  10e6 11e6 
.PRINT HB_FD  VM(OUT) VP(OUT)  VM(INP) VP(INP) VM(OUT)
.PRINT HB_TD  V(OUT) V(INP)
.END

tvrusso, 2021-05-23:
Unfortunately, the information you have provided so far makes it difficult for us to reproduce and diagnose the problem.  So far your experience is not in line with our own, so we need a little more information to try to pin down what is going on.

Could you please provide:
- The exact CMake invocation you used for building Trilinos
- The exact CMake invocation you used for building Xyce
- Whether or not your build of Xyce passed all of the Xyce regression suite

Did you also build Xyce 7.2 with the experimental CMake system, or did you use the autotools build we have been recommending for most of Xyce's history?  Have you tried building Xyce 7.3 with the autotools build instead, and if so does it do the same thing?  This could help us pin down whether it's a bug in the new build system or not.

eskiyerli, 2021-05-23:
cmake invocation for Trilinos:
    cmake .. \
            -G "Unix Makefiles" \
            -DCMAKE_C_COMPILER=gcc \
            -DCMAKE_CXX_COMPILER=g++ \
            -DCMAKE_Fortran_COMPILER=gfortran \
            -DCMAKE_CXX_FLAGS="$FLAGS" \
            -DCMAKE_C_FLAGS="$FLAGS" \
            -DCMAKE_Fortran_FLAGS="$FLAGS" \
            -DCMAKE_INSTALL_PREFIX:PATH=/usr \
            -DCMAKE_MAKE_PROGRAM="make" \
            -DTrilinos_ENABLE_NOX:BOOL=ON \
            -DNOX_ENABLE_LOCA:BOOL=ON \
            -DTrilinos_ENABLE_EpetraExt:BOOL=ON \
            -DEpetraExt_BUILD_BTF:BOOL=ON \
            -DEpetraExt_BUILD_EXPERIMENTAL:BOOL=ON \
            -DEpetraExt_BUILD_GRAPH_REORDERINGS:BOOL=ON \
            -DTrilinos_ENABLE_TrilinosCouplings:BOOL=ON \
            -DTrilinos_ENABLE_Ifpack:BOOL=ON \
            -DTrilinos_ENABLE_Isorropia:BOOL=ON \
            -DTrilinos_ENABLE_AztecOO:BOOL=ON \
            -DTrilinos_ENABLE_Belos:BOOL=ON \
            -DTrilinos_ENABLE_Teuchos:BOOL=ON \
            -DTeuchos_ENABLE_COMPLEX:BOOL=ON \
            -DTrilinos_ENABLE_Amesos:BOOL=ON \
            -DAmesos_ENABLE_KLU:BOOL=ON \
            -DTrilinos_ENABLE_Sacado:BOOL=ON \
            -DTrilinos_ENABLE_Kokkos:BOOL=OFF \
            -DTrilinos_ENABLE_ALL_OPTIONAL_PACKAGES:BOOL=OFF \
            -DTrilinos_ENABLE_CXX11:BOOL=ON \
            -DTPL_ENABLE_AMD:BOOL=ON \
            -DAMD_LIBRARY_DIRS:PATH="/usr/lib" \
            -DTPL_AMD_INCLUDE_DIRS:PATH="/usr/include" \
            -DTPL_ENABLE_BLAS:BOOL=ON \
            -DTPL_ENABLE_LAPACK:BOOL=ON 
    make VERBOSE=1
cmake invocation for Xyce:
    cmake \
    	-B "${pkgname}-Release-${pkgver}/build" \
    	-S "${pkgname}-Release-${pkgver}" \
        -D Xyce_PLUGIN_SUPPORT=ON \
        -DCMAKE_INSTALL_PREFIX='/opt/Xyce' 

I haven't tried regression tests.

FYI, I used the following invocation as you suggested and HB anaysis is now working:
	./configure \
	--enable-shared \
	--enable-xyce-shareable  \
	--enable-silent-rules 	\
    --prefix=/opt/Xyce 	\
	CXXFLAGS="-O3 -std=c++11" \
    CPPFLAGS="-I/usr/include/" 
	make -j2
	cd utils/
	make buildxyceplugin

tvrusso, 2021-05-24:
This has nothing to do with anything, really, but "make buildxyceplugin" should not be necessary --- buildxyceplugin is made by default if "--enable-shared --enable-xyce-shareable" are given.  And "--enable-silent-rules" should not be needed, either, as it's the default.

We definitely recommend running the regression test suite with every build, just to make sure that your build passes the same tests that our own builds do.  See https://xyce.sandia.gov/documentation/RunningTheTests.html for details.

We'll have to try your CMake build and see if we can reproduce your failures, but without the exact failing netlist and all its included files, it may be difficult.

TBird2001, 2021-05-24:
Sorry I'm coming to this late. The only thing that jumps out at me regarding the first CMake invocation for Xyce is the setting of `CMAKE_BUILD_TYPE='None'`. I see that Arch Linux [recommends](https://wiki.archlinux.org/title/CMake_package_guidelines#CMake_can_automatically_override_the_default_compiler_optimization_flag) using that capability when creating packages, so I'm guessing that is why you specified it. 

Arch Linux uses `-O2` by default when building in their package system. I'm not sure what happens when specifying `CMAKE_BUILD_TYPE='None'` in Manjaro, outside of the Arch packaging environment (I suppose it depends on the environment variables). That could be what is causing the failures.

I notice that the Autotools invocation has `-O3` set explicitly. In the Xyce CMake, `CMAKE_BUILD_TYPE=Release` is the default, so eliminating that line completely would make the CMake process look the most like the Autotools build. However, in the second Xyce CMake invocation you listed, you did just that. If that build had the same issue, this is probably a red herring.

For what it's worth, I built Xyce with CMake in Majaro on an ancient laptop using the recommended build process, and all the tests passed. That was a few months ago, though.

eskiyerli, 2021-05-24:
Hi Jason,

 

Indeed, I added that line because it was recommended in Arch Linux wiki and then removed it when I had the errors. I made a search for that particular error in Google and noticed that a few Arch/Linux users were complaining about the exactly same error after compiling their software. These errors happened after April 2021. Thus it could be an error related to Arch Linux cmake or gcc package. I am not very knowledgeable about intricacies of build system and makefile to hope for much progress. 

 

I managed to build the package using autobuilds which allows to test Xyce in my circuits. So far, so good.

closed 2021-06-22 by eskiyerli