Xyce/Xyce #28: Xyce 7.3 doesn't call external output interface

pepijndevos, 2021-06-14:
We are working on a Julia wrapper for Xyce that is reaching a point where you can get Xyce up and running in a single command.
However, since upgrading to Xyce 7.3 the external output interface does not receive any data.

I've added some print statements to `requestedOutputs` and `outputFieldNames` and it appears that it requests the field names, but never gets back with the actual field names.

```
*****
***** Welcome to the Xyce(TM) Parallel Electronic Simulator
*****
***** This is version Xyce DEVELOPMENT-202105171928-g-9dd4374f-opensource
***** Date: ma jun 14 10:14:54 CEST 2021

***** Executing netlist test/VRC.sp

***** Reading and parsing netlist...
Netlist warning: No print specified
***** Setting up topology...

***** Device Count Summary ...
       C level 1 (Capacitor)                  1
       R level 1 (Resistor)                   1
       V level 1 (Independent Voltage Source) 1
       ----------------------------------------
       Total Devices                          3
***** Setting up matrix structure...
***** Number of Unknowns = 3
***** Initializing...

 addOutputInterface got object of output type 1
requesting output fieldnames
TIME
V(*)
I(*)
***** Beginning DC Operating Point Calculation...

***** Beginning Transient Calculation...
```

Xyce is built as follows:

tvrusso, 2021-06-14:
I will not be able to look at this immediately, but I do know that testing of the external output interface, including bug fixes we've made for things you've found, is done every night and every time anyone makes any change in the code.  It is puzzling that you are finding it so obviously broken in 7.3.

I note also that our test suite contains a test that uses V(*) and I(*) and *depends* on outputFieldNames returning the real names of fields.  This test has passed consistently, including in 7.3.

I will likely not be able to try to reproduce it in your Julia setup, and if you could try to produce a simple C++ code that shows the same behavior it would be helpful.

pepijndevos, 2021-06-15:
Alright, here is a C++ example
```c++
#include <Xyce_config.h>
#include <N_CIR_GenCouplingSimulator.h>

void report_handler(const char *m, unsigned i)
{
    std::cout << m;
}

// by default Xyce aborts on an error.
// As a library we don't like that
void set_report_handler() {
    Xyce::set_report_handler(&report_handler);
}

class OutputHandler final : public Xyce::IO::ExternalOutputInterface
{
public:
    OutputHandler(std::string name,
    Xyce::IO::OutputType::OutputType type,
    std::vector<std::string> outputs)
        : requested_fieldnames(outputs), type(type), name(name) { }

    std::string getName()
    {
        return name;
    }

    Xyce::IO::OutputType::OutputType getOutputType()
    {
        return type;
    }

    void requestedOutputs(std::vector<std::string> &outputVars)
    {
        std::cout << "requesting output fieldnames\n";
        for(int i=0;i<requested_fieldnames.size();i++) {
            std::cout << requested_fieldnames[i] << std::endl;
        }
        outputVars = requested_fieldnames;
    }

    void outputFieldNames(std::vector<std::string> &outputNames)
    {
        std::cout << "getting output fieldnames\n";
        for(int i=0;i<outputNames.size();i++) {
            std::cout << outputNames[i] << std::endl;
        }
        fieldnames = outputNames;
        real_data.resize(outputNames.size());
    }

    std::vector<std::string> getFieldnames() {
        return fieldnames;
    }

    std::vector<double> getRealData(unsigned int idx) {
        return real_data[idx];
    }

    void outputReal(std::vector<double> &outputData)
    {
        for(int i=0;i<outputData.size();i++) {
            real_data[i].push_back(outputData[i]);
        }
    }

    void outputComplex(std::vector<std::complex<double>> &outputData)
    {
        // TODO: how to expose std::complex to julia?
    }

    std::string name;
    Xyce::IO::OutputType::OutputType type;
    std::vector<std::string> requested_fieldnames;
    std::vector<std::string> fieldnames;
    std::vector<std::vector<double>> real_data;
};

int main(int argc, char** argv) {
    Xyce::Circuit::GenCouplingSimulator sim;
    sim.initialize(argc, argv);
    std::vector<std::string> vec;
    vec.push_back("TIME");
    vec.push_back("V(*)");
    vec.push_back("I(*)");
    auto oi = OutputHandler("handler", Xyce::IO::OutputType::OutputType::TRAN, vec);
    sim.addOutputInterface(&oi);
    sim.runSimulation();
    std::cout << "the field names" << std::endl;
    for(auto f : oi.getFieldnames()) {
        std::cout << f << std::endl;
    }
}
```

On my side this never prints the second callback and prints no fields at the end.

I'm worried it might be some nasty linking issues, but hopefully it's easy to reproduce and fix :crossed_fingers: 

If it's useful I could look into supplying an `rr` trace to make it more reproducible. Also happy to do more debugging on my side.

tvrusso, 2021-06-19:
Unfortunately, it may take me a while before I can get to trying out your test case and finding how it differs from our own (which clearly works).

It is high on my list of priorities to look into this, but I may not get to it for a week or two.

tvrusso, 2021-06-21:
The problem here is your use of "initialize" prior to the addOutputInterface call.  This is incorrect usage.

You must use "initializeEarly" instead, then call the addOutputInterface, then call "initializeLate".

That is, the addOutputInterface calls must happen between initializeEarly and initializeLate.  initializeLate does final setup, and calling "addOutputInterface" after initializeLate has been called means that Xyce thinks it is already done setting up outputters, so your subsequent calls are not actually doing the right thing.

"initialize" is basically the same as calling initializeEarly and initializeLate one after the other.

This example C++ code will work:
```

#include <Xyce_config.h>
#include <N_CIR_GenCouplingSimulator.h>
#include <N_IO_ExtOutInterface.h>
#include <iostream>
#include <fstream>
#include <iomanip>

class ioTestInterface : public Xyce::IO::ExternalOutputInterface
{
public:
  ioTestInterface(std::string filename,Xyce::IO::OutputType::OutputType theOutputType=Xyce::IO::OutputType::TRAN)
    :
    theOutputType_(theOutputType)
  {
    os_ =new std::ofstream(filename.c_str(),std::ios_base::out);
  };

  ~ioTestInterface()
  {
    delete os_;
  };

  Xyce::IO::OutputType::OutputType getOutputType()
  {
    return theOutputType_;
  }

  void addOutputString(std::string theString)
  {
    outputStrings_.push_back(theString);
  };

  void requestedOutputs(std::vector<std::string> &outputVars)
  {
    outputVars = outputStrings_;
  };

  void reportParseStatus(std::vector<bool> & statusVec)
  {
    for (int i=0;i<statusVec.size();i++)
    {
      std::cout << "String " << outputStrings_[i] << " is "
                <<  ( (statusVec[i])? " parseable" : " unparseable")
                << std::endl;
    }
  };

  void outputFieldNames(std::vector<std::string> & outputNames)
  {
    // The current "external output" feature doesn't do Xyce's "Expand
    // complex types" manipulation that some output formats do.  So
    // for frequency domain output, we *ALWAYS* get a single column
    // name, and always get complex data for each column.  Some of
    // these are nonsense, especially index and frequency, for which
    // we should only output the real part.  There are others like VR, VI, VDB
    // VM, VP and so on, but we are NOT handling those here.  This is a
    // demo code, not a fully functional standard print formatter.

    hideImag_.resize(outputNames.size(),false);
[... 121 lines cut ...]
    run_status = xyce.runSimulation();
  }
  catch (std::exception &x)
  {
    run_status=Xyce::Circuit::Simulator::ERROR;
  }

  // the "finalize" function closes output files and
  // releases memory.
  if (run_status != Xyce::Circuit::Simulator::DONE)
  {
    try
    {
      xyce.finalize();
    }
    catch (std::exception &x)
    {
      run_status = Xyce::Circuit::Simulator::ERROR;
    }
  }
  exit ((run_status==Xyce::Circuit::Simulator::ERROR)?1:0);
}
```

If the "initializeEarly" call is replaced with "initialize" and the "initializeLate" call removed, this will be as broken as your example.

closed 2021-06-21 by tvrusso

pepijndevos, 2021-06-22:
Thank you! I will try this later, but assume it will work. Also thanks for finding the current bug. I was kind of assuming I just needed to add the equivalent of ngspice's `.savecurrents`.