Xyce/Xyce #30: Verilog-A model simulation fails.

eskiyerli, 2021-06-22:
Hello,

I tried to simulate a simple verilog-a file in Xyce. It works until one of the inputs reaches zero and then fails. The idea is that the output should flip phase of the input when that input goes below zero. I attached the input files as well as the log file. Tried to a few options statement but not much success. 

* Xyce Simulation Deck for /home/eskiyerli/Documents/Glade/tsmc18/Simulations/design_tsmc18/tb_phaseDetector/tb_phaseDetector_schematic.cir
.PREPROCESS REPLACEGROUND TRUE
.OPTIONS OUTPUT PHASE_OUTPUT_RADIANS=true 
.OPTIONS OUTPUT PRINTFOOTER=false 
.OPTIONS DEVICE TEMP = 25 
.OPTIONS DEVICE TNOM = 25 
.OPTIONS DEVICE GMIN = 1e-12 
.OPTIONS DEVICE VOLTLIM = 0 
.OPTIONS DEVICE LAMBERTW = 0 
.OPTIONS TIMEINT METHOD = 8 
.OPTIONS TIMEINT NEWLTE = 3 
.OPTIONS TIMEINT ABSTOL = 1e-12 
.OPTIONS TIMEINT RELTOL = 1e-5 
.OPTIONS PARSER MODEL_BINNING = 1
.INC "/home/eskiyerli/Documents/Glade/tsmc18/Simulations/design_tsmc18/tb_phaseDetector/tb_phaseDetector_schematic.cir"
.INC "/home/eskiyerli/OneDrive/Documents/Glade/tsmc18/Simulations/design_tsmc18/tb_phaseDetector/includeFile.inc"
.MODEL phaseDetectorModel phaseDetector gain = 1 
*.HDL /home/eskiyerli/OneDrive/Documents/Glade/tsmc18/behavioural/phaseDetector.va 
.OP
.TRAN 2p 5.5n 0 
.PRINT TRAN V(*) 
.END
-----
*******************************************************************************
* CDL netlist
*
* Library : design_tsmc18
* Top Cell Name: tb_phaseDetector
* View Name: schematic
* Netlist created: 22.Jun.2021 14:30:28
*******************************************************************************

*.SCALE METER
*.GLOBAL gnd!

YphaseDetector I0 lo rf if phaseDetectorModel  gain=0.5
RI4 if gnd! r=1
VI1 rf gnd! dc 0 ac 1 0 sin(1.1 1 1G 0 0)
VI2 lo gnd! dc 0 ac 0 0 sin(0 1 100Meg 0 0)
.END

----------------------------- 
.OPTIONS NONLIN-TRAN MAXSTEP=50
.OPTIONS TIMEINT NLNEARCONV=1
-------------------------------

`include "disciplines.vams"
`include "constants.vams"

// Phase Detector
module phaseDetector(lo, rf, if_);
inout lo, rf, if_;
electrical lo, rf, if_;
parameter real gain=1 (* type="instance" xyceAlsoModel="yes" *);
analog function real chopper;
input sw, in;
real sw, in;
//chopper = sw > 0 ? in : -in;
begin
if (sw > 0)
	chopper = in;
else 
    chopper = 0.1*in;
end
endfunction // chopper
analog V(if_) <+ gain*chopper(V(lo),V(rf));
[... 464 lines cut ...]
*****
***** End of Xyce(TM) Simulation
*****

Timing summary of 1 processor
                 Stats                   Count       CPU Time              Wall Time
---------------------------------------- ----- --------------------- ---------------------
Xyce                                         1        0.027 (100.0%)        0.068 (100.0%)
  Analysis                                   1        0.023 (85.31%)        0.063 (93.25%)
    Transient                                1        0.023 (84.96%)        0.063 (93.11%)
      Nonlinear Solve                      435        0.015 (55.63%)        0.040 (59.78%)
        Residual                           872        0.005 (16.86%)        0.013 (18.59%)
        Jacobian                           437        0.002 ( 7.29%)        0.005 ( 7.98%)
        Linear Solve                       437        0.004 (14.25%)        0.011 (15.54%)
      Successful DCOP Steps                  1        0.000 ( 0.00%)        0.001 ( 1.68%)
      Successful Step                      386        0.004 (14.02%)        0.011 (16.66%)
      Failed Steps                          48        0.000 ( 0.15%)        0.000 ( 0.67%)
  Netlist Import                             1        0.002 ( 7.75%)        0.002 ( 3.15%)
    Parse Context                            1        0.001 ( 2.54%)        0.001 ( 1.01%)
    Distribute Devices                       1        0.001 ( 1.96%)        0.001 ( 0.78%)
    Verify Devices                           1        0.000 ( 0.02%)        0.000 (<0.01%)
    Instantiate                              1        0.000 ( 0.42%)        0.000 ( 0.18%)
  Late Initialization                        1        0.001 ( 3.22%)        0.001 ( 1.27%)
    Global Indices                           1        0.000 ( 0.14%)        0.000 ( 0.06%)
  Setup Matrix Structure                     1        0.000 ( 0.55%)        0.000 ( 0.22%)

eskiyerli, 2021-06-28:
It seems that no response is coming. Will close the issue.

closed 2021-06-28 by eskiyerli

tvrusso, 2021-06-28:
Murat:  I apologize for not getting to your query, but I have been out of commission for a few weeks.

Your problem looks very much like the kind of thing that results when abrupt transitions are present in the simulation, which is often the case when behavioral models switch expressions based on a single "if".

Since your phase detector verilog is very simple, you might be better served by replacing it with a simple "B" source in "voltage source" mode.  For example, instead of 
```
YphaseDetector I0 lo rf if phaseDetectorModel gain=0.5
```
you could do:
```
Bphasedetector_l0 if gnd! V={0.5*((V(lo)>0)?V(rf):-V(rf))}
```

This should do exactly the same thing as your original Verilog without the need for compiling a plug-in, and should behave exactly as badly.   But what it *does* buy you is the possibility of using the "smoothbsrc" option that inserts a tiny RC network with a 1ms time constant in parallel with the source that smooths the transitions and generally helps improve convergence of such abrupt transitions.

If you really want to keep using the verilog-A version of this phase detector, you might try adjusting the conditional so it makes a smoother transition than simply changing abruptly when the control voltage changes sign, or by doing what the B source does with "smoothbsrc" enabled, inserting a small RC delay circuit in parallel.

I'm afraid I can't be much more help than that.  Timestep-too-small errors have a wide variety of causes, but abrupt transitions are the most common (the same problem tends to show up in digital circuits that use the low-level (1-6) MOSFETs with model cards that define no capacitances and simply use the default capacitances of zero).

tvrusso, 2021-06-28:
Just a quick correction:  the default RC constant used by "smoothbsrc" is 1ns, not 1ms.  It is effectively a 1u ohm resistor in parallel with a 1mF capacitor, both in parallel with the voltage source.  This tiny time constant is often enough to get past nasty timestep-too-small issues like this.

You could either do this by substituting an equivalent B source for your Verilog phase detector, or adding the same sort of parallel RC to your Verilog.

eskiyerli, 2021-07-05:
Hi Tom,

Thanks for the feedback. I want to stay with Verilog-A as I created a small GUI to convert a Verilog-A code to a Glade symbol. Once a symbol is created it can be used without further effort in a schematic. Revolution EDA GUI will take care of keeping track of models, model or instance parameters. I was hoping to create a simple PLL to show the concept. I guessed that it is due to abrupt switching of in the if statement. However, the output is actually very close to 0 when it cannot handle the time-steps. Will try to incorporate your idea.

Thanks,

tvrusso, 2021-07-06:
At the very least, making the B source substitution I suggested would make it a lot easier to experiment with whether the convergence issue (and timestep-too-small errors always amount to convergence problems) will go away with the simple addition of an RC delay circuit in parallel with the source.  It would just be a matter of trying it with and without the "smoothbsrc" option.

If it *doesn't* help to use the B source with "smoothbsrc" then you know the problem isn't the discontinuity in the model, and that adding it to your Verilog-A wouldn't help, either.  And if it *does* help, then you can modify your verilog-a to have the additional delay circuitry included.