Xyce/Xyce #33: Using step with subcircuit parameters

eskiyerli, 2021-07-30:
I wonder if it is possible to use .STEP statements with subcircuit parameters. For example, the user guide has this example for a subcircuit definition:
.SUBCKT SUB5 1 2 PARAMS: A1=5 FSIN=1
It is called with 
X2 e f SUB5 PARAMS: A1=10
What if I were to step A1 using another parameter? For example,

.PARAM aparam = 5
.STEP aparam 5 10 1
X2 e f SUB5 PARAMS: A1={aparam}

I am now trying to sweep gate widths of a 1p8v nmos device in skywater PDK. If there is no stepping, it works, but if I were to step gate width, it complains about source and drain conductances.

eskiyerli, 2021-07-30:
I got it working.

eskiyerli, 2021-08-03:
Reopening the issue as the issue seems to be quite complex. I updated the example in the user guide for parameters to approximate the models in the skywater PDK as shown below:
*subckt parameter sweep example
.options parser scale=1.0e-6
Vin in 0 dc 1
xm1 out in  0 0 xnmos PARAMS: l=0.11 w={wm1} m=2
RL vdd out 100
Vdd vdd 0 1.8

.PARAM pres=50
.PARAM wm1=5
.STEP wm1 5 20 5
.DC Vin 0 1 0.05
.PRINT DC V(*) I(*)

.SUBCKT subres in out PARAMS: res=1
R1 in out {res}
.ENDS

.SUBCKT xnmos d g s b PARAMS: w=1 l=1 m=1
mn1 d g s b nch w = {w} l = {l} m={m}
* model binning
.model nch.0 nmos(level=14
+ lmin = 0.1u lmax = 20u
+ wmin = 0.1u wmax = 10u)
.model nch.1 nmos(level=14
+ lmin = 0.1u lmax = 20u
+ wmin = 10u wmax = 100u)
.ENDS

.END

If .STEP is not used, the RLOAD current is at VIN=1.0 and VDD=1.8V is around 5.35mA.

However if the .STEP statement is activated,  the RLOAD current is constant at -2.34107358e-07 (0.234uA).

What could be going wrong?

Karlsefni2012, 2021-08-03:
I am looking at this now.  I'll see what I can figure out.

Karlsefni2012, 2021-08-03:
I was able to reproduce your error using Xyce version 7.3.  However, the error is not present my build of the Xyce development branch.  About a month ago, I fixed a few issues with using .option parser scale with dependent parameters, so possibly that bugfix also fixed this problem.  That was an issue that was exposed by people who were attempting to use Xyce with the Skywater 130 PDK.

eskiyerli, 2021-08-03:
Thanks. As you guessed, I encountered this problem when trying to port Skywater models into Xyce. Does it mean that I need to recompile Xyce from GitHub? Are there any other problems others might have faced already?

Karlsefni2012, 2021-08-03:
@eskiyerli , if you want to get this fix quickly (rather than wait for Xyce 7.4), you'll have to compile Xyce from github.

There were a handful of other issues encountered with Xyce and Skywater, but many of them could be addressed via scripting.  I don't recall all of them, but  I think most of them have been taken care of by now.  Some got fixed before the Xyce 7.3 release, and some came after.    Another issue, which I think got fixed for Xyce 7.3 was the use of relative paths for include files and libraries.
 
If you have access to the Skywater slack channel, you can see a lot of discussion about Xyce and Skywater there.   Some of the issues that emerged on the slack channel were then documented in this github issue by @mguthaus under @RTimothyEdwards   open-pdk repository:  
https://github.com/RTimothyEdwards/open_pdks/issues/124

There is also an older issue on github, under skywater-pdk about Xyce, which you can see here:  
https://github.com/google/skywater-pdk/issues/186.

eskiyerli, 2021-08-03:
Thanks. I actually read that thread and indeed joined slack #xyce channel. This is how I was able to simulate the device initially. I had compiled Xyce from GitHub. I think I could do it again.

closed 2021-08-03 by eskiyerli

eskiyerli, 2021-08-05:
Can confirm that the issue is fixed when using the latest fit clone from Xyce repo.

Karlsefni2012, 2021-08-05:
Great, glad to hear it!