Xyce/Xyce #39: Xyce/ADMS: Conditional operator cannot assign to integer variable when probe-dependant condition used

ngwood, 2021-09-07:
Tested with Xyce v7.3.0 and ADMS v2.3.7.

Minimal working example: [demo.va.txt](https://github.com/Xyce/Xyce/files/7123282/demo.va.txt)

Summary of errors reported by ```buildxyceplugin -d -o demo demo.va.txt```.
```
N_DEV_ADMSdemo.C: In member function 'virtual bool Xyce::Device::ADMSdemo::Instance::updateIntermediateVars()':
N_DEV_ADMSdemo.C:602:1: error: 'd_var_dV_b_GND' was not declared in this scope
  602 | d_var_dV_b_GND = (((probeVars[admsProbeID_V_a_GND])>=(probeVars[admsProbeID_V_b_GND]))?0.0:0.0);
      | ^~~~~~~~~~~~~~
N_DEV_ADMSdemo.C:603:1: error: 'd_var_dV_a_GND' was not declared in this scope
  603 | d_var_dV_a_GND = (((probeVars[admsProbeID_V_a_GND])>=(probeVars[admsProbeID_V_b_GND]))?0.0:0.0);
      | ^~~~~~~~~~~~~~
```

I believe that the problem is not the missing declarations, rather the attempt to calculate the derivatives.

Replacing ```integer``` with ```real``` works in the sense that the C++ code compiles, but then the variable no longer behaves as an integer. Replacing the conditional operator statement with an equivalent ```if-else``` statement produces correct results.

ngwood, 2021-09-07:
I should also point out that Xyce/ADMS uses C++ implicit conversion from ```double``` to ```int``` which performs truncation of the fraction part. This is inconsistent with the LRM-2.4 (4.2.1.1 Real to integer conversion) which states the following. "Real numbers are converted to integers by rounding the real number to the nearest integer, rather than by truncating it. Implicit conversion takes place when a real number is assigned to an integer. If the fractional part of the real number is exactly 0.5, it shall be rounded away from zero."

Verilog-A (var should be assigned 2):
```
integer var;

var = 1.5;
```

C++ (var actually assigned 1):
```
int var=0.0;
//End of Block-local variables
var = 1.5;
```

tvrusso, 2021-09-08:
Yes, you are correct, the error is most likely that when we emit the derivative of an expression that includes integer variables, we're using variables for the derivative of the integer variables that appear on the RHS.  We should probably be assuming that those derivatives are zero instead.

tvrusso, 2023-01-09:
FWIW, this issue has been fixed internally and will be in the next batch of code pushed to Github (which could be as early as tomorrow, but may be three or more weeks from now).  Unfortunately I neglected to mention this github issue in the commit messages and so update and closure of this issue won't happen automatically.   You'll need to watch for a batch of commits dated around January 4th and all referencing an issue number 106 (which is the issue number in our internal issue tracker associated with this Github issue).

tvrusso, 2023-02-01:
This was fixed in commit e33759d, just pushed to github yesterday.  There are several other commits in Xyce and Xyce_Regression related to documenting and testing the fixes, these may be recognized by looking for "Xyce/backlog/project-backlog\#106" in their commit messages (that's the issue number in the internal issue tracker associated with this github issue).

If you have other issues related to handling of integer variables, please open a new ticket and someone on the team may be able to address them.

closed 2023-02-01 by tvrusso