Xyce/Xyce #41: Xyce/ADMS: Missing standard atan2 function

ngwood, 2021-09-07:
The Accellera standard lists ```atan2``` as a standard function. This is currently not supported by Xyce/ADMS though.

tvrusso, 2021-09-08:
This is something that probably isn't too difficult to add.

tvrusso, 2021-09-27:
Neal:  Do you actually have a model that uses atan2 that I can use to verify correctness of any fix I make to this, or is this just a general request for more compliance with the standard?

tvrusso, 2021-09-27:
This has been fixed on our development branch, which will be synced to github sometime next week.

If you would like to try out the fix, you can apply the following patch to your github clone (remember to "git reset --hard" before pulling the updated code next week, though).

[0001-Add-ATAN2-support-to-Xyce-ADMS.txt](https://github.com/Xyce/Xyce/files/7238473/0001-Add-ATAN2-support-to-Xyce-ADMS.txt)

I would appreciate very much if you could test it out on your own model, because I have only tested it out using toy models that don't actually *run* (I just looked at the code generated).  I have tested use in both normal context and inside analog functions, and they both appear to be correct.

tvrusso, 2021-09-27:
That patch has an error that only impacts second derivatives (which will only be used if you do a ddx() of something, and then use that ddx thing downstream in contributions --- don't do that).

I've fixed that error, too, and it'll be in next week's github sync.

ngwood, 2021-09-29:
Hi Tom.

My specific use case is to calculate the argument of a complex number; this cannot be done simply with atan(y / x) due to its limited input range and the inability to individually determine the sign of y and x.

Specifically, I use it inside of an analog function that calculates the complex cube root.

/*
DESCRIPTION: Compute the cube root of a complex number using De
Moivre's Theorem.
RETURN:
cmplx_cbrt Zero
INPUT:
a Real part of complex number
b Real part of imaginary part of complex number
OUTPUT:
c Real part of complex number
d Real part of imaginary part of complex number
LOCAL:
tmp1..2 Temporary variables
*/
analog function real cmplx_cbrt;
input a, b;
output c, d;
real a, b, c, d, tmp1, tmp2;
begin
tmp1 = cbrt(hypot(a, b));
tmp2 = atan2(b, a) / 3.0;
c = tmp1 * cos(tmp2);
d = tmp1 * sin(tmp2);
end
endfunction

This is used to handle an edge case of another analog function, which expects it's first input argument to be probe-dependent.

/*
DESCRIPTION: Compute the n-channel junction field-effect transistor
channel saturation voltage for a uniform rectangular
channel with unilateral step junction in the framework
of the gradual channel approximation and depletion
approximation.
RETURN:
njf_sg_sat_volt Saturation drain-source voltage / V
INPUT:
phigs Gate-source potential / V
phipo Pinch-off potential / V
xi Transport parameter / V⁻¹
LOCAL:
tmp1..6 Temporary variables
dummy Placeholder variable
*/
analog function real njf_sg_sat_volt;
input phigs, phipo, xi;
real phigs, phipo, xi, tmp1, tmp2, tmp3, tmp4, tmp5, tmp6, dummy;
begin
if (xi == 0.0)
njf_sg_sat_volt = phipo - phigs;
else begin
tmp1 = 1.0 / xi;
tmp2 = tmp1 - phigs;
tmp3 = 1.5 * sqrt(phipo) * tmp1 - cbsqrt(phigs);
tmp4 = cb(tmp2) + sq(tmp3);
if (tmp4 >= 0.0) begin
tmp5 = cbrt(tmp3 + sqrt(tmp4));
njf_sg_sat_volt = sq(tmp2 / tmp5 - tmp5) - phigs;
end
else begin
dummy = cmplx_cbrt(tmp3, sqrt(-tmp4), tmp5, tmp6);
njf_sg_sat_volt =
sq(tmp2 * tmp5 / (sq(tmp5) + sq(tmp6)) - tmp5) - phigs;
end
end
end
endfunction

Working through the function you'll find that the input arguments to cmplx_cbrt would both be probe-dependent in my use case.

If you untar the attachment I am sending you and run the standard buildxyceplugin script with just the resistor.va file it should hopefully work with your version of Xyce/ADMS (that has support for atan2).

If you need me to strip out any non-essential parts of the model in order to make it easier for you to debug I can do so; however, I would not be able to do this until Friday, unfortunately.

Regards, Neal.

Sent with [ProtonMail](https://protonmail.com/) Secure Email.

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

ngwood, 2021-09-29:
Hi Tom.

My specific use case is to calculate the argument of a complex number; this cannot be done simply with `atan(y / x)` due to its limited input range  and the inability to individually determine the sign of `y` and `x`.

Specifically, I use it inside of an analog function that calculates the complex cube root.

```
/*
    DESCRIPTION:        Compute the cube root of a complex number using De
                        Moivre's Theorem.
    RETURN:
        cmplx_cbrt      Zero
    INPUT:
        a               Real part of complex number
        b               Real part of imaginary part of complex number
    OUTPUT:
        c               Real part of complex number
        d               Real part of imaginary part of complex number
    LOCAL:
        tmp1..2         Temporary variables
*/
analog function real cmplx_cbrt;
    input a, b;
    output c, d;
    real a, b, c, d, tmp1, tmp2;
    begin
        tmp1 = cbrt(hypot(a, b));
        tmp2 = atan2(b, a) / 3.0;
        c = tmp1 * cos(tmp2);
        d = tmp1 * sin(tmp2);
    end
endfunction
```

This is used to handle an edge case of another analog function, which expects its first input argument to be probe-dependent.

```
/*
    DESCRIPTION:        Compute the n-channel junction field-effect transistor
                        channel saturation voltage for a uniform rectangular
                        channel with unilateral step junction in the framework
                        of the gradual channel approximation and depletion
                        approximation.
    RETURN:
        njf_sg_sat_volt Saturation drain-source voltage / V
    INPUT:
        phigs           Gate-source potential / V
        phipo           Pinch-off potential / V
        xi              Transport parameter / V⁻¹
    LOCAL:
        tmp1..6         Temporary variables
        dummy           Placeholder variable
*/
analog function real njf_sg_sat_volt;
    input phigs, phipo, xi;
    real phigs, phipo, xi, tmp1, tmp2, tmp3, tmp4, tmp5, tmp6, dummy;
    begin
        if (xi == 0.0)
            njf_sg_sat_volt = phipo - phigs;
        else begin
            tmp1 = 1.0 / xi;
            tmp2 = tmp1 - phigs;
            tmp3 = 1.5 * sqrt(phipo) * tmp1 - cbsqrt(phigs);
            tmp4 = cb(tmp2) + sq(tmp3);
            if (tmp4 >= 0.0) begin
                tmp5 = cbrt(tmp3 + sqrt(tmp4));
                njf_sg_sat_volt = sq(tmp2 / tmp5 - tmp5) - phigs;
            end
            else begin
                dummy = cmplx_cbrt(tmp3, sqrt(-tmp4), tmp5, tmp6);
                njf_sg_sat_volt =
                    sq(tmp2 * tmp5 / (sq(tmp5) + sq(tmp6)) - tmp5) - phigs;
            end
        end
    end
endfunction
```

Working through the function you'll find that the input arguments to `cmplx_cbrt` would both be probe-dependent in my use case.

I will check everything work in my model and let you know how I get on.

tvrusso, 2021-09-29:
Great.  If you can try out the patch before the weekend it'd be cool, but if you have to wait until we push to github that'll work for me, too.

ngwood, 2021-09-30:
I ran into an error relating to unbalanced parentheses.
```
N_DEV_ADMSwhitej.C: In function 'double Xyce::Device::ADMSwhitej::AnalogFunctions::d_cmplx_cbrt(double, double, double, double, double, double, double&, double&)':
N_DEV_ADMSwhitej.C:300:30: error: expected ';' before ')' token
  300 | d_tmp2_d_a=(-b/(b*b+a*a))/3.0);
      |                              ^
      |                              ;
N_DEV_ADMSwhitej.C: In member function 'Xyce::Device::ADMSwhitej::AnalogFunctions::cmplx_cbrtEvaluator::returnType Xyce::Device::ADMSwhitej::AnalogFunctions::cmplx_cbrtEvaluator::evaluator_(double, double, Xyce::Device::ADMSwhitej::AnalogFunctions::cmplx_cbrtEvaluator::returnType&, Xyce::Device::ADMSwhitej::AnalogFunctions::cmplx_cbrtEvaluator::returnType&)':
N_DEV_ADMSwhitej.C:375:30: error: expected ';' before ')' token
  375 | d_tmp2_d_a=(-b/(b*b+a*a))/3.0);
      |                              ^
      |                              ;
```
I fixed the issue on line 797 of `xyceAnalogFunction_nosac.xml` and was able to get my models to compile.

Subsequently, I chose some parameters that forced one of my models to always operate in a way that required it to rely on the `cmplx_cbrt` analog function that contains the `atan2` call. I saw no convergence issues. I also compared the output characteristic from this test with the equivalent output when using an approximation to the `atan2` function that performs a smoothed interpolation between the various quadrants and they matched up exactly.

If there was an issue with the calculations and the derivatives I would expect to see convergence issues due to the introduction of a discontinuity in the current voltage characteristics. I made sure to sweep the voltages in such a way to make sure no such discontinuity was present.

I don't contribute variable output by the ddx operator, so that would not affect my tests.

tvrusso, 2021-09-30:
I'll take a look at the error on line 797 of xyceAnalogFunction_nosac.xml.

Thanks for checking.

tvrusso, 2021-09-30:
Fixed. Thank you.

This patch and the fixes to the fixes will be on github next week.

tvrusso, 2021-10-03:
Code is now published.  Enjoy.

closed 2021-10-03 by tvrusso