Xyce/Xyce #42: Xyce/ADMS: Xyce parser fails for plugins that have no parameters

ngwood, 2021-09-17:
The Xyce parser cannot handle a Xyce ``.model`` being used with a ``-plugin`` instance whose Verilog-A model does not define at least one parameter.

[foo.va.txt](https://github.com/Xyce/Xyce/files/7186843/foo.va.txt):
 ```
  1 `include "disciplines.vams"
  2 
  3 module foo(a, b);
  4     inout a, b;
  5     electrical a, b;
  6 
  7 //  parameter real bar = 1.0;
  8 
  9     analog
 10         I(a, b) <+ V(a, b);
 11 endmodule
```

[foo.cir.txt](https://github.com/Xyce/Xyce/files/7186856/foo.cir.txt):
```
  1 * xyce parser bug demo
  2 v1 1 0 dc 2.0
  3 .model baz foo
  4 yfoo bar 1 0 baz
  5 .op
  6 .end
```

The following invocation results in an error.
```
bash> buildxyceplugin -d -o foo foo.va.txt . 1> /dev/null
bash> xyce -plugin foo.so foo.cir.txt 

*****
***** Welcome to the Xyce(TM) Parallel Electronic Simulator
*****
***** This is version Xyce DEVELOPMENT-202109131838-g-d7d64f70-opensource
***** Date: Fri Sep 17 16:49:00 BST 2021

***** Executing netlist foo.cir.txt

***** Reading and parsing netlist...
Netlist warning: No print specified
Netlist error in file foo.cir.txt at or near line 4
 Unrecognized parameter BAZ for device YFOO!BAR
Netlist error in file foo.cir.txt at or near line 4
 Unrecognized fields for device YFOO!BAR
Simulation aborted due to error.  There are 0 MSG_FATAL errors and 2 MSG_ERROR
 errors
*** Xyce Abort ***
Simulation aborted due to error.  There are 0 MSG_FATAL errors and 2 MSG_ERROR
 errors

*** Xyce Abort ***
```

If I uncomment line number 7 of [foo.va.txt](https://github.com/Xyce/Xyce/files/7186843/foo.va.txt) then I do not get these errors. Also, if I remove the model name from line number 4 of [foo.cir.txt](https://github.com/Xyce/Xyce/files/7186856/foo.cir.txt) then I do not get this error either.

I'm not sure if this behaviour is intensional or not, but I see no mention of this in the documentation.

tvrusso, 2021-09-17:
A device type that does not have any model parameters will not allow you to associate an instance with a model card.  This is somewhat deliberate, and yeah, it is probably not explicitly documented.  We could probably toss something into the documentation about it.  But it is not a bug.

The reason here is that if the device does not define any model parameters, then Xyce/ADMS does not even generate code to associate a .model card type with the device, and therefore the top level parser knows that the device doesn't take one and rejects instance lines that provide one.

A device that defines itself as one of the standard device types (like R, M, D, Q...) automatically has at least one model parameter, level, and therefore must have an associated model card.

If you look at the ["toy" resistor in the Xyce source tree](https://github.com/Xyce/Xyce/tree/master/utils/ADMS/examples/toys/resistor.va) and generate its C++ code with ADMS (either by running buildxyceplugin with the -d option to preserve the C++ source, or just running Xyce/ADMS by hand:
```
  export ADMSDIR=/path/to/Xyce/ADMS/XML/files
  admsXml -x -e $ADMSDIR/adms.implicit.xml -e $ADMSDIR/xyceVersion_nosac.xml -e $ADMSDIR/xyceBasicTemplates_nosac.xml -e $ADMSDIR/xyceAnalogFunction_nosac.xml -e $ADMSDIR/xyceHeaderFile_nosac.xml -e $ADMSDIR/xyceImplementationFile_nosac.xml resistor.va
```
you will find that it produces a line of code in the .C file, down in the `registerDevice` function:
```
    .registerModelType("r", 6);
```

This is what associates the "r" model card of level "6" with this device model.  The same thing happens to any device with no module attributes, but with model parameters --- since there are model parameters, there MUST be an associated model type, and the default type is the same name as the module.  Uncomment your model parameter in foo.va and you'll get a registerModelType for model type foo level 1.

Comment out or remove the module attributes in that file:
```
   `attr(xyceModelGroup="Resistor" xyceSpiceDeviceName="R" xyceLevelNumber="6") 
```

so that the module line is just:
```
module resistor (p,n);
```
and regenerate the code, you'll find that there is no longer a registerModelType line, and so the parser knows that this device cannot have a model on its instance line.

This is fairly generic behavior for Xyce --- device types that have no model parameters do not have associated model cards, and it is illegal to specify a model name.  In reality, it's not so much that it's illegal to specify one, but Xyce knows that there is not supposed to be one and does not recognize "extra" fields on the instance line that it isn't expecting.

These days there are very few devices in Xyce that *don't* take model cards, even basic SPICE devices like resistors (the original spice resistor had no model parameters, but we've extended it a lot and now there are such parameters).  So this doesn't come up much.  

You'd have had the same problem if you defined your foo.va file's parameter with `(* type="instance" *)` --- since there is no MODEL parameter in the device, there is no associated model card type, and the parser expects no model on the instance line.

closed 2021-09-17 by tvrusso

ngwood, 2021-09-17:
I can see the point of doing this actually. Funnily enough, I originally noticed this behaviour after adding ``(* type="instance" *)`` on the solitary parameter of an example module. It does mean that if you want to use just instance parameters you'd need to define at least one model parameter as well; that's what made me lean towards thinking it was a bug than a feature.

tvrusso, 2021-09-17:
You don't need to define a model parameter.  You simply leave the model card off of the instance line.

I should probably fix up the Xyce/ADMS documentation so that it is clear that the model on the instance line is only required if the module has model parameters (or is of a standard SPICE device type other than "Y").

ngwood, 2021-09-18:
What I mean to say is that you cannot set an instance parameter without having a model card on the instance line; i.e.,
```
ymodule_name instance_name 1 0 instance_parameter=1.0
```
wouldn't work. In order to add a model card to the instance line one would need to first make available a model parameter, even if it's never used. As you say, this is an unlikely scenario indeed.

tvrusso, 2021-09-18:
This is a false statement.  You can indeed do exactly that.

resistor_Y.va:
```
`include "disciplines.vams"
`include "constants.vams"
// Xyce's K is not the one in constants.vams!
`define X_K (1.3806226e23)

`define attr(txt) (*txt*)

module resistor (p,n);
   electrical p,n;
   inout p,n;
   
   parameter real R=1 from (0:inf) `attr(info="Resistance" type="instance");
   
   real           Ir;
   real           tdevK;
   real           conductance;
   
   analog
     begin
        tdevK=$temperature;
        conductance=1.0/R;
        Ir=V(p,n)*conductance;
        I(p,n) <+ Ir;
        I(p,n) <+ white_noise(4*`X_K*tdevK*conductance,"thermal");
     end
endmodule
```

resistor.cir:
```
Netlist to Test the Xyce Resistor Model
Yresistor R1 1 0 R=1K 
V1 1 0 5V
.DC V1 0 5V 1V
.PRINT DC V(1) I(V1)
.END
```

Compile the verilog-a with buildxyceplugin and run the netlist with the plugin enabled, and it will work just fine and produce an I-V curve for a 1KOhm resistor.

ngwood, 2021-09-18:
You are entirely correct, sorry.

tvrusso, 2021-09-20:
This deficiency in the Xyce/ADMS users guide has been addressed and the updated page will be published when Xyce 7.4 is released (most likely sometime in November).