Xyce/Xyce #43: Xyce/ADMS: @(initial_model) variables can be corrupted by instance parameters

ngwood, 2021-09-18:
Currently, it it possible for an instance parameter on one instance line to alter the model variables used by all other instance lines.
```
`include "disciplines.vams"

module foo(a, b);
    inout a, b;
    electrical a, b;

    parameter integer level = 1;
    parameter real bar = 1.0 (* type = "instance" *);

    real baz;

    analog begin
	@(initial_model) begin
	    baz = 1.0;
	end

	@(initial_instance) begin
	    baz = bar;
	end

	I(a, b) <+ V(a, b) / baz;
    end
endmodule
```
```
* inital model bug demo (a)
v1 1 0 dc 0.0
.model foo foo
yfoo r1 1 2 foo ; Behaves as though bar=2.0
yfoo r2 2 0 foo bar=2.0
.dc lin v1 0.0 1.0 0.1
.print dc v(1) v(2) {-i(v1)}
.end
```
One would expect the `v(2)` to equal two thirds of `v(1)`, not one half of it.

The same thing happens if more than one instance line sets an instance parameter.
```
* inital model bug demo (b)
v1 1 0 dc 0.0
.model foo foo
yfoo r1 1 2 foo bar=2.0 ; Behaves as though bar=4.0
yfoo r2 2 0 foo bar=2.0 ; Behaves as though bar=4.0
.dc lin v1 0.0 1.0 0.1
.print dc v(1) v(2) {-i(v1)}
.end
```

tvrusso, 2021-09-18:
Oooh.  Nasty.

Yes, clearly there is a bug in handling model-scoped variables.  I am not entirely sure how to fix it, since at the moment a variable used in the initial_model block *must* be a member of the model class, and it is perfectly reasonable to *use* a model-scoped variable (other than by assigning to it) in the instance scope.  It is illegal to access an instance-scoped variable at model scope (as there can be a many-to-one relationship between instances and models).  So in order to get done what your example intends, there would have to be two "baz" variables, a model-scoped and an instance-scoped version, with the instance-scoped version being initialized to the model scoped version, but overridable by code at instance scope.  Only the instance-scoped version would be used in the main body of the model (which is always instance-scoped).  

To get *that* done, we'd have to detect that a variable that is used at model scope *and* assigned to at instance scope, and trigger this more complex declaration/usage issue.  Either that, or have all model variables mirrored at instance scope and initialized to the value in the model, and all instance-scope access of that variable use the instance version.  These are just the ideas that spring to mind initially, but there are probably other ways to handle it.  It will not be particularly easy.

In the meantime,  the workaround until the bug is fixed would be to define a variable that is intentionally assigned to only at local scope, so it becomes an instance class variable.

```
`include "disciplines.vams"

module foo(a, b);
    inout a, b;
    electrical a, b;

    parameter integer level = 1;
    parameter real bar = 1.0 (* type = "instance" *);

    real bazM, basI;

    analog begin
	@(initial_model) begin
	    bazM = 1.0;
	end

	@(initial_instance) begin
            basI = basM;
	    bazI = bar;
	end

	I(a, b) <+ V(a, b) / bazI;
    end
endmodule
```

Obviously in this model the existence of barM is pointless, but I imagine you have less useless concrete reasons for wanting to do this.

tvrusso, 2021-09-18:
This issue has been opened in our internal issue tracker as issue 287.