Xyce/Xyce #49: Xyce/ADMS: analog function ternary operator bug

ngwood, 2021-12-27:
The following analog function works correctly.
```
analog function real cbrt;
    input x;
    real x, sng;
    begin
        if (x >= 0.0)
            sng = 1.0;
        else
            sng = -1.0;
        cbrt = sng * pow(sng * x, 0.33333333333333333333);
    end
endfunction
```
The equivalent analog function results in ambiguous C++ code.
```
analog function real cbrt;
    input x;
    real x, sng;
    begin
        sng = x >= 0.0 ? 1.0 : -1.0;
        cbrt = sng * pow(sng * x, 0.33333333333333333333);
    end
endfunction
```
```
Compiling C++ for whitej...
libtool: compile:  /opt/local/bin/g++ -std=gnu++11 -I. -I/usr/local/xyce/devel/include -DXyce_NOX_LOCA_SUPPORT -I/usr/local/xyce/devel/include -I/opt/local/include -O3 -std=c++11 -c N_DEV_ADMSwhitej.C -fPIC -DPIC  -fno-common -DPIC -o .libs/N_DEV_ADMSwhitej.o
In file included from N_DEV_ADMSwhitej.C:44:
./N_DEV_ADMSwhitej.h: In instantiation of 'ScalarT Xyce::Device::ADMSwhitej::AnalogFunctions::cbrt(ScalarT) [with ScalarT = double]':
N_DEV_ADMSwhitej.C:291:24:   required from here
./N_DEV_ADMSwhitej.h:885:22: error: call of overloaded 'adms_ternary_op(bool, double, double)' is ambiguous
  885 | sng = adms_ternary_op((x>=0.0),1.0,(-1.0));
      |       ~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~
In file included from N_DEV_ADMSwhitej.C:44:
./N_DEV_ADMSwhitej.h:142:12: note: candidate: 'Tin Xyce::Device::ADMSwhitej::adms_ternary_op(bool, const Tin&, const Tin&) [with Tin = double]'
  142 | static Tin adms_ternary_op(const bool cond, const Tin &ifTrue, const Tin &ifFalse)
      |            ^~~~~~~~~~~~~~~
./N_DEV_ADMSwhitej.h:151:12: note: candidate: 'Tin Xyce::Device::ADMSwhitej::adms_ternary_op(bool, const Tin&, const double&) [with Tin = double]'
  151 | static Tin adms_ternary_op(const bool cond, const Tin &ifTrue, const double &ifFalse)
      |            ^~~~~~~~~~~~~~~
./N_DEV_ADMSwhitej.h:160:12: note: candidate: 'Tin Xyce::Device::ADMSwhitej::adms_ternary_op(bool, const double&, const Tin&) [with Tin = double]'
  160 | static Tin adms_ternary_op(const bool cond, const double &ifTrue, const Tin &ifFalse)
      |            ^~~~~~~~~~~~~~~

```

tvrusso, 2021-12-28:
Thank you for this report.

The multiple templating of adms_ternary_op and the template specializations are a relic of the old Sacado implementation, where the ternary op was a huge mess and very much more complicated to handle than it is now.

The Xyce team is on holiday until next week, and I'm actually taking it.  I believe this is simply a matter of removing the multiple templates output by xyceHeaderFile_nosac.xml, replacing:

```
// Figure out how to template this shiznit!
//-----------------------------------------------------------------------------

template&lt;typename Tin&gt;
static Tin adms_ternary_op(const bool cond, const Tin &amp;ifTrue, const Tin &amp;ifFalse)
{
if (cond)
return ifTrue;
else
return ifFalse;
}

template&lt;typename Tin&gt;
static Tin adms_ternary_op(const bool cond, const Tin &amp;ifTrue, const double &amp;ifFalse)
{
if (cond)
return ifTrue;
else
return Tin(ifFalse);
}

template&lt;typename Tin&gt;
static Tin adms_ternary_op(const bool cond, const double &amp;ifTrue, const Tin &amp;ifFalse)
{
if (cond)
return Tin(ifTrue);
else
return ifFalse;
}
```

with

```// Figure out how to template this shiznit!
//-----------------------------------------------------------------------------

template&lt;typename Tin&gt;
static Tin adms_ternary_op(const bool cond, const Tin &amp;ifTrue, const Tin &amp;ifFalse)
{
if (cond)
return ifTrue;
else
return ifFalse;
}
```

(that is, remove the two specializations and leave only the full templated version).  That should do the trick.  Now that there is no use of Sacado at all, this could probably be even further simplified to a straight inline function:

```
inline double adms_ternary_op(bool cond, double ifTrue, double ifFalse)
{
if (cond)
return ifTrue;
else
return ifFalse;
}
```
in the same file.

I will not be able to test this out until at least mid January, but if you care to try it it might fix the entire issue.

I think maybe it might not even be necessary to use "adms_ternary_op" anymore at all, since we should be able simply to emit the C++ ternary operator of the same form.  That is, in fact, what is already being done everywhere *except* in analog functions.  The analog function code simply wasn't completely brought up to date after Sacado was removed.

ngwood, 2021-12-28:
Thank you.

Both removing the legacy templates and replacing the templates with an inline function in `xyceHeaderFile_nosac.xml` worked fine. Also, removing the templates and then removing

```
<admst:when test="[$globalCurrentScope='AF']">
  <admst:variable name="expression" select="adms_ternary_op($a1,$a2,$a3)"/>
</admst:when>
```

from `xyceBasicTemplates_nosac.xml` worked, too.

tvrusso, 2021-12-28:
Great.  When the holidays are over I'll see if I can clean that up.  It never got fixed when the Sacado stuff was replaced, and now it is just a bit of vestigial ugliness.

tvrusso, 2022-01-04:
Fixed on our end.  Will appear here on github when we do our next repo sync (probably next week).

I should note for anyone reading this report that the specific bug is only present when calling an analog function that uses ternary operators from *another* analog function.

Analog functions that use ternary operators but are only called from the module itself would not have tripped the bug.

ngwood, 2022-01-04:
Thank you, again.

tvrusso, 2022-01-04:
Please leave open until the code is pushed.  Comments in the commits that fix it will close this issue automatically.

closed 2022-01-09 by tvrusso