Xyce/Xyce #54: Rebuilding xyce with shareable options

ratul619, 2022-01-21:
Hi , I am trying to reinstall xyce with enable-shared option (for veriloga adms plugin)
../configure \
CXXFLAGS="-O3" \
ARCHDIR="$HOME/XyceLibs/Serial" \
CPPFLAGS="-I/usr/include/suitesparse" \
--enable-stokhos \
--enable-amesos2 \
--enable-shared  --enable-xyce-shareable \
But make is stopping with below errors :
ake[4]: Leaving directory '/home/ratul619/installs/Xyce-7.4/my_build/src/test'
make[3]: Leaving directory '/home/ratul619/installs/Xyce-7.4/my_build/src/test'
make[3]: Entering directory '/home/ratul619/installs/Xyce-7.4/my_build/src'
  CXX      Xyce.o
  CCLD     libxyce.la
ar: `u' modifier ignored since `D' is the default (see `U')
  CXXLD    Xyce
/usr/bin/ld: ./.libs/libxyce.so: undefined reference to `Xyce::Device::Units::unitTable'
/usr/bin/ld: ./.libs/libxyce.so: undefined reference to `Xyce::IO::MMIO::mm_read_mtx_crd_size(_IO_FILE*, int*, int*, int*)'
/usr/bin/ld: ./.libs/libxyce.so: undefined reference to `Xyce::IO::MMIO::mm_read_mtx_array_size(_IO_FILE*, int*, int*)'
/usr/bin/ld: ./.libs/libxyce.so: undefined reference to `Xyce::Device::Units::unitTableSize'
/usr/bin/ld: ./.libs/libxyce.so: undefined reference to `Xyce::IO::MMIO::mm_read_mtx_crd_data(_IO_FILE*, int, int, int, int*, int*, double*, char*)'
/usr/bin/ld: ./.libs/libxyce.so: undefined reference to `Xyce::IO::MMIO::mm_write_banner(_IO_FILE*, char*)'
/usr/bin/ld: ./.libs/libxyce.so: undefined reference to `Xyce::Device::Units::descriptionTable'
/usr/bin/ld: ./.libs/libxyce.so: undefined reference to `Xyce::Device::Units::descriptionTableSize'
/usr/bin/ld: ./.libs/libxyce.so: undefined reference to `Xyce::IO::MMIO::mm_read_banner(_IO_FILE*, char (*) [4])'
/usr/bin/ld: ./.libs/libxyce.so: undefined reference to `Xyce::IO::MMIO::mm_write_mtx_array_size(_IO_FILE*, int, int)'
collect2: error: ld returned 1 exit status
make[3]: *** [Makefile:783: Xyce] Error 1
make[3]: Leaving directory '/home/ratul619/installs/Xyce-7.4/my_build/src'
make[2]: *** [Makefile:875: all-recursive] Error 1
make[2]: Leaving directory '/home/ratul619/installs/Xyce-7.4/my_build/src'
make[1]: *** [Makefile:645: all] Error 2
make[1]: Leaving directory '/home/ratul619/installs/Xyce-7.4/my_build/src'
make: *** [Makefile:526: all-recursive] Error 1

tvrusso, 2022-01-21:
There is not enough information to determine why your compilation is failing.  The options you are using are basically the same options that many of us use, so we cannot guess at why it's failing for you.  The undefined references you're reporting are all for functions that are provided in our own source code, and with just the information you've provided, there is no way to tell why they're not being found.

What operating system are you compiling on?

Can you please capture the full output of "make" (starting from a completely fresh build directory) into a file (e.g. "make > make.txt 2>&1") and attach that file directly to this issue (don't cut and paste the text, attach the whole file)?  That will tell us perhaps why the files that contain the functions being complained about aren't being properly inserted into the library as they should be.

ratul619, 2022-01-22:
Hi , 

Thanks . Seems rebuilding in new my_build directory  - installation is done.

closed 2022-01-22 by ratul619