Xyce/Xyce #58: Missing contributions in OpenModels

vpinon, 2022-03-06:
Hello dear Xyce developers,

When I tried to converted my PDK Spectre/HSpice models with XDM, I noted that several parameters were dropped as they are not implemented in Xyce :
- for D: JSW, TCV, IKR, TRS, CTA, TPB, CJSW, PHP, MJSW, CTP, TPHP, JTUN, JTUNSW, NTUN, KEG, XTITUN, FCS, TREF, TLEV, TLEVC. I'm not interested in all of them, but mainly the *SW components ("SideWall"=perimetric scaling) that do have an impact at high frequencies.
- for R: KF, AF, EF. I know ignoring poly resistor's flicker in some of my circuits could be a mistake (non-negligible contribution).
- for Q: SUBS, TREF, TLEV, TLEVC, TBF1, TBF2, TBR1, TBR2, TIKF1, TIKR1, TIRB1, TNC1, TNE1, TNF1, TNF2, TRE1, TNR1, TRB1, TRC1, CTC, CTE, TVJC, TVJE ; I've not studied these, just kept them for the record (as long as it doesn't affect my bandgap circuit response?)
- also D & Q don't seem to support M (multiplicity) parameter?

I believe several of these parameters (or alias names) are handled in NGSpice, so the equations should be around. Note that for flicker, [this](https://github.com/KenKundert/flicker-noise) might be worth to read?
I'm ready to help with testing, comparing results with commercial simulators.
I don't want to load you with requests your group is not interested in, so if it could be better, I'm also OK to contribute such code.
But then, I know you can't accept code from external contributors: isn't there a way around?
I have no problem signing a CLA for such contributions (already did it before with other projects), leaving my copyright to you...

Best regards

vpinon, 2022-03-06:
I'm just realizing these model evolution might be achieved by a new verilog-a implementation of basic devices, but then would it have an impact on performance? How to tell XDM how to proceed with the translation?

Karlsefni2012, 2022-03-28:
Thanks for your suggestions.  We can take a look at these various issues.  At least for some of them, they should be easy  to do.

Regarding all the unrecognized model parameters, I can't promise anything at the moment, but if those parameters are easy to add we can (probably) do that.  We'll take a look at them.  Usually, if unrecognized params are associated with proprietary models there isn't much we can do about them.  But if they've been published in the open domain then that makes it feasible.  It sounds like some of these are.

Regarding the multiplicity factors (M=), they are actually already supported in the diode, as of Xyce 7.4.   You are correct that they are not in the BJT, however.  They should be, so I've opened an issue for this in our internal issue tracker.  It should be easy to do.

Regarding flicker noise, I am familiar with that paper, etc. I believe it wouldn't be that hard to add it to the Xyce resistor model.  I've opened an issue on this in our internal tracker as well.

Regarding Verilog-A devices, if you want to go that route.  The main way to run verilog-A models in Xyce is to use the ADMS model compiler.  XDM isn't involved in that process.  Currently, using ADMS with Xyce requires that you build Xyce from source.  We have instructions for doing so (building Xyce and also using ADMS) at our website:

https://xyce.sandia.gov/documentation-tutorials/building-guide/

https://xyce.sandia.gov/documentation-tutorials/xyce-adms-users-guide/

Performance-wise, ADMS-generated models in Xyce are competitive vs.  hand-written models.  So, there shouldn't be a  performance hit from using them.

Note, since you are talking about augmenting existing Xyce models.  Most of the models you refer to are old, and predate the standardization of Verilog-A as a modeling language.    So, there wouldn't be a straightforward way to modify existing Verilog-A code of (for example) the BJT in Xyce, since there isn't an existing Verilog-A BJT model in our repository.  It is a hand-coded model.  To modify that model, someone would have to hack the C++ code directly.  The alternative, which would avoid C++ hand-hackage, would be to create an entirely new model via Verilog-A.

tvrusso, 2023-09-03:
FWIW, a number of commits back in December introduced sidewall computations into Xyce's diode  (not one of which are those shown above, which as an artifact of how Xyce's internal issue tracker numbers mentioned in commit logs confuse github, and the fact that we forgot to mention this github issue in the commits that actually had something to do with it).

Those commits are:  28a88fd0, e2755f63, ed98213d, 189d1c6e, and 2c43fc77, and they're documented in commit c3cc8c90.

Only sidewall computations were added, not any of the tunneling stuff.  So now Xyce's diode supports the PJ instance parameter and the JSW, NS, CJSW, CJP, PHP, VJSW, MJSW, and FCS model parameters.  They were tested against ngspice's diode code when they were committed and agreed.

These updated diode computations were not in Xyce 7.6 and were only just released in Xyce 7.7 last week, but have been in the master branch since right before Christmas of last year if you were building from source out of github clones.

tvrusso, 2023-09-04:
I will note, however, that while Xyce 7.7 supports these new parameters in the diode and Xyce has had them since December, XDM 2.7 was not updated to reflect that change, and will still ignore those parameters if one tries to convert a netlist from Spectre or Hspice to Xyce.  The xyce.xml file in the XDM source tree still lacks those parameters 

Pinging @gjtempl , @Karlsefni2012, and @TBird2001  to get this on their radar.  There may be other ways that XDM's "xyce.xml" isn't up to date.