Xyce/Xyce #59: Add ability to parameterize transient end time

olde-fortran, 2022-04-04:
```
.param endtime = '34 * clk_period'
.tran 50p 'endtime'
```

This is really handy when I want to parameterize the clock period, but make sure that, say, 32 clock cycles are captured.  

ng-spice allows this.  Xyce gives me: `Netlist error: Cannot convert '{endtime}' to double for expression TSTOP`

Right now I either have to adjust the end time manually, OR if the measurement isn't affected I can plug in a large enough value to satisfy all the clk_period values that I'm testing, but the simulations take longer.  (I'm sweeping the clk_period by launching Xyce multiple times)

Karlsefni2012, 2022-04-05:
I think this is something we can add to Xyce.  I'll add it to our internal issue tracker.

Karlsefni2012, 2023-06-01:
I wanted to mention that this issue got fixed on the Xyce development branch a while ago.  I meant to give an update here, but forgot.     It was (mostly) fixed in this commit:

https://github.com/Xyce/Xyce/commit/3d521da0654a698cbf5b8b0450d70d9d6f5492a9

yakimetsd, 2025-05-12:
As this issue is still open, I would like to propose another approach to the end time definition: once all the defined measurements are completed, transient simulation may be optionally automatically stopped. I think it is called "autostop" in spectre and ngspice. What do you think of such a proposal?