Xyce/Xyce #61: What to do if an error is reported in the test with Xyce

Cahui1, 2022-05-27:
The test suite be run in between the "make" and "make install" steps.
What should I do if such a mistake occurs?
[image]

Karlsefni2012, 2022-05-27:
It is hard to say what might be wrong without more details.  Are you only seeing errors for the `EMBEDDED` and `EMBEDDED2` tests?  Or are you seeing more errors throughout the test suite?

tvrusso, 2022-05-27:
@Karlsefni2012 is correct, it is impossible to deduce what's wrong just from the console output of run_xyce_regression, which only tells you whether a test passed or not.  If it's only the EMBEDDED and EMBEDDED2 tests and only the MC tests within that set (as it appears from your screenshot), it might be a hint as to what's wrong, but only a vague hint.

To be sure what's really going wrong, you have to dig down into the output directories for those tests and look at what actually happened.  For example, if you look at the file Xyce_Test/Netlists/EMBEDDED/normalParamNormalMC.cir.stderr, it should tell what the statistics for that test were.  If it were passing it should look something like this:

```
Try # 0 :

V(1) sampling mean is 875.163, analytical mean is 875
V(1) sampling variance is 119.888, analytical variance is 119.62
V(1) sampling stddev is 10.9494, analytical stddev is 10.93
V(1) sampling skew is 0.123302, analytical skew is 0
V(1) sampling kurtosis is 3.00492, analytical kurtosis is 3

V1 mean result = 0.000186251018381731 
V1 stddev result = 0.00177178658191325 
```

Yours almost certainly says something other than that, though, since your system failed this test.