Xyce/Xyce #66: some model parameter should be integer but ...

laurentc2, 2022-07-01:
Some mode parameter should be written as integer such as MODMOB=0 or BINUNIT=1
But they come from the foundry as MODMOB=0.00 or BINUNIT=1.00
Then Xyce stops with an error ... of course it is easy to correct, but ... it requires to open the models from the foundry.

Best regards,
Laurent

Karlsefni2012, 2022-07-01:
Thanks for your feedback about this.  This has come up before in other contexts for other parameters.  There is a fix for it but we've applied it conservatively on a parameter-by-parameter basis.   I assume you are working with one of the BSIM3 or BSIM4 models?  Are you getting this error for any other parameters in those models or just MOBMOD and BINUNIT?

laurentc2, 2022-07-02:
Hi Eric,
Actually, it was at least on those 2 ones, but then, I have corrected all parameters from 1.0000 to 1 and all 0.0000 to 0 and all 2.0000 to 2, so I don't know if there were other parameters with the same error.
By the way, there was another error in the parameters :.param+ abc = '1.123'   * explanation of abc parameterThe * after the expression is not allowed in Xyce while it is swallowed by other simulators.
BRgds,Laurent