Xyce/Xyce #68: Documentation doesn't describe Xyce features

yurivict, 2022-07-11:
Documentation on https://xyce.sandia.gov/ covers the installation process in details, regression suite, and then goes to FAQ and Xyce/ADMS.

There is no chapter explaining product features. For example, stokhos and amesos2 fatores could be explained there among many other options that I am sure Xyce has.

hkthorn, 2022-07-11:
The supported features in Xyce are found in the Reference and User Guides:

https://xyce.sandia.gov/documentation-tutorials/

The discussion therein does relate the supported features in Xyce to packages in Trilinos.  Specifically, you will find Stokhos mentioned in the Polynomias Chaos Expansion (PCE) section of the User Guide.  The solvers in Amesos2 are not supported features of Xyce at this time, so they are not found in the User Guide.

tvrusso, 2022-07-11:
The documentation and tutorials page was always intended to be THE place for finding documentation and tutorials.

The dropdown menu only is an automatically generated feature of wordpress, and is a new thing that crept into our site quite unbidden when it was migrated from a previous content management system.  It lists ONLY those things that are actually wordpress items.  The many other bits of documentation are all PDF files linked to through wordpress, so they don't get picked up automatically by whatever it is that generates that dropdown.

We will have to see if it is possible to add links to non-wordpress items into that dropdown.  If not, perhaps the dropdown should just be disabled.   We never had it before, and thousands of Xyce users managed to find all of our documentation up to now.

I should note that if you CLICK the "Documentation and Tutorials" thing instead of  just dropping it down, it takes you to the Documentation and Tutorials page, where everything is available.

tvrusso, 2022-07-12:
There is no longer a "Documents and Tutorials" dropdown.  You click the "Documents and Tutorials" menu entry, and you get to the "Documents and Tutorials" page where all the documents and tutorials live.

closed 2022-07-12 by tvrusso