Xyce/Xyce #70: YLIN touchstone with AC+LIN gives always 1+0j in output file for all frequencies

mysol, 2022-08-03:
Hi!
A have a questions about importing s-params and .LIN analysis. I want to get s-parameters output for network, consists two-port element as touchstone file. I'm using YLIN for this purpose. But when i'm trying to get output Xyce always gives just all 1s in real part S11 and S22 for all frequencies and 0s for all other parameters.
As i studied in manuals YLIN works only with HB type analysis. Is this mean that i cannot get s-parameters of network with YLIN? Is there any other way to put element, described by touchstone file? Maybe any manual metod.
Thank You!

tvrusso, 2022-08-03:
Could you please provide a netlist and input files that demonstrate the problem you are having?

Are you trying to import S-parameters through YLIN to describe one component and *also* output S-parameters of a larger circuit containing that YLIN from the same simulation?

mysol, 2022-08-03:
> Could you please provide a netlist and input files that demonstrate the problem you are having?
> 
> Are you trying to import S-parameters through YLIN to describe one component and _also_ output S-parameters of a larger circuit containing that YLIN from the same simulation?

Exactly)

My netlist is below

.MODEL YLIN_MOD1 LIN TSTONEFILE=TQP3M9008.s2p
YLIN YLIN1 1 0 2 0 YLIN_MOD1
P1 1 0 port = 1 z0=50
P2 2 0 port = 2 z0=50
.AC DEC 100 1e9 6e9
.LIN FORMAT=TOUCHSTONE DATAFORMAT=MA sparcalc=1

All files is in attachement, includes netlist, touchstone file of device, file with results
[Netlist_device_result.zip](https://github.com/Xyce/Xyce/files/9255455/Netlist_device_result.zip)

tvrusso, 2022-08-03:
Yes, at the moment the YLIN device has only been implemented for harmonic balance and not .AC.   So trying to use it in a circuit with .LIN analysis will not work, as this requires AC analysis.

mysol, 2022-08-03:
> Yes, at the moment the YLIN device has only been implemented for harmonic balance and not .AC. So trying to use it in a circuit with .LIN analysis will not work, as this requires AC analysis.

Thank You for answer!

closed 2022-08-04 by mysol