Xyce/Xyce #71: Transcient simulation error

laurentc2, 2022-08-16:
Hi,

Although my command is : 
.tran 25n 30u

In the middle of the simulation, I got the warning : Netlist warning: Current time exceeds stop time!

At the end, the data shows the simulation is only over 30us. I don't understand this warning.

Laurent

tvrusso, 2022-08-16:
That is a strange thing to see.  Could you please provide a netlist for us to review.

Karlsefni2012, 2022-08-16:
The "stop time" internally doesn't necessarily refer to the final time point of the simulation.  It is just the time of the next breakpoint, whatever that might be.  So, if you have a PWL source, or a PULSE source in the netlist, you'll have breakpoints before the final time.

This warning is emitted by the trapezoid integrator, and it may just be a roundoff issue.   Landing precisely on breakpoints can be tricky due to roundoff.  If the simulation made it all the way thru to the end, and the results otherwise look OK, it probably isn't anything to worry about.

laurentc2, 2022-08-17:
Hi Eric,
Yes, I use PWL data to simulate a digital circuit based on MOS transistor. 
So far, the simulation is wrong (the output is wrong) because I have difficulties to setup gmin, reltol, abstol ... I use continuation=mos.Do you have any hints for the convergence and simulation of a large CMOS digital circuit included a small SRAM in analog ?Unfortunately, the spice netlist cannot be shared.

Thanks, BRgds,Laurent

Karlsefni2012, 2022-08-17:
Are you having convergence problems, or is the circuit running but producing the wrong answer?  If you are producing the wrong answer then most of the solver options won't help you much.  That is usually going to be a problem with the circuit itself, or with a netlist translation mistake.   A lot of digital circuits are designed for codes like Hspice, so if running an Hspice circuit in Xyce be sure to use the "Xyce -hspice-ext all " command line option.

To address the possibility you are having convergence problems:  If you are having convergence problems during the transient phase (after DCOP) then setting `continuation=mos` won't help.  Continuation methods are only used during the DCOP phase, so it won't affect the solves during time-stepping.

BTW, in my experience there aren't that many circuits that `continuation=mos` will solve that cannot be solved by the default DCOP solvers.  So, we don't use it much these days.

If you are having convergence difficulties in transient (post-DCOP), especially near transitions, I might suggest switching from trapezoid to gear.  You can do that via `.options timeint method=gear`.    If you haven't tried this already, that is.  Sometimes trapezoid can start ringing, and that can lead to problems.

You could also (assuming your problems are happening near transitions) experiment with adding some capacitance to the circuit, via `.options device mincap=1uf` or something like that.  (the value should be small).   This is mostly a debug option; if it works it suggests that your models set their capacitance-related parameters to zero, and the best fix is to set those params, rather than using mincap.

It sounds, however, like your circuit is running all the way thru but getting the wrong answer.  That's a little harder to debug without seeing the netlist.  (which I realize you can't share)  Can you share anything about how it is incorrect?