Xyce/Xyce #72: Compilation issue on Ubuntu 20

platise, 2022-08-24:
had to add find package for FLEX command, as otherwise it didn't found FLEX_EXECUTABLE to the `XyceSuperBuild.cmake`

    diff --git a/XyceSuperBuild.cmake b/XyceSuperBuild.cmake
    index c51b55eb..0c2593cb 100644
    --- a/XyceSuperBuild.cmake
    +++ b/XyceSuperBuild.cmake
    @@ -3,6 +3,7 @@ cmake_minimum_required(VERSION 3.13 FATAL_ERROR)
     
     include(ExternalProject)
     find_package(Git)
    +find_package(FLEX)
     # Find specific folder locations
     
     # Version 3.5.0 required for Trilinos version

tvrusso, 2022-08-24:
The Xyce CMake setup, including superbuild, is still under development and is not our recommended build yet.  We expect the entire superbuild system to be rewritten.

At this time, the autotools build remains our recommended build for now up until our next release, at which time we expect the CMake build to be really ready.

closed 2022-08-24 by tvrusso

platise, 2022-08-24:
Anyway it worked already as it is, with the above patch already ;)

tvrusso, 2022-08-24:
It "works" but is not complete and is in total flux.  It is not safe to rely on it right now.  The autoconf build is stable for now.