Xyce/Xyce #74: Xyce/ADMS: Verilog-A system task inconsistencies

ngwood, 2022-09-12:
Xyce/ADMS recognises three display system tasks:

`$strobe` displays when the simulator has converged on a solution for all nodes;
`$display` provides the same capabilities as `$strobe`;
`$write` provides the same capabilities as `$strobe`, but with no trailing newline.

It doesn't recognise the two remaining display system tasks though:

`$monitor` provides the same capabilities as `$strobe`, but only displays if a change has occurred since the last accepted step.
`$debug` displays for each iteration of the solver.

Notably, the way that `$strobe` and `$display` are loosely implemented in Xyce/ADMS is actually how `$debug` is loosely supposed to be implemented though; namely to display after each solver iteration, rather than after an accepted iteration.

Adding support for `$debug` would therefore be very easy. Is there an easy way to modify things so as to skip `$strobe`, `$display` and `$write` calls except on accepted iterations though? More so, for `$monitor` with the additional requirement of only when one of its arguments change?

Xyce/ADMS also recognises one simulation control system task:

`$finish` exit after the current solution is complete.

It doesn't recognise the five remaining display system tasks though:

`$stop` suspend after the current solution is complete;
`$fatal` display message and exit at current iteration;
`$error` display message and exit after the current solution is complete;
`$warning` display message after the current solution is complete, in a tool specific way;
`$info` display message after the current solution is complete.

As noted in the Xyce/ADMS user guide, the way `$finish` is implemented is more like how `$fatal` is supposed to behave, except that it doesn't let you provide a message.

Adding support for `$fatal` would therefore be fairly straightforward. Is fixing `$finish` to exit only after the current solution is complete going to be difficult though? From what I can see,`$warning` and `$info` can be treated much like how `$strobe` and `$display` should be handled and `$error` is basically equivalent to a `$strobe` followed by `$finish`.`$stop` isn't going to be useful in Xyce/ADMS.

In summary, finding a way to wait until the current solution is complete seems to be the main issue to achieve more compliant behaviour for system tasks. Is this something that can be easily done though?

Thank you in advance.

Neal

mithro, 2022-09-13:
This might be of interest to your team @miesli

miesli, 2022-09-13:
@mithro Thank you for highlighting this issue for us. 

Yes, ADMS as has many limitations, displaying messages is one of many points, why we chose to implement a new Verilog-A compiler.

Our Compiler [OpenVAF](https://gitlab.com/DSPOM/OpenVAF) will support the Verilog-A standard as closely as possible, and this includes these tasks. Sadly, the standard is not always easy or even meaningful. If Xyce can support these messages and implements them accordingly, it will be possible to display them using OpenVAF.