Xyce/Xyce #86: error in building dev. version on Ubuntu with Xyce_GRAPH_DEBUG=ON

bizhishui, 2023-01-31:
When I built Xyce (serial and debug mode) with all debug flags on, the following error occurs:
N_UTL_Graph.h:661:32: error: passing ‘const Key1Map’ {aka ‘const std::unordered_map<int, Xyce::NodeID, std::hash<int>, std::equal_to<int>, std::allocator<std::pair<const int, Xyce::NodeID> > >’} as ‘this’ argument discards qualifiers [-fpermissive]

After looking inside the code, I found it was the "Xyce_GRAPH_DEBUG" flag. This error is not difficult to fix, as shown the diffs attached.

[diff_Xyce_GRAPH_DEBUG.txt](https://github.com/Xyce/Xyce/files/10541194/diff_Xyce_GRAPH_DEBUG.txt)

hkthorn, 2023-04-10:
Thanks for the error report.  We will get it taken care of soon!

hkthorn, 2023-04-18:
This has been fixed in the development branch and will be included in the Xyce 7.7 release.  Thanks again for submitting this issue.

closed 2023-04-18 by hkthorn