Xyce/Xyce #87: SANDreport.cls missing

laurentc2, 2023-02-21:
SANDreport.cls is missing in the doc files, to run make

Laurent

tvrusso, 2023-02-21:
SANDreport.cls is an internal documentation style used only for official Sandia Laboratories publications and cannot be distributed.

The documentation will build without it, but you have to make a minor modification that is documented in the Reference and Users guide LaTeX files.  Once that change is made you can use "make" to get the documentation built, but it will not look like an official Sandia publication, just a regular LaTeX document.

For example, `Xyce/doc/Reference_Guide/Xyce_RG.tex` has this comment right where it makes use of SANDreport.cls:

```
% When compiling at Sandia, uncomment 'sand' and SANDreport
% Outside of Sandia, uncomment 'report' and scrreprt:
\documentclass[11pt,report]{SANDreport}
\usepackage[sand]{optional}
%\documentclass[11pt,letterpaper]{scrreprt}
%\usepackage[report]{optional}
```

This is the best we can do here.

closed 2023-02-21 by tvrusso

tvrusso, 2023-02-21:
I should also remark that we never really expected users outside of Sandia to be running LaTeX over the documentation, as we provide it in PDF form already processed and in official published form at https://xyce.sandia.gov/documentation-tutorials/.  You are better off just downloading from there unless you really need some reference material that was added since the prior release and publication of its documentation.  In that case, just make the edits described above and it'll work.