Xyce/Xyce #88: Mixed-Signal Segmentation Fault

yrrapt, 2023-04-07:
When using the mixed-signal interface as detailed [here](https://xyce.sandia.gov/files/xyce/AppNote-MixedSignal.pdf) I get segmentation faults with some combinations of ADC and DACs but not others.

I am using version `Xyce DEVELOPMENT-202303092313-(Public_Release-7.6.0-146-g5330643a)-opensource`.

The python is from the example:
```python
from xyce_interface import xyce_interface

# this calls the xyce_interface.open() method to
# make a xyce object
xyceObj = xyce_interface()
print(xyceObj)

argv = ["run_dac.cir"]
print("calling initialize with netlist %s" % argv[0])

result = xyceObj.initialize(argv)
print("return value from initialize is %d" % result)
print("Calling runSimulation...")

print(xyceObj.getDACDeviceNames())

result = xyceObj.runSimulation()
print("return value from runSimulation is %d" % result)
print("calling close")
xyceObj.close()
```

The netlist is very simple:
```spice
.model dac_default dac
ydac 1 1 0 dac_default
* ydac 2 2 0 dac_default

.model adc_default adc
yadc 3 3 0 adc_default

R1 1 0 1k
R2 2 0 1k
R3 3 0 1k

.TRAN 0 1
.END
```

The simulation runs fine like this, but if I uncomment `ydac 2 2` then I get segmentation faults:
`[1]    20134 segmentation fault  python xyce_test.py`

There's happens with different arrangements of ADCs & DACs but this is the simplest example.

Do you know what might be going wrong?

Thanks in advance.

yrrapt, 2023-04-07:
Runing with `valgrind python xyce_test.py` provides this output:

```bash
return value from initialize is 1
Calling runSimulation...
==20655== Invalid write of size 1
==20655==    at 0x483F0AC: strcpy (in /usr/lib/x86_64-linux-gnu/valgrind/vgpreload_memcheck-amd64-linux.so)
==20655==    by 0x6394F67: xyce_getDACDeviceNames (in /tools/xyce/install/serial/lib/libxycecinterface.so.0.0.0)
==20655==    by 0x5CC3FF4: ??? (in /usr/lib/x86_64-linux-gnu/libffi.so.7.1.0)
==20655==    by 0x5CC3409: ??? (in /usr/lib/x86_64-linux-gnu/libffi.so.7.1.0)
==20655==    by 0x4864305: _ctypes_callproc (in /usr/lib/python3.8/lib-dynload/_ctypes.cpython-38-x86_64-linux-gnu.so)
==20655==    by 0x48649DB: ??? (in /usr/lib/python3.8/lib-dynload/_ctypes.cpython-38-x86_64-linux-gnu.so)
==20655==    by 0x5F7055: _PyObject_MakeTpCall (in /usr/bin/python3.8)
==20655==    by 0x57107D: _PyEval_EvalFrameDefault (in /usr/bin/python3.8)
==20655==    by 0x569CE9: _PyEval_EvalCodeWithName (in /usr/bin/python3.8)
==20655==    by 0x5F6A12: _PyFunction_Vectorcall (in /usr/bin/python3.8)
==20655==    by 0x56BBDE: _PyEval_EvalFrameDefault (in /usr/bin/python3.8)
==20655==    by 0x569CE9: _PyEval_EvalCodeWithName (in /usr/bin/python3.8)
==20655==  Address 0x0 is not stack'd, malloc'd or (recently) free'd
==20655== 
==20655== 
==20655== Process terminating with default action of signal 11 (SIGSEGV)
==20655==  Access not within mapped region at address 0x0
==20655==    at 0x483F0AC: strcpy (in /usr/lib/x86_64-linux-gnu/valgrind/vgpreload_memcheck-amd64-linux.so)
==20655==    by 0x6394F67: xyce_getDACDeviceNames (in /tools/xyce/install/serial/lib/libxycecinterface.so.0.0.0)
==20655==    by 0x5CC3FF4: ??? (in /usr/lib/x86_64-linux-gnu/libffi.so.7.1.0)
==20655==    by 0x5CC3409: ??? (in /usr/lib/x86_64-linux-gnu/libffi.so.7.1.0)
==20655==    by 0x4864305: _ctypes_callproc (in /usr/lib/python3.8/lib-dynload/_ctypes.cpython-38-x86_64-linux-gnu.so)
==20655==    by 0x48649DB: ??? (in /usr/lib/python3.8/lib-dynload/_ctypes.cpython-38-x86_64-linux-gnu.so)
==20655==    by 0x5F7055: _PyObject_MakeTpCall (in /usr/bin/python3.8)
==20655==    by 0x57107D: _PyEval_EvalFrameDefault (in /usr/bin/python3.8)
==20655==    by 0x569CE9: _PyEval_EvalCodeWithName (in /usr/bin/python3.8)
==20655==    by 0x5F6A12: _PyFunction_Vectorcall (in /usr/bin/python3.8)
==20655==    by 0x56BBDE: _PyEval_EvalFrameDefault (in /usr/bin/python3.8)
==20655==    by 0x569CE9: _PyEval_EvalCodeWithName (in /usr/bin/python3.8)
==20655==  If you believe this happened as a result of a stack
==20655==  overflow in your program's main thread (unlikely but
==20655==  possible), you can try to increase the size of the
==20655==  main thread stack using the --main-stacksize= flag.
==20655==  The main thread stack size used in this run was 8388608.
==20655== 
==20655== HEAP SUMMARY:
==20655==     in use at exit: 1,911,622 bytes in 6,597 blocks
==20655==   total heap usage: 14,513 allocs, 7,916 frees, 11,343,523 bytes allocated
==20655== 
==20655== LEAK SUMMARY:
==20655==    definitely lost: 0 bytes in 0 blocks
==20655==    indirectly lost: 0 bytes in 0 blocks
==20655==      possibly lost: 33,680 bytes in 56 blocks
==20655==    still reachable: 1,877,942 bytes in 6,541 blocks
==20655==                       of which reachable via heuristic:
==20655==                         multipleinheritance: 192 bytes in 6 blocks
==20655==         suppressed: 0 bytes in 0 blocks
==20655== Rerun with --leak-check=full to see details of leaked memory
==20655== 
==20655== Use --track-origins=yes to see where uninitialised values come from
==20655== For lists of detected and suppressed errors, rerun with: -s
==20655== ERROR SUMMARY: 1732 errors from 153 contexts (suppressed: 0 from 0)
[1]    20655 segmentation fault  valgrind python xyce_test.py
```

TBird2001, 2023-04-10:
I'm just glancing at this right now, but when you say that you, "uncomment `ydac 2 2`", does start of the netlist look like the following?
```
.model dac_default dac
ydac 1 1 0 dac_default
ydac 2 2 0 dac_default
```
If so, then the problem is that you have two devices with the same name. Xyce should not be crashing, though, and instead be throwing an error, so we will still be troubleshooting the issue.

tvrusso, 2023-04-10:
Remember that Y devices take their names from the combination of the first two tokens on the line, which are concatenated with a "!" between them.  So those two devices are "YDAC!1" and "YDAC!2" --- not identically named.

TBird2001, 2023-04-10:
Ah, right. My bad. The memory error appeared to be originating with `xyce_getDACDeviceNames`, which put my mind in the wrong space. We'll need to look deeper into the segfault.

tvrusso, 2023-04-10:
I believe you will find the answer to the problem in the definition of the Python function "getDACDeviceNames", where the code is incorrectly calling getNumDevices with the YADC base name instead of the YDAC basename.

```
  def getDACDeviceNames( self ):
    basename = "YADC"
    cBaseName = c_char_p(basename.encode('utf-8'))
    cNumDeviceNames = c_int( 0 )
    cMaxDeviceNameLength = c_int( 0 )
    DACnames = []

    # if the call to xyce_getNumDevices() fails then return an empty array
    status = self.lib.xyce_getNumDevices( byref(self.xycePtr), cBaseName, byref(cNumDeviceNames), byref(cMaxDeviceNameLength) )
    if status != 1:
      return (status, DACnames)
```

This mistake is in both the xyce_interface.py.cmake file in Xyce/utils/XyceCInterface and the one in the Xyce tests suite under Xyce_Regression/Netlists/MIXEDSIGNAL/Python/xyce_interface.py.

rich1149, 2023-04-12:
This has been fixed on the delopment branch of Xyce.  It will show up here on the next push from our internal repository.

yrrapt, 2023-04-13:
Thank you all for your timely help in fixing this bug!

closed 2023-04-13 by yrrapt