Xyce/Xyce #95: YGENEXT in HB

wenjicn, 2024-01-05:
I referred to the document named "AppNote-GenExt.pdf" and created a custom independent voltage source. 
numInternalVars = 1
[image]
When I used this YGENEXT type independent voltage source for tran simulation, I obtained completely correct results. 
But when I use it for HB simulation, this result always approaches 0. 
Even when I set ".OPTION HBINT STARTUPPERIODS=2 TAHB=1", the starting guess in the "startup" file is correct.
The results of "TD" and "FD" files always approach 0.

How can I solve this problem ?

tvrusso, 2024-01-05:
The problem here (and @hkthorn can correct me if I'm wrong) is that HB only works with sources that have been coded to support HB.  It doesn't work if your independent source is an explicitly time-dependent B source, for example.  The same problem obtains with trying to use a YGENEXT device as an independent source.

The reference guide shows this:

[image]

and the same limitation would apply to YGENEXT devices that are strictly time dependent in the same way.

tvrusso, 2024-01-05:
You will find that if you tried to create a B source that did exactly the same thing as your YGENEXT device (just a plain old sine source) you'd get the same results --- transient is correct (including transient startup for transient-assisted HB), but HB will be completely wrong.

That said, if you implemented a computeXyceFDVectors method in your GENEXT device that returned the the correct frequency-domain F vector and dF/dX matrix for your source, it is possible that you could make it work correctly (again, I defer to @hkthorn for specifics).  There is an example of the implementation of that method for the capacitor in the AppNote, and it is explained in section 3.3.2, but of course that is for a device that has no explicit time dependence.  Basically, the F vector you return must  be the fourier transform of the time domain version (which in your case is simple to return analytically) and the matrix must be its derivatives w.r.t. solution variables.

In saying this I'm speaking only hypothetically, as I have never tried to implement an HB-correct source in the YGENEXT device.  But since YGENEXT supports frequency-domain loading, it *should* be possible.  You just can't rely on the time-domain-only loads when the time-domain behavior is explicitly time dependent.

tvrusso, 2024-01-05:
Actually, I think it would be essential that your YGENEXT device return the time-dependent voltage in the B vector instead of F, and the fourier transform of B in frequency domain.  There are no examples of using B in the app note (indeed, all of the examples there explicitly clear B and never even set it to a non-zero length), but it does describe what B means and how it fits into the nonlinear system that Xyce is solving.

wenjicn, 2024-01-08:
@tvrusso Thank you very much !
After my experiment, as long as two points are achieved:
1. The computeXyceFDVectors function must be implemented.
2. Place the Fourier transform result of the signal in B. And no need to consider negative frequency.
The key code is shown in the figure
[image]

tvrusso, 2024-01-08:
I'm glad it worked out for you.  You might look over Xyce/src/DeviceModelPKG/OpenModels/N_DEV_Vsrc.C to see how the built-in sine source is implementedop, since it contains not only a frequency but also a DC offset and phase.  I believe that what you have implemented is the fourier transform of a pure *cosine* source, not a *sine* source, because you've chosen to put the full amplitude in the real part of the B vector.   So you could wind up with an odd phase issue showing up in the time-domain output of HB, which is the inverse Fourier transform of the HB solution.

But that's a subtlety.  The main point is what you've already gotten done, that the GENEXT cannot be used as a source for HB unless its frequency domain features are enabled and implemented (both computeXyceFDVectors and haveFDLoads).

wenjicn, 2024-01-09:
As long as the code works, all that's left is some mathematical problems. DC bias is easily understood as the amplitude at 0 frequency. Phase is only a complex rotation.
 I am doing this just to achieve a truly limited power voltage source.

tvrusso, 2024-01-10:
Yeah, that's what I meant by "But that's a subtlety."  You've solved the main issue, the rest is just getting little details right.

tvrusso, 2024-01-10:
BTW, since this isn't actually a bug in Xyce that requires the Xyce team to fix something, you might want to go ahead and close this issue.

closed 2024-01-10 by wenjicn