Xyce/Xyce #96: .STEP with YGENEXT

wenjicn, 2024-01-18:
When I use the ygenext device for ".STEP" simulation, the parameters of ygenext cannot be updated.
Even if I call the getDParams function of the simulator in the computeXyceVectors function, the parameter value is not updated with ".STEP" .
[image]

How should I make the parameter values of ygenext also follow STEP synchronous update ?

tvrusso, 2024-01-18:
I don't think that `.step` works with parameters that are inside "vector composite" parameters, but @Karlsefni2012 can answer definitively.

Even if it were possible to use .step with parameters that live only inside a vector-composite parameter, the YGENEXT device's handling of parameters is not very advanced --- the only thing we provide is a way for you to query the contents of the vector param and process it yourself.

In addition, as originally conceived, there is no provision for any user code to be executed from the Instance or Model class processParams function (these are usually called when .step updates a parameter).  

It may be that in order for YGENEXT to be able to reprocess user-defined device parameters from .step, the Xyce team might have to beef up the YGENEXT device so that it is possible to provide custom code called by processParams (and your user code would then have to use the appropriate API calls to obtain the updated parameters and do with them as you require).

wenjicn, 2024-01-22:
Thank you for your answer ！

closed 2024-01-22 by wenjicn