Xyce/Xyce #100: Build error on Ubuntu 20.04 LTS

chenshih1, 2024-06-17:
I try to build Xyce on Ubuntu 20.04 LTS, but the compilation failed at the last step.

The dependencies are prepared following the instructions on the official website with Trilinos 12.12.1, and the compiler is gcc 9.4.0.

Here's the compilation command.
Trilinos 12.12.1 compilation
`cmake -B build -DCMAKE_C_COMPILER=mpicc -DCMAKE_CXX_COMPILER=mpic++ -DCMAKE_Fortran_COMPILER=mpif77 -DCMAKE_CXX_FLAGS="-O3 -fPIC" -DCMAKE_C_FLAGS="-O3 -fPIC" -DCMAKE_Fortran_FLAGS="-O3 -fPIC" -DTPL_ENABLE_MPI=ON -DTrilinos_ENABLE_NOX=ON -DNOX_ENABLE_LOCA=ON -DTrilinos_ENABLE_EpetraExt=ON -DEpetraExt_BUILD_BTF=ON -DEpetraExt_BUILD_EXPERIMENTAL=ON -DEpetraExt_BUILD_GRAPH_REORDERINGS=ON -DTrilinos_ENABLE_TrilinosCouplings=ON -DTrilinos_ENABLE_Ifpack=ON -DTrilinos_ENABLE_Isorropia=ON -DTrilinos_ENABLE_AztecOO=ON -DTrilinos_ENABLE_Belos=ON -DTrilinos_ENABLE_Teuchos=ON -DTrilinos_ENABLE_COMPLEX_DOUBLE=ON -DTrilinos_ENABLE_Amesos=ON -DAmesos_ENABLE_KLU=ON -DTrilinos_ENABLE_Amesos2=ON -DAmesos2_ENABLE_KLU2=ON -DAmesos2_ENABLE_Basker=ON -DTrilinos_ENABLE_Sacado=ON -DTrilinos_ENABLE_Stokhos=ON -DTrilinos_ENABLE_Kokkos=ON -DTrilinos_ENABLE_Zoltan=ON -DTrilinos_ENABLE_ALL_OPTIONAL_PACKAGES=OFF -DTPL_ENABLE_AMD=ON -DAMD_LIBRARY_DIRS="/usr/lib" -DTPL_AMD_INCLUDE_DIRS="/usr/include/suitesparse" -DTPL_ENABLE_BLAS=ON -DTPL_ENABLE_LAPACK=ON -DCMAKE_INSTALL_PREFIX="/usr/local" && cmake --build build --parallel $(nproc) && cmake --install build`

Xyce 7.8.0 compilation
`./bootstrap && ./configure CXXFLAGS="-O3" CPPFLAGS="-I/usr/include/suitesparse" CC=mpicc CXX=mpic++ F77=mpif77 --enable-mpi --enable-stokhos --enable-amesos2 --prefix=/usr/local && make -j $(nproc) && make install`

Here's the config.log of Xyce.
[config.log](https://github.com/user-attachments/files/15866297/config.log)

Here's the information produced by the compiler.
[build.log](https://github.com/user-attachments/files/15866133/build.log)

tvrusso, 2024-06-17:
I'm looking into your issue on Ubuntu 22.04, as I have no access to Ubuntu 20.04 anymore.  It will be some time before I can reproduce your problem (but I probably can't, because I'm pretty sure where the problem lies and it isn't likely with Ubuntu).

The errors you're getting point rather strongly at a problem with the Trilinos build.  Errors like that are almost *always* due to a mismatch between what Xyce needs for Trilinos and how Trilinos was actually built.  Your Trilinos build process *looks* like it should match, but something clearly went wrong.

Is there any possibility that you have installed any version of Trilinos from a package management system?  If so, the presence of Trilinos libraries and headers in /usr/lib and /usr/include will be causing problems, because of the way you've chosen to deviate from our recommendations (the /usr/lib and /usr/include stuff will be found *instead* of the ones you built from source, and no package management system we know of has a version of Trilinos built in such a way that it will work with Xyce).  I am pretty confident that a system-installed Trilinos from package management is the issue, because if you look at config.log where it is probing for -lteuchoscore, there is no "-L/usr/local/lib" before the "-lteuchoscore" on the link line, so it's probably finding an incompatible one in /usr/lib or /lib.

For what it's worth, the Xyce team has always *strongly* advised against installing Trilinos and Xyce directly into /usr/local/lib and /usr/local/bin because serial and parallel versions of Xyce and Trilinos could interfere with each others.  This is why we always advocated installing serial and parallel versions of Trilinos into uniquely named directories.  Our prototypical example was usually something like `XyceLibs/Linux_serial` and `XyceLibs/Linux_parallel` (both of which could be directly under /usr/local if you like, or just under your own home directory, or whatever), and then specifying that directory as the ARCHDIR to Xyce's configure. 

Furthermore, by doing that you can avoid mismatch between conflicting versions of Trilinos lying around your system that Xyce's configure might find otherwise, because our guidance is designed to force the issue and make configure look first where you tell it, instead of just picking up random versions from elsewhere.

Run the way you're trying to do, with no ARCHDIR, and the Trilinos stuff installed to /usr/local/lib and /usr/local/include, configure could easily pick up a system-installed, completely incompatible version of Trilinos /usr/lib and /usr/include --- that's because in general those latter directories would be searched first by configure's probes unless it's told otherwise.

The very first thing I'd recommend to you is for you to make sure you do NOT have any trilinos package installed from your system's package management.  If you do, *remove* it unless it is a requirement for something else you must have other than Xyce.  Its presence is almost certainly a problem, especially given your choices to deviate from our recommendations.

Then remove the Trilinos libraries and headers you installed in /usr/local.  Use the "install_manifest.txt" file that cmake created to find all the files you'll need to delete, since they'll be scattered all around /usr/local/lib and/usr/local/include (ease of cleanup is yet another reason to use a uniquely named install directory).   Finally, change your cmake invocation for Trilinos to:

```
cmake -B build \
-DCMAKE_C_COMPILER=mpicc \
-DCMAKE_CXX_COMPILER=mpic++ \
-DCMAKE_Fortran_COMPILER=mpif77 \
-DCMAKE_CXX_FLAGS="-O3 -fPIC" \
-DCMAKE_C_FLAGS="-O3 -fPIC" \
-DCMAKE_Fortran_FLAGS="-O3 -fPIC" \
-DTPL_ENABLE_MPI=ON \
-DTrilinos_ENABLE_NOX=ON \
  -DNOX_ENABLE_LOCA=ON \
-DTrilinos_ENABLE_EpetraExt=ON \
  -DEpetraExt_BUILD_BTF=ON \
  -DEpetraExt_BUILD_EXPERIMENTAL=ON \
  -DEpetraExt_BUILD_GRAPH_REORDERINGS=ON \
-DTrilinos_ENABLE_TrilinosCouplings=ON \
-DTrilinos_ENABLE_Ifpack=ON \
-DTrilinos_ENABLE_Isorropia=ON \
-DTrilinos_ENABLE_AztecOO=ON \
-DTrilinos_ENABLE_Belos=ON \
-DTrilinos_ENABLE_Teuchos=ON \
-DTrilinos_ENABLE_COMPLEX_DOUBLE=ON \
-DTrilinos_ENABLE_Amesos=ON \
  -DAmesos_ENABLE_KLU=ON \
-DTrilinos_ENABLE_Amesos2=ON \
  -DAmesos2_ENABLE_KLU2=ON \
  -DAmesos2_ENABLE_Basker=ON \
-DTrilinos_ENABLE_Sacado=ON \
-DTrilinos_ENABLE_Stokhos=ON \
-DTrilinos_ENABLE_Kokkos=ON \
-DTrilinos_ENABLE_Zoltan=ON \
-DTrilinos_ENABLE_ALL_OPTIONAL_PACKAGES=OFF \
-DTPL_ENABLE_AMD=ON \
-DAMD_LIBRARY_DIRS="/usr/lib" \
-DTPL_AMD_INCLUDE_DIRS="/usr/include/suitesparse" \
-DTPL_ENABLE_BLAS=ON \
-DTPL_ENABLE_LAPACK=ON \
-DCMAKE_INSTALL_PREFIX="/usr/local/XyceLibs/Linux_parallel"
```
That is, simply change the install prefix to something unique that will contain *only* Trilinos libraries and headers (and supporting CMake noise).   Make sure the source directory is completely pristine (since you chose not to use an out-of-source build as we recommend, your source tree will have been polluted with build artifacts that you must clean out.  `make clean` might get most of them, but not all.  `git clean -fdx` should get rid of them all if you actually cloned Trilinos and checked out the 12.12.1 branch, but not if you downloaded a tarball.

Finally, rebuild and install Trilinos all over again.  Since this version of trilinos is very unlikely to be useful for anything *other* than building Xyce, having it live in its own little sandbox off to the side of /usr/local or your home directory shouldn' [... cut ...]
Doing things this way will assure that Xyce's configure picks up and uses the version of Trilinos you have built with MPI and not some other version that might be somewhere else.   Installing Xyce to a uniquely named subdirectory of /usr/local assures that the serial and parallel versions of Xyce (both of which install their own component libraries) don't interfere with each other.  You can always symlink "Xyce_serial" and "Xyce_parallel" to /usr/local/XyceSerial/bin/Xyce and /usr/local/XyceParallel/bin/Xyce if you don't want to tinker with your PATH.

chenshih1, 2024-06-18:
@tvrusso Thanks a lot, I'll try it later.

tvrusso, 2024-06-18:
In that case, if there really was no other trilinos installed in /usr/lib, the reason Xyce's link might have failed *could* have been because without ARCHDIR, the "Makefile.export" files from Trilinos weren't located.

These files tell configure exactly what interdependent Trilinos libraries must be linked in and in what order.

Without the Makefile.export files, Xyce's configure does its best to figure out what Trilinos libraries are needed and how to link them, but because the interdependencies are not fixed and predictable, it does not do a very thorough job and this capability has decayed over the years.

The reason for the problem is that Trilinos is a very, very complex collection of interdependent libraries, and the exact interdependencies change depending on exactly which version of Trilinos is in use, but even worse, it depends on exactly how Trilinos was configured.  

The Makefile.export system (which the Trilinos team dropped after the 12.x series of releases, because they no longer wanted to maintain support for downstream projects that did not use CMake) contains all the needed interdependency information, and Xyce's configure uses these files to figure out exactly which extra libraries for *that* compilation of Trilinos.  If those Makefile.export files are not found, configure falls back on the much, much less reliable technique of trying to pick which libraries are needed by itself.  It does so by enumerating the libraries we *know* in advance that we need, and trying them out in simple ways.   But it can't pick up that some library we know we need depends on some other library we don't know about.   It may have missed some dependency in the process.

This issue of missing interdependencies is one of the reasons that the Xyce project is being forced to move to CMake in order to work with any version of Trilinos later than version 14 -- if both Xyce and Trilinos are working from the same CMake build system, the interdependencies are communicated through CMake artifacts, not clumsy hacks.  

But because Trilinos is a moving target that pushes the bleeding edge of CMake versions, because Xyce's needs are sufficiently complex, and because the team is very short staffed, the Xyce CMake build system took a *lot* longer than expected, had to be restarted from scratch at least 3 times, and as far as I know is still not considered completely finished or even stable enough to drop autoconf completely (though I believe the team is now very close to declaring it so).

It is thus *very* important to follow the guidance in the building guide very closely, even when it looks like we're making weird suggestions --- all that guidance is there to help navigate the extreme complexity of the Trilinos issue.  Deviation from that guidance is the path to madness.

tvrusso, 2024-10-22:
This is not an accurate list of the apt installs necessary for a normal install of Xyce.  The complete list needed for building against the recommended version of Trilinos being used by the original poster is given in the building guide on the Xyce web site.

olisnr, 2024-10-22:
thats possible. but its difficult to install Xyce, and i needed much time to get it run, so i try to help others to save time (not SSD space ;)...

but please give us an better list of the things we need for Xyce.

i will also use it in the future!

tvrusso, 2024-10-22:
The complete list of apt installs needed on Ubuntu is found in the [building guide at the Xyce web site](https://xyce.sandia.gov/documentation-tutorials/building-guide/).  Everything that is needed is right there.

olisnr, 2024-10-23:
@tvrusso the thing is: there are many different modules to build. as example the trillios stuff is build from a lot different libraries. 
so there are the things You need for Xyce, the things for ADMS, for the Trillios & the parallel-stuff. 

i tried to follow all the instructions, on two new installed Linux-Mint 20.3 (ubuntu 22.04) and there where still things missing. i needed long time to get everything compiled, and in the end i looked in the bash history what all was installed. this is what i uploaded. 

probably there are a few library to much, but its working for me. so its possible, that somebody else could use it...

tvrusso, 2024-10-23:
That be as it may be, the list in the building guide has been thoroughly tested at each release up to the current one, and your list is excessive.

olisnr, 2024-10-23:
@tvrusso You are absolutely correct: as example my list installs `gmp-static` 3 times :) not very deliberate...