Xyce/Xyce #102: build requires deprecated trilinos packages

Kreijstal, 2024-07-17:
while building trilinos
```
********************************************************************
********  WARNING WARNING WARNING WARNING WARNING WARNING  *********
********************************************************************
* Several Trilinos packages are scheduled for deprecation in 2025.
* The following deprecated packages are enabled (either explicitly
* or implicitly due to a dependency)
*
*    Triutils
*    ThyraEpetraExtAdapters
*    ThyraEpetraAdapters
*    Ifpack
*    EpetraExt
*    Epetra
*    AztecOO
*    Amesos
*
********************************************************************
```

So xyce won't be able to be compiled with trilinos in a year, which means you'll have to update it, or fork trilinos and vendor it, not very nice solutions.

https://github.com/msys2/MINGW-packages/issues/21150

Karlsefni2012, 2024-07-22:
We have been aware of this issue for a long time and have been working on it.

tcaduser, 2024-07-23:
you can also use git submodules and lock in the version of the tools that you need by commit hash.

Kreijstal, 2024-07-23:
> you can also use git submodules and lock in the version of the tools that you need by commit hash.

That's short sighted

Kreijstal, 2024-11-27:
it's almost december of 2024.

Karlsefni2012, 2024-11-27:
Today's date doesn't change the amount of work necessary to make this happen.   There are substantial changes necessary in the Xyce source code (as well as on the Trilinos side) that is necessary to make this happen.

Several Xyce developers are also Trilinos developers (they are both Sandia products).  This change didn't catch us by surprise.  Like I said in July we've been working on this.