Xyce/Xyce #103: xyce-7.8.0 fails to configure with trilinos-16.0.0: -ltriutils isn't found

yurivict, 2024-07-24:
```
checking for main in -ltriutils... no
```

trilinos-16.0.0 doesn't install this library any more.

FreeBSD 14.1

hkthorn, 2024-07-24:
Trilinos has deprecated Epetra-based packages, so all of these packages (like Triutils) will need to be explicitly enabled now.   Can you include your configure script to this issue so we can check if you have all the packages you need enabled?

yurivict, 2024-07-24:
cmake arguments are:
-DCMAKE_INSTALL_PREFIX:PATH=/usr/local -DBUILD_SHARED_LIBS:BOOL=ON -DTrilinos_ENABLE_ALL_PACKAGES:BOOL=ON -DTrilinos_ENABLE_ALL_OPTIONAL_PACKAGES:BOOL=ON -DTrilinos_ENABLE_SEACAS:BOOL=OFF -DTrilinos_ENABLE_PyTrilinos:BOOL=OFF -DTrilinos_ENABLE_OpenMP:BOOL=OFF -DUMFPACK_INCLUDE_DIRS=/usr/local/include/suitesparse -DTPL_AMD_INCLUDE_DIRS=/usr/local/include/suitesparse -DAMD_LIBRARY_DIRS=/usr/local/lib -DCMAKE_C_COMPILER:STRING="cc"  -DCMAKE_CXX_COMPILER:STRING="c++"  -DCMAKE_C_FLAGS:STRING="-O2 -pipe  -fstack-protector-strong -fno-strict-aliasing "  -DCMAKE_C_FLAGS_DEBUG:STRING="-O2 -pipe  -fstack-protector-strong -fno-strict-aliasing "  -DCMAKE_C_FLAGS_RELEASE:STRING="-O2 -pipe  -fstack-protector-strong -fno-strict-aliasing  -DNDEBUG"  -DCMAKE_CXX_FLAGS:STRING="-O2 -pipe -fstack-protector-strong -fno-strict-aliasing  -DBOOST_STACKTRACE_GNU_SOURCE_NOT_REQUIRED "  -DCMAKE_CXX_FLAGS_DEBUG:STRING="-O2 -pipe -fstack-protector-strong -fno-strict-aliasing  -DBOOST_STACKTRACE_GNU_SOURCE_NOT_REQUIRED "  -DCMAKE_CXX_FLAGS_RELEASE:STRING="-O2 -pipe -fstack-protector-strong -fno-strict-aliasing  -DBOOST_STACKTRACE_GNU_SOURCE_NOT_REQUIRED  -DNDEBUG"  -DCMAKE_EXE_LINKER_FLAGS:STRING=" -lexecinfo -Wl,-rpath=/usr/local/lib/gcc13  -L/usr/local/lib/gcc13 -fstack-protector-strong "  -DCMAKE_MODULE_LINKER_FLAGS:STRING=" -lexecinfo -Wl,-rpath=/usr/local/lib/gcc13  -L/usr/local/lib/gcc13 -fstack-protector-strong "  -DCMAKE_SHARED_LINKER_FLAGS:STRING=" -lexecinfo -Wl,-rpath=/usr/local/lib/gcc13  -L/usr/local/lib/gcc13 -fstack-protector-strong "  -DCMAKE_INSTALL_PREFIX:PATH="/usr/local"  -DCMAKE_AUTOGEN_PARALLEL:STRING="7"  -DCMAKE_BUILD_TYPE:STRING="Release"  -DTHREADS_HAVE_PTHREAD_ARG:BOOL=YES  -DCMAKE_INSTALL_RPATH_USE_LINK_PATH=YES  -DCMAKE_VERBOSE_MAKEFILE:BOOL=ON  -DFETCHCONTENT_FULLY_DISCONNECTED:BOOL=ON -DEpetraExt_BUILD_BTF:BOOL=ON -DEpetraExt_BUILD_EXPERIMENTAL:BOOL=ON -DEpetraExt_BUILD_GRAPH_REORDERINGS:BOOL=ON -DTrilinos_ENABLE_COMPLEX_DOUBLE:BOOL=ON -DTPL_ENABLE_UMFPACK:BOOL=ON -DTPL_ENABLE_AMD:BOOL=ON -DPython_ADDITIONAL_VERSIONS=3.11 -DPython_EXECUTABLE:FILEPATH="/usr/local/bin/python3.11" -DPython3_EXECUTABLE:FILEPATH="/usr/local/bin/python3.11" -DBOOST_PYTHON_SUFFIX:STRING=311

hkthorn, 2024-07-24:
@yurivict Would you mind including your Xyce configure script as well?  Many thanks!

yurivict, 2024-07-24:
Xyce cmake arguments:
-DPython_ADDITIONAL_VERSIONS=3.11 -DPython_EXECUTABLE:FILEPATH="/usr/local/bin/python3.11" -DPython3_EXECUTABLE:FILEPATH="/usr/local/bin/python3.11" -DBOOST_PYTHON_SUFFIX:STRING=311

hkthorn, 2024-07-30:
@yurivict Due to the pending deprecation of the Epetra-based Trilinos packages, the logic that enables ALL packages in Trilinos no longer includes the packages the Xyce needs to compile.  Therefore, several packages will need to be explicitly enabled in Trilinos 16 using these cmake options:
```
-DTrilinos_ENABLE_Ifpack=ON \
-DTrilinos_ENABLE_AztecOO=ON \
-DTrilinos_ENABLE_Amesos=ON \
```
We are in the process of integrating the Tpetra-based Trilinos packages so this is will not be necessary in the long run.

Karlsefni2012, 2024-08-22:
If you are building Xyce with a recent version of Trilinos (more recent than 12.12.1), you need to use CMake instead of the autotools.

The autotools-based build process is being deprecated and is only guaranteed to work with Trilinos 12.12.1.

For guidance on how to use CMake to build Xyce, see the INSTALL.md file at the top of the Xyce source tree.

yurivict, 2024-08-22:
When Xyce is switched to cmake it fails to configure with trilinos-16.0.0 built with the above options:

```
===>  Performing out-of-source build
/bin/mkdir -p /usr/ports/cad/xyce/work/.build
-- Looking for Trilinos
   Required packages:
        Amesos Epetra EpetraExt Ifpack NOX Teuchos Sacado
        Triutils AztecOO Belos TrilinosCouplings
   Optional packages:
        Isorropia Zoltan ShyLU ShyLU_DDCore Amesos2 Stokhos ROL MKL
-- Enabled Kokkos devices: SERIAL
CMake Warning at /usr/local/lib/cmake/Kokkos/KokkosConfig.cmake:284 (MESSAGE):
  The installed Kokkos configuration does not support CXX extensions.
  Forcing -DCMAKE_CXX_EXTENSIONS=Off
Call Stack (most recent call first):
  /usr/local/lib/cmake/TeuchosCore/TeuchosCoreConfig.cmake:178 (include)
  /usr/local/lib/cmake/Teuchos/TeuchosConfig.cmake:176 (include)
  /usr/local/lib/cmake/Amesos/AmesosConfig.cmake:156 (include)
  /usr/local/lib/cmake/Trilinos/TrilinosConfig.cmake:123 (include)
  CMakeLists.txt:42 (find_package)

CMake Error at CMakeLists.txt:42 (find_package):
  Found package configuration file:

    /usr/local/lib/cmake/Trilinos/TrilinosConfig.cmake

  but it set Trilinos_FOUND to FALSE so package "Trilinos" is considered to
  be NOT FOUND.  Reason given by package:

  ERROR: Could not find component 'EpetraExt'!

```

rich1149, 2024-08-22:
Trilinos can be built with a variety of sub-packages active and there isn't a "default" build of Trilinos that will work with Xyce.  The error you hit above suggests that the Trilinos sub-pacakge "EpetraExt" is not available.  If you look at this page in the Xyce building guide, https://xyce.sandia.gov/documentation-tutorials/building-guide/ about half way down the page, It shows you the Cmake invocation to use to build Trilinos.  The invocation shown for Trilinos 12.12 works with the current version Trilnos too and it specifies the need package within Trilinos that are needed (i.e. with such lines as -DTrilinos_ENABLE_EpetraExt=ON)

closed 2024-08-23 by yurivict