Xyce/Xyce #108: Ubuntu Xyce compilation fail.

georgtree, 2024-10-12:
Hello!
Firstly I compiled trilinos:
```
  FLAGS="-O3 -fPIC"
  cmake \
  -DCMAKE_C_COMPILER=/usr/bin/gcc \
  -DCMAKE_CXX_COMPILER=/usr/bin/g++ \
  -DCMAKE_Fortran_COMPILER=/usr/bin/gfortran \
  -D TPL_ENABLE_Matio:BOOL=OFF \
  -G "Unix Makefiles" \
  -DCMAKE_CXX_FLAGS="$FLAGS" \
  -DCMAKE_C_FLAGS="$FLAGS" \
  -DCMAKE_Fortran_FLAGS="$FLAGS" \
  -DCMAKE_INSTALL_PREFIX=$ARCHDIR \
  -DCMAKE_MAKE_PROGRAM="make" \
  -DTrilinos_ENABLE_NOX=ON \
    -DNOX_ENABLE_LOCA=ON \
  -DTrilinos_ENABLE_EpetraExt=ON \
    -DEpetraExt_BUILD_BTF=ON \
    -DEpetraExt_BUILD_EXPERIMENTAL=ON \
    -DEpetraExt_BUILD_GRAPH_REORDERINGS=ON \
  -DTrilinos_ENABLE_TrilinosCouplings=ON \
  -DTrilinos_ENABLE_Ifpack=ON \
  -DTrilinos_ENABLE_AztecOO=ON \
  -DTrilinos_ENABLE_Belos=ON \
  -DTrilinos_ENABLE_Teuchos=ON \
  -DTrilinos_ENABLE_COMPLEX=ON \
  -DTrilinos_ENABLE_COMPLEX_DOUBLE=ON \
  -DTrilinos_ENABLE_Amesos=ON \
    -DAmesos_ENABLE_KLU=ON \
  -DTrilinos_ENABLE_Amesos2=ON \
  -DAmesos2_ENABLE_KLU2=ON \
  -DAmesos2_ENABLE_Basker=ON \
  -DTrilinos_ENABLE_Sacado=ON \
  -DTrilinos_ENABLE_Stokhos=ON \
  -DTrilinos_ENABLE_Kokkos=ON \
  -DTrilinos_ENABLE_ALL_OPTIONAL_PACKAGES=OFF \
  -DTrilinos_ENABLE_CXX11=ON \
  -DTPL_ENABLE_AMD=ON \
  -DAMD_LIBRARY_DIRS="/usr/lib" \
  -DTPL_AMD_INCLUDE_DIRS="/usr/local/include/suitesparse/" \
  -DTPL_ENABLE_BLAS=ON \
  -DTPL_ENABLE_LAPACK=ON \
  -DCMAKE_INSTALL_PREFIX=/home/georgtree/trilinos_serial/build \
  /home/georgtree/trilinos_serial/Trilinos-trilinos-release-12-12-1/

  make -j16 install
```

Then I try to configure Xyce with next command:
```
 /home/georgtree/Downloads/Xyce-7.8/configure \
CXXFLAGS="-O3" \
ARCHDIR="/home/georgtree/trilinos_serial/build/" \
CPPFLAGS="-I/usr/include/suitesparse" \
--enable-stokhos \
--enable-amesos2 \
--prefix=/home/georgtree/xyce_serial/
```
and then got:

```
checking whether HAVE_TEUCHOS_COMPLEX is declared... no
configure: error: Your Trilinos was not built with the correct options and does not have COMPLEX support built into Teuchos.  Please rebuild Trilinos using the options specified in the Xyce building guide.
```

I used the recommended procedure and still no success..thank you in advance.

tvrusso, 2024-10-12:
Please attach the "config.log" file that configure produced.  This will have detailed information about *why* configure is concluding that Trilinos isn't usable.

Something is clearly wrong here, but the information you've provided isn't enough to figure out what it is.  Clearly it's not as simple as Trilinos not having been configured properly, since your trilinos cmake invocation does look correct.  It is identical in every important respect to the one I use when I build on Ubuntu.

georgtree, 2024-10-12:
I found a error - with clean configuration of trilinos error dissapeared

closed 2024-10-12 by georgtree

olisnr, 2024-10-22:
for ubuntu 20.04 install:

sudo apt install gmp-static
sudo apt install -y libgmp-dev
sudo apt install -y libmpfr-dev
sudo apt install -y libatlas-base-dev
sudo apt install -y gfortran make cmake bison flex libfl-dev libfftw3-dev libsuitesparse-dev libblas-dev liblapack-dev libtool autoconf automake git libopenmpi-dev openmpi-bin

sudo apt install gmp-static
sudo apt install -y libgmp-dev
sudo apt install -y libmpfr-dev
sudo apt install -y libatlas-base-dev
sudo apt install nvidia-cuda-toolkit

sudo apt install -y bison flex libfl-dev libfftw3-dev libsuitesparse-dev libblas-dev liblapack-dev libtool libopenmpi-dev openmpi-bin  libtpl-dev libsuitesparse-dev libumfpack5

sudo apt install gmp-static
sudo apt install -y libgmp-dev
sudo apt install -y libmpfr-dev
sudo apt install -y libatlas-base-dev

sudo apt install libnetcdf-dev

then try again

tvrusso, 2024-10-22:
That list of apt installs is far more than are needed on Ubuntu for a normal install.

olisnr, 2024-10-22:
thats possible. but its difficult to install Xyce, and i needed much time to get it run, so i try to help others to save time (not SSD space ;)...

but please give us an better list of the things we need for Xyce.

i will also use it in the future!

tvrusso, 2024-10-22:
The complete list of apt installs needed on Ubuntu is found in the [building guide at the Xyce web site](https://xyce.sandia.gov/documentation-tutorials/building-guide/).  Everything that is needed is right there.