Xyce/Xyce #111: Xyce 7.7.0 compilation fails unless `-D Trilinos_ENABLE_Epetra=ON` is set when building Trilinos

dolphingarlic, 2024-10-28:
I followed [the installation guide for release 7.7.0](https://github.com/Xyce/Xyce/blob/4d3ddea894689560ab68d1a65a42ef585818231c/INSTALL.md) and couldn't get it to work until I explicitly enabled Epetra while building Trilinos. Has this issue been fixed in the latest release?

hkthorn, 2024-10-28:
@dolphingarlic This depends upon the version of Trilinos you are using.  Trilinos is in the process of deprecating Epetra, and the solver packages supported by Epetra, in favor of Tpetra (a more modern implementation).  So, the development branch of Trilinos is disabling Epetra by default now.  Xyce is in the process of transitioning to Tpetra as well, but cannot use it yet.

However, in the instructions for building Trilinos:

https://github.com/Xyce/Xyce/blob/4d3ddea894689560ab68d1a65a42ef585818231c/INSTALL.md#building-trilinos  

There is a SHA provided that we test with just to be sure that Trilinos can work with Xyce during this transition.  Did you try building Trilinos with that specific SHA?

dolphingarlic, 2024-10-29:
Yes, I used that specific SHA