Xyce/Xyce #115: Illegal instruction errors on all parallel tests FreeBSD 13.4

tvrusso, 2024-11-14:
After updating my system from FreeBSD 13.3 to 13.4, newly compiled parallel Xyce fails all regression tests with illegal instruction errors.  This happens even in "parallel build/serial run" mode, which should make it easier to find.   I suspect, but have not yet proven, that this is an issue similar to #113, with some bit of code trying to do &(vector[0]) on an STL vector of size zero.  If so, they are certainly inside MPI ifdefs.

I will spend some time hunting this down and will issue a pull request with fixes if I find them.

tvrusso, 2024-11-14:
I am quite wrong about where the problem lies.  It has apparently nothing to do with &(vector[0]) where vector is empty.

A large number of the crashes are due to this code in N_PDS_MPI.C, near line 540:

```
    for (unsigned i = 0; i < size; ++i)
          dest[i] = std::string(&receive[receive_displacement[i]], &receive[receive_displacement[i] + receive_count[i]]);
```

That is, the std::string constructor is being called with two char pointers, presumably the beginning and ending of the string to copy.

I can find no current documentation about a two-pointer constructor of this form in the std::string class.  Perhaps it is an ancient thing?

The code crashes hard here.

Changing this to:
```
      dest[i] = std::string(&receive[receive_displacement[i]], receive_count[i]);
```
i.e., copying receive_count characters instead of trying to copy the data between two pointers, which *IS* a currently documented version of the std::string constructor, fixes a large number of my failures, but not all of them.

I see several uses of exactly this sort of string constructor in N_PDS_MPI.C, so perhaps the other failures I'm seeing are happening in other places but for the same reason.  I'll continue digging.

tvrusso, 2024-11-14:
I found more issues with &(vector[0]) in N_TOP_ParLSUtil.C that were exactly the same as #113's issue --- attempting unconditional access even when the vector is empty.

There may be many more such issues throughout the code.  Fixing the ones in  `ParLSUtil::testVoltageNodeConnectivity_()` fixed some of my failures.  Not sure if it's fixed them all yet.

tvrusso, 2024-11-14:
Yeah, fixing the `ParLSUtil::testVoltageNodeConnectivity_()` misuse of &(vector[0]) only fixed about 56 of my failing parallel netlists, leaving about 900 more to sort out.

I've exhausted my budget for volunteer debugging services for the week, and will come back to this next week if I can.   

I am wondering if this sort of thing is why the Xyce mailing list got a query from someone on RHEL9 who was seeing all his parallel runs crash.  Perhaps something in that OS's STL implementation is exposing the same sort of problem mine is.

There are a lot of uses of &(vector[0]) throughout the code and many of them may be unsafe, and the apparently-illegal usage of the `std::string()` constructor with two char pointer arguments isn't limited to the one place I fixed already in my local repo, so it might be useful for the team to revisit the work of commit 92e522bf and hunt them down more systematically.

The best I can do is pick at this in my spare time until my parallel runs of the test suite stop failing.

Karlsefni2012, 2024-11-14:
Thanks for reporting this.  None of us here had (yet) observed these issues.  But it was probably just a matter of time.

tvrusso, 2024-11-14:
Yeah.  When you're limited to only a few internal OSen, and never bleeding edge versions of those, either, it will be hard to catch things like this.

FreeBSD 13.3 used Clang 17, and this version uses Clang 18.  They must have changed the STL headers that go with.

My old builds of Xyce from 13.3 still run fine without any of these problems, it's only the newly compiled versions that break.  Only thing that changed was the OS version and Clang version.

tvrusso, 2024-11-15:
FWIW, I went back and ran parallel Xyce with no changes (other than the ones I'd fixed in serial for PR #114), and found that Xyce crashed on 1914 tests out of 2083, and the ones that passed were mostly those that are supposed to exit with an appropriate error message (as with things like syntax errors or other netlist mistakes).

After fixing just the two issues I mentioned in previous comments (the weird string constructor thing and the &(vector[0]) thing just in `ParLSUtil::testVoltageNodeConnectivity_()`, it "only" crashes on 903 tests out of 2083, so those two are responsible for a large fraction of the problems.

I had a little trouble narrowing down the remaining problems yesterday, and haven't done any more work on it today.  I do know that ABM/smoothbsrc.cir is one of the 903 crashing netlists, and it's puking in  `Xyce::Parallel::Migrate<Xyce::NodeID, Xyce::Topo::ParNode>::operator()` in `ParallelDistPKG/N_PDS_Migrate.h`, but when I tried fixing the handful of &(vector[0]) issues in that function yesterday it didn't fix even that one netlist, so I undid that and left it for when I really have a block of time to debug.  I think my issue yesterday was that I hadn't built Xyce with all optimization and inlining turned off, and the stack trace was misleading.  Today it seems to point more clearly to a line that makes sense.  But again I've used up my volunteer hours for the day.

tvrusso, 2024-11-15:
So, after realizing that optimization and inlining were confusing the issue for gdb, I did spend a few more minutes on this and tracked down a few more vector[0] issues, enough to get ABM/smoothbsrc.cir to run in parallel.  It turned out to require 8 lines of code to be tweaked in two files.  They were all the same change ( `&(vector[0])` changed to `vector.data()`, which technique wouldn't have existed in the vector class back when those lines were originally written, and to be safe back then would have required checking that the vector wasn't empty first).

After fixing those necessary to get that one test running, I find that the test suite now fails "just" 421 tests, so I'm making progress.

Even after I fix everything on my end to make the whole test suite pass, I am *certain* that many more such landmines will exist throughout the code.  

A quick scan of what I've done so far tells me that the code in question is all about 20 years old and appears to be primarily in the parallel distribution services and topology packages (with one instance in linear algebra services), and the "use &(vector[0]) without checking if the vector is empty" paradigm is all over that code.   

The point being that it would behoove the team to go farther than I can and go hunting them down and fixing them all, even if the test suite passes.  This issue is going to bite you folks someday, even if it's mostly invisible now.  Given that the &(vector[0]) issue was the subject of two internal bug reports (according to the commit message for 92e522b I quoted earlier), it must have bitten you before this somehow.

tvrusso, 2024-11-18:
Quick update:  I've gotten my failure count down into the 80s, but have encountered the first of them that is happening down in a Trilinos function rather than in Xyce:  `Ifpack_CrsRiluk::Factor()`.  I am still using Trilinos 12.12.1 per ancient guidance, but after I track that one down and see if it's fixed in a more recent version, I'm probably not going to dig any deeper into any other failure that is happening inside Trilinos rather than Xyce itself.

tvrusso, 2024-11-18:
> Quick update: I've gotten my failure count down into the 80s, but have encountered the first of them that is happening down in a Trilinos function rather than in Xyce: `Ifpack_CrsRiluk::Factor()`. I [...]

Yeah, the crash inside Trilinos is exactly the same sort of thing, trying to access the pointer to underlying storage of an STL container using &vector[index] when index is out of bounds.   It's in a bit of code related to Epetra, so I presume the Trilinos team will not be interested in any sort of bug report about it.

The failures are coming up in tests in the EMBEDDED directory.

I might never address them and just live without that capability (whatever will I do?!?), or just address them by patching my local copy and maybe forking Trilinos with a bug fix for my use as a submodule for my XyceBundle package.

hkthorn, 2024-11-18:
FWIW, we are refactoring the code this year to use Tpetra and the Tpetra-dependent solver stack.  The Epetra-dependent solver stack is deprecated.  The hope is in the long term that this will not be an issue that needs to be managed.  Thanks again for reporting this issues that will, no doubt, arise when we start developing with and supporting newer compilers.

hkthorn, 2024-11-18:
@tvrusso I have a theory that this is the offending line, Ifpack_CrsRiluk:382:

 int MaxNumEntries = L_->MaxNumEntries() + U_->MaxNumEntries() + 1;

If that line was changed to a "+ 2" instead of a "+ 1", then the InI and InV vectors would be valid for the call on line Ifpack_CrsRiluk:415

tvrusso, 2024-11-18:
nice.   line 415 is indeed where it crashes.  i will check it out

tvrusso, 2024-11-18:
:tada: @hkthorn your one line fix to Trilinos fixed 23 failing tests in parallel on BSD.

The remaining 60 failures are mostly output and error message tests, but there are a few actual computation tests failing I'll have to hunt down.  Hopefully these are just more of the same and local to Xyce.

tvrusso, 2024-11-19:
PR incoming.  Got 'em all nailed. Film at 11.

hkthorn, 2024-11-19:
> For the benefit of other FreeBSD weirdos, the patches to Trilinos 12.12.1 needed to make parallel Xyce not puke on a few netlists in its test suite are here: [trilinos_patch_set.txt](https://github.com/user-attachments/files/17807310/trilinos_patch_set.txt)

Thanks!  I need to submit some fixes to Trilinos to address some errors found during the Tpetra transition effort.  I'll include these in the merge request.

hkthorn, 2024-11-25:
@tvrusso Thank you for your Trilinos patch.  It has been merged into the Trilinos development branch.

tvrusso, 2025-03-24:
Unfortunately, the PR I had submitted to fix this issue didn't actually get merged, and @jdschic closed that original PR when the master branch was re-rooted.  I've submitted a rerun PR.  Hopefully *that* one will get merged.

An enormous number of parallel tests fail on FreeBSD if the fixes in that PR are not applied (and one little fix in Trilinos, that @hkthorn has applied on its development branch, but which has to be applied manually to 12.12.1).

tvrusso, 2025-03-24:
As always, a mail-formatted patch that can be applied using "git am" for the PR I just re-submitted can be had at https://github.com/Xyce/Xyce/pull/136.patch

rich1149, 2025-03-24:
I downloaded your patch.  Tested it out and merged it in.  You should see it on the next push out to GitHub

closed 2025-03-31 by jdschic