Xyce/Xyce #117: Can't compile Xyce because Trillinos packages are deprecated

Kreijstal, 2024-11-17:
-- Looking for Trilinos
   Required packages:
        Amesos Epetra EpetraExt Ifpack NOX Teuchos Sacado
        Triutils AztecOO Belos TrilinosCouplings
   Optional packages:
        Isorropia Zoltan ShyLU ShyLU_DDCore Amesos2 Stokhos ROL MKL
-- Enabled Kokkos devices: SERIAL
CMake Error at cmake/tps.cmake:80 (find_package):
  Found package configuration file:

    E:/msys64/ucrt64/lib/cmake/Trilinos/TrilinosConfig.cmake

  but it set Trilinos_FOUND to FALSE so package "Trilinos" is considered to
  be NOT FOUND.  Reason given by package:

  ERROR: Could not find component 'Amesos'!

  ERROR: Could not find component 'EpetraExt'!

  ERROR: Could not find component 'Ifpack'!

  ERROR: Could not find component 'Triutils'!

  ERROR: Could not find component 'AztecOO'!

Kreijstal, 2024-11-17:
Epetra has to be ported to  Tpetra

tvrusso, 2024-11-17:
The Xyce team is well aware of this issue and is actively working to port Xyce to tpetra.  It is an enormous effort to do so, and it will be quite some time before the team is done.

Until it is done, one must follow the team's instructions for how to build Trilinos for use with Xyce, which either requires building the deprecated packages or building the recommended, older version of Trilinos.

Kreijstal, 2025-01-18:
duplicate of https://github.com/Xyce/Xyce/issues/102
I created this because at the time issues dissapeared, I guess they magically reappeared

closed 2025-01-18 by Kreijstal