Xyce/Xyce #121: Compilation error due to `N_NLS_NOX::PseudoTransientBased`

adrienluitot, 2024-12-22:
Hello everyone,

I am try to build Xyce 7.8 on ArchLinux, I followed the building guide (I might have missed something).

I built Trilinos 12.12.1 without a problem using the given `reconfigure` script (tried for both Serial and Parallel).
I didn't change anything unless the Trilinos path.

But when I compile Xyce, using the given configure command, I get this error (both with Serial and Parallel):
```
../../../src/NonlinearSolverPKG/N_NLS_NOX_Interface.C: In member function 'int Xyce::Nonlinear::N_NLS_NOX::Interface::pseudoTransientSolve(Xyce::Nonlinear::N_NLS_NOX::ParameterSet*)':
../../../src/NonlinearSolverPKG/N_NLS_NOX_Interface.C:1224:20: error: invalid new-expression of abstract class type 'Xyce::Nonlinear::N_NLS_NOX::PseudoTransientBased'
 1224 |         maxStepSize));
      |                    ^
In file included from ../../../src/NonlinearSolverPKG/N_NLS_NOX_Interface.C:58:
../../../src/NonlinearSolverPKG/N_NLS_NOX_PseudoTransientSolver.h:67:9: note:   because the following virtual functions are pure within 'Xyce::Nonlinear::N_NLS_NOX::PseudoTransientBased':
   67 |   class PseudoTransientBased : public NOX::Solver::Generic {
      |         ^~~~~~~~~~~~~~~~~~~~
In file included from /home/user/XyceLibs/Serial/include/NOX.H:53,
                 from ../../../src/NonlinearSolverPKG/N_NLS_NOX_ParameterSet.h:55,
                 from ../../../src/NonlinearSolverPKG/N_NLS_NOX_Interface.h:50,
                 from ../../../src/NonlinearSolverPKG/N_NLS_NOX_Interface.C:57:
/home/user/XyceLibs/Serial/include/NOX_Solver_Generic.H:113:39: note:     'virtual NOX::StatusTest::StatusType NOX::Solver::Generic::getStatus()'
  113 |   virtual NOX::StatusTest::StatusType getStatus() = 0;
      |                                       ^~~~~~~~~
```

Does anyone have a hint in how to fix this?

PS: I will try the same procedure on Ubuntu 24.04, and will keep you in touch on how it's going.

tvrusso, 2024-12-22:
I'm at a loss to explain this myself, given that the getStatus method of the PseudoTransientBased class gets declared for that class around line 50 of `N_NLS_NOX_PseudoTransientSolver.h` and implemented around line 150 of `N_NLS_NOX_PseudoTransientSolver.C`.  It's hard to guess why it could fail on Arch and not on Ubuntu with the same configure options and compilation options.  The function is clearly defined and the pure virtual function overridden by that definition.

You could help the team diagnose the problem if you can compare other differences between the version of Arch you're using and Ubuntu 22.04, such as what compiler is in use (`gcc --version`).  If Arch is using a very new version of GCC it is possible that could be the source of this problem.   That's the only thing I can guess at without further information, and I have no Arch system to play with for experimenting.

adrienluitot, 2024-12-22:
Thank you for your fast answer!

First, to compile the `N_NLS_NOX_Interface.C` file, they use almost the same command:
```
/bin/sh ../../libtool  --tag=CXX   --mode=compile mpicxx -std=gnu++11 -DHAVE_CONFIG_H    [...Lots of includes...]    '-Wl,-rpath,$ORIGIN/../lib' -O3 
-MT N_NLS_NOX_Interface.lo -MD -MP -MF .deps/N_NLS_NOX_Interface.Tpo -c -o N_NLS_NOX_Interface.lo ../../../src/NonlinearSolverPKG/N_NLS_NOX_Interface.C
```
The only difference is that my Arch uses `/bin/sh` while Ubuntu uses `/bin/bash`, but I don't think this makes a big difference.

But there is a difference with `mpicxx`: Arch has `g++ (GCC) 14.2.1 20240910` whereas Ubunut has `g++ (Ubuntu 13.3.0-6ubuntu2~24.04) 13.3.0`.
The problem might come from here.

tvrusso, 2024-12-22:
If the same problem exists on Arch with GCC 13, I am at a total loss to explain how the same code can trip a serious C++ error on one system but sail through on a different one.  The code for the derived class clearly does implement the method that your compilation is flagging because it's pure virtual in the base class, so this error is really strange. 

Perhaps someone on the Xyce team can figure it out (I know the team has at least one Arch linux user, but since Arch is not an approved Linux distro inside Sandia, the team really can't support it officially, and that person isn't one who gets on Github to answer questions like this directly).  But I wouldn't count on it until after the winter holidays, when Sandia is shut down until early January.

The team is still very active, but has not pushed its latest efforts to github in a long time.  Xyce is maintained in a completely separate git repository internal to Sandia, and the team must take special steps to transfer its updates to github.  Unfortunately, due to staffing issues, they are not able to take those steps as frequently as I used to do before I retired (I used to update github from the internal repository every 3 weeks or so, but there is no longer anyone at Sandia who is available to do that as often).  They tried to push an update a few months ago and there was a serious problem with that attempt that had to be undone.    They have not tried again since.

Hopefully the 7.9 release will get pushed without too much further delay.  I know the team had it all ready to release back in November, but the github problem they encountered probably has them scrambling to assure that they don't have a repeat of that issue and they are probably understandably gunshy.

tvrusso, 2024-12-22:
Any chance you could post your exact trilinos CMake invocation and exact Xyce reconfigure script?  I know you said you followed the building guide, but there might be some hint in there as to why there's confusion for the compiler in there.  Your config.log file would probably shed some light on the issue, too, though I can't promise I can spot it.

adrienluitot, 2024-12-22:
I agree that it's pretty odd... maybe I missed something like a package, but it's strange that it only happens at this moment, and it's not detected earlier like in a configure scripts.

I understand for the winter holiday, I'm not really in a hurry. It works on one of my system, that means I can already work with that!

Glad to hear the team is still active! 
The GitHub procedure seems to be a pain... why not directly use the GitHub repository? or set-up the GH as a miror? (like Linux or Ngspice do). I mean... I know it's a lot of work, and it seems you were the one working on it...

Sure, I will try to push all the information I have.

adrienluitot, 2024-12-22:
Here are some pastebin:

The `reconfigure` of Trilinos 12.12.1 for Xyce Parallel: https://pastebin.com/JkmBfvzg and its output: https://pastebin.com/cY4GkN2d (I ran it a second time to get the logs, so it might not be 100% accurate)
`config.log` for Xyce 7.8: https://pastebin.com/65i3ViwY

Here I tried to compile everything by using gcc-13, even Trilinos, but nothing has changed. And I had the very same results with mpi (gcc-14) instead.

However I could reproduce another error:
```
  CXX      N_NLS_NOX_Interface.lo
  CXX      N_NLS_NOX_PseudoTransientSolver.lo
../../../src/NonlinearSolverPKG/N_NLS_NOX_PseudoTransientSolver.C: In member function 'virtual void Xyce::Nonlinear::N_NLS_NOX::PseudoTransientBased::init()':
../../../src/NonlinearSolverPKG/N_NLS_NOX_PseudoTransientSolver.C:113:34: error: 'parseObserver' is not a member of 'NOX::Solver'
  113 |   prePostOperator = NOX::Solver::parseObserver(paramsPtr->sublist("Solver Options"));
      |                                  ^~~~~~~~~~~~~
make[3]: *** [Makefile:567: N_NLS_NOX_PseudoTransientSolver.lo] Error 1
```

This error only shows up if I run make with more than 1 job.
I don't know if it's the first error I described that causes this error, or if they are related at all. Maybe it may help.
Just for information, `N_NLS_NOX_PseudoTransientSolver` is compiled right after `N_NLS_NOX_Interface`. But the error of `N_NLS_NOX_PseudoTransientSolver` shows up before (as long as I have more than 1 job).

tvrusso, 2024-12-22:
The reason the team maintains an internal repository instead of hosting it completely on github is that the project history extends back much further than the Xyce team has permission to release (Sandia only gave approval for Xyce to be open sourced in 2013, and the project had been in existence for almost 14 years by then), and so the github repository is based on a fake initial commit from the day the team began hosting the public repo on github.

Also, the github repo is pretty much a one-way repo, unlike many other fully open source projects.   Sandia National Labs has internal uses of Xyce that are inconsistent with the GPL, and so can't just take code from the open source world and use it themselves --- but the team is perfectly happy to *share* their code under the GPL.

There are other drivers for keeping the internal working git repository completely separate from the public facing one, but the fact that the team only has permission to release code dating back no further than 2013 is the big one.  See the log for commit 4a1d7ef.

The team might choose some other technique for sharing the code on github going forward, but the technique we used when I set it up was designed to have minimal impact on the other developers while fully exposing all the code we had permission to expose.  Every other technique we had for keeping those other bits private would have put unacceptable burdens on all the developers, instead of just burdening the one person handling the mirror of the subset of the tree.

tvrusso, 2024-12-22:
Your second error does suggest a reason for this.  (Doing a make with more than one job just means that files that would have been compiled after `N_NLS_NOX_Interface.C` in a strictly serial make get built at the same time as that file, and so you see more errors.

The code complained about in the second error is inside an `#ifdef Xyce_NOX_SOLVERSTATS`, which is only defined if configure detects a more recent version of Trilinos than 12.12.1. 

Somehow, configure is detecting that Trilinos has the NOX `NOX_SolverStatistics.hpp` header, and setting `Xyce_NOX_SOLVERSTATS` based on that location.  However, Trilinos 12.12.1 does NOT have this header, and configure shouldn't have found it.

Something is very confused, and I suspect that you have some second version of Trilinos installed somewhere on your system that is mucking up Trilinos feature detection.  That could also have mucked up the virtual override thing, because when `Xyce_NOX_SOLVERSTATS` is defined, `getStatus` is a "const" pure virtual method (as it is in later versions of Trilinos), but when it's not defined it's not a const method (as it is not in Trilinos 12.12.1).  Thus, the function defined in Xyce doesn't match the prototype of the pure virtual function in the base class of `NOX::Solver::Generic` and so doesn't override it, leaving the base class method unimplemented.

Try to do a `find /usr -name NOX_SolverStatistics.hpp` and see what it comes up with.  I bet you'll find that some system package has installed Trilinos headers somewhere in /usr/include or something, and configure is picking that up before the ones in your ARCHDIR.

tvrusso, 2024-12-22:
FWIW, your config.log does show configure finding `NOX_SolverStatistics.hpp`, which it shouldn't have if all you had installed was Trilinos 12.12.1.

Since you DO have Trilinos 12.12.1 installed locally in your home directory, it must be picking up relics of a second install from elsewhere.
[image]

adrienluitot, 2024-12-23:
Okay I understand a little better, it seems very complex... The team could fork the project and work on the fork, so there is no link anymore with Xyce and they would be free to do anything! 🙃 I'm kidding, I know this wouldn't change anything.
But yes indeed the GPL3 is quite restrictive, wouldn't the GPL2 or MIT suit the project better?
Anyway, as long as we have an access to Xyce and its updates, it's already awesome!

You really know your stuff! I feel dumb, but yes, I first installed one of the latest Trilinos Version, I thought I cleaned up everything, but apparently not... I can't assure the problem comes from that because I don't have access to my computer to verify right now, but it is very likely.
I'll search for those left overs and clean them up, it must be the cause. I'll get back to you in few days.

tvrusso, 2024-12-23:
Yep, if you installed a newer version of Trilinos into your "ARCHDIR" first, then just reinstalled 12.12.1, it would have left files in there that are new in the newer version, and that would completely confuse configure (which performs various test to see what version of Trilinos you have based on what files it can compile against).   

The best approach would be to delete the ARCHDIR directory entirely and reinstall a fresh 12.12.1 into an empty archdir.  I bet that will fix your problem.

tvrusso, 2024-12-23:
And a «joyeux Noël» to you.

adrienluitot, 2024-12-23:
I think I did worst and installed Trilinos in the /usr… anyway I will have to do a better clean up…

Thank you, joyeux Noël to you too!

adrienluitot, 2024-12-28:
Hello!
I hope you had a great Christmas!

I couldn't find the particular file you mentioned anywhere on my computer, but I did found some leftovers of my recent Trilinos install in my `/usr`. So I made a cleanup and deleted those files.

Anyway, it worked! I could compile Xyce 7.8!

Sorry for the time I stole you... and thanks a lot for your help!
At least I hope this might be useful for anyone in the future...

closed 2024-12-28 by adrienluitot

tvrusso, 2024-12-28:
That's because I got the name of the file wrong, it was `NOX_SolverStats.hpp` not `NOX_SolverStatistics.hpp`.  Sorry.   

Glad you got it working.