Xyce/Xyce #130: Xyce does not recognize the model card correctly on Ubuntu

jiaxinmax, 2025-01-19:
**Problem Description**:Xyce does not recognize the model card when running the simulation on Ubuntu, causing the simulation to fail. The same netlist file works fine on Windows. The error message on Ubuntu reads as follows:
***** Executing netlist EKV.cir

***** Reading and parsing netlist...
Netlist error in file EKV.cir at or near line 2
 Model is required for device M1 and no valid model card found.
Simulation aborted due to error.  There are 0 MSG_FATAL errors and 1 MSG_ERROR
 errors
*** Xyce Abort ***

**Steps to reproduce:**
1. Use the netlist in the attachment
2. Run the command: /home/user/XyceInstall/Serial/bin/Xyce -l EKV.log -r EKV.raw EKV.cir

**Expected behavior:** The simulation should complete successfully, without errors, and generate a EKV.raw data file.
[](url)
**Actual behavior:** Xyce terminates abnormally with the error message:
Model is required for device M1 and no valid model card found.

Hello, thank you for reviewing and addressing this issue!

[EKV.zip](https://github.com/user-attachments/files/18470918/EKV.zip)

tvrusso, 2025-01-20:
EKV 3.01 is only available in the "Non Free" version of Xyce, which is only available in binary form.  The Windows binary you can download from the Xyce is the non-free version.

The open source version of Xyce cannot include the EKV 3.01 model for licensing reasons.   EKV 3.01 is licensed to the Xyce team under a restrictive license that forbids the distribution of its source code.  The Xyce team doesn't provide binaries for Ubuntu, so you must be running the open source version built from the Github source.  This is why your Windows run works and your Ubuntu run doesn't.

The EKV 2.6 model is open source.

There is no way to fix this until EKV 3.02 (which has been open sourced) is imported into Xyce.

jiaxinmax, 2025-01-20:
Thank you for the detailed explanation and clarification!