Xyce/Xyce #132: The. raw file format generated by Xyce is abnormal, causing ReadRaw parsing to fail (block size mismatch)

jiaxinmax, 2025-02-06:
Hello, when I simulate using Xyce 7.8 on Ubuntu, the generated. raw file cannot be parsed by ReadRaw and reports an error of block size mismatch. The specific error message is:
Error in calculating block size. Expected 32 bytes, but found 60 bytes.

**Reproduction steps:**
Run the simulation using Xyce on Ubuntu and generate a. raw file.
Attempt to read the. raw file using ReadRaw.
Failed to read, indicating block size mismatch error.

**Analysis and Discovery:**
ReadRaw can successfully parse. raw files generated by other tools:
The. raw files generated by Ngspice and LTspice can be parsed by ReadRaw normally, indicating that ReadRaw's parsing ability is not a problem.
LTspice also uses binary format:
Although both LTspice and Xyce store. raw files in binary format, ReadRaw can successfully parse LTspice files but cannot parse Xyce files, indicating that the problem lies in the. raw file format generated by Xyce.
Possible format incompatibility or errors:
Due to incompatibility or formatting errors between the. raw file generated by Xyce and the standard format, ReadRaw cannot parse correctly.

**Conclusion:**
The root cause of the problem may be that the. raw file format generated by Xyce does not fully comply with the specifications supported by ReadRaw, or the file contains incompatible additional content. This makes ReadRaw unable to parse the. raw files generated by Xyce.

**Suggestions and Expectations:**
I hope Xyce can confirm the issue and fix the related problems with the. raw file format, or provide more detailed file format instructions for compatibility adaptation.

[error.zip](https://github.com/user-attachments/files/18685725/error.zip)

Karlsefni2012, 2025-02-06:
Thanks for reporting this.  

I am not familiar with the "ReadRaw" tool.  Assuming it is open-source can you provide a link to it?

In general, we attempted to make Xyce's raw file format match that of other simulators.   But the format isn't supported yet for every type of analysis.   Sensitivity analysis, for example, doesn't yet support it for various reasons.

kev-cam, 2025-02-06:
Architecturally, you want to break out things like file format readers & writers and have them as DLLs or scripts so that they're easier to fix independently.

E.g. I think a lot of people would like LTspice/Qspice compatibility, and would be able to work on that independently if it isn't hardwired into Xyce.

Karlsefni2012, 2025-02-06:
That is a good point.  And, actually for file output Xyce has a feature like this.   At the moment the only way to invoke it is via the "Xyce as a library" option for building Xyce.       But for that variation of Xyce, you can create a C++ output class which is derived from a base class, and pass that class in thru the API.   Several people have used this to create custom outputters.

Karlsefni2012, 2025-02-06:
For reading file input (like netlists), this is something we need to do, but it hasn't happened yet.

kev-cam, 2025-02-07:
If anyone wants to go the Cygnus route and get paid for maintaining a commercially supported version, let me know.

Killing off HSpice, Spectre, LTspice, SIMPLIS, Qspice, GNUcap, and various other unnecessary simulators is one of my goals in life, creating the Verilog-AMS standard didn't make much of a dent.

tvrusso, 2025-02-07:
One further comment:  if Xyce is invoked with "override raw file" (as the user clearly must be doing) as "Xyce -r xyce.raw PMOS.net" you get the CSV output stuffed into the rawfile improperly.

If, however, the two .print lines are changed to specify exactly where the output should go and in what format and Xyce invoked without the "-r xyce.raw" option, Xyce's raw file is fine, it just doesn't contain the sensitivity output (because, as Eric notes, Xyce's raw file outputter has not been extended to support sensitivity):

```
spice
.options ksparse
V1 1 0 PWL 0 0 1 1
R1 1 2 1
R2 2 0 1
M1 2 1 0 0 PMOS
.model PMOS PMOS (W=1.5u L=1u)
.TRAN 0 1
.OPTIONS OUTPUT initial_interval=0.1
.sens objfunc={V(2)} param=R2:R
.print TRANADJOINT FORMAT=CSV FILE=tranadj.csv V(1) R1:R R2:R
.options SENSITIVITY STDOUTPUT=1 adjoint=1 direct=0
+ adjointTimePoints=0.25,0.5,0.75
.PRINT TRAN FORMAT=RAW file=xyce.raw R1:R R2:R V(1) V(2)
.end
```

Karlsefni2012, 2025-02-07:
Thanks @tvrusso for your observations.   Based on this, the first, easiest thing to do is to ensure that unsupported analyses get excluded from raw files.    That's pretty straightforward.

The next, less easy thing (though not that hard) would be to add raw support to .SENS.

Regarding why raw_read doesn't like Xyce's text-formatted raw files, I'll defer that question for later.  

Note we have some of our own python-based file IO tools (including the ability to read raw files) in the Xyce test repository.  These were mostly created for testing different output file formats.

jiaxinmax, 2025-02-08:
Thank you all for your valuable responses！

Karlsefni2012, 2025-02-08:
Hello again; I have managed to fix the most immediate issue, which was to ensure that an unsupported analysis is not included in a raw file.

For formats specified on the PRINT line, transient adjoints were already automatically excluded (and the format resets to STD, or *prn).

However for raw files specified via "-r", that involves a different path through the output code, and this case wasn't getting caught.   It was pretty easy to fix this.  It was a one-line change.  The change is in the N_IO_OutputMgr.C file, in the OutputMgr::parsePRINTBlock function, around line 1640 in my local copy.

This if statement needs to be changed from:

```
  // -r output is not made for these formats
  if ( (print_type == PrintType::SENS) || 
       (print_type == PrintType::HOMOTOPY) || 
       (print_type == PrintType::SPARAM) ||
       (print_type == PrintType::ES) ||
       (print_type == PrintType::PCE) )
  {
    print_parameters.formatSupportsOverrideRaw_= false; 
  }
```

to this:

```
  // -r output is not made for these formats
  if ( (print_type == PrintType::SENS) || 
       (print_type == PrintType::HOMOTOPY) || 
       (print_type == PrintType::SPARAM) ||
       (print_type == PrintType::ES) ||
       (print_type == PrintType::PCE) ||
       (print_type == PrintType::TRANADJOINT) )
  {
    print_parameters.formatSupportsOverrideRaw_= false; 
  }
```

I probably never noticed this myself because, even though I often use sensitivity analysis, I don't use the "-r" option very often in my own work.

Anyway, this change will get into our internal repo immediately.    We've started doing regular synchronizations again with github, but I'm not sure how soon the next one will be.    But when it happens this change will be in the code.

Karlsefni2012, 2025-02-12:
Also, the "extra text in the raw file" problem has been fixed, at least in our internal git repo.   Once it is synchronized that bug fix will be in this repo as well.

tvrusso, 2025-03-24:
For what it's worth, this got fixed in commit 4a1ed0, which was pushed today.  This issue wasn't mentioned in its commit log as "Closes GH-132" and so the issue didn't get closed automatically when the commit was pushed.

I do not have permission to close this issue myself, so maybe @Karlsefni2012 can do it?

closed 2025-03-26 by Karlsefni2012

Karlsefni2012, 2025-05-08:
> Spicelib maintainer here.
> 
> spicelib now has support for xyce files (in a separate branch at the moment, will be merged soon.)
> 
> The main problem WAS the extra text stuff behind the binary part (plus a small issue for the ascii files). That is solved now (the text is ignored) for spicelib, but others will likely still have the issue.
> 
> The main problem that remains now is that the xyce simulator does not tell that it is xyce that wrote the raw file. ngspice had the same problem not too long ago. To solve that, please add a "Command:" line with "xyce" and version.
> 
> And telling who you are is not only nice but will also help detecting quirks and needed different interpretations.
> 
> Others say:
> 
> ```
> Command: Linear Technology Corporation LTspice
> Command: ngspice-44.2, Build 
> Command: QSPICE64, Build Feb 11 2025 08:06:48
> ```
> 
> whatever, but please announce yourself.

Hello, I wanted to mention that (as it turns out) this feature is actually already available in Xyce's raw files.  However, it currently must be turned on from the netlist.     If you add the following to the netlist:
```
.OPTIONS OUTPUT  outputversioninrawfile=true
```

The text rawfile header will have an extra line which says something like:
```
Version: Xyce Release 7.9.0 
```

I would have mentioned this earlier, but embarrassingly, I didn't remember that this feature had been added to Xyce.   Today I was revisiting this issue and rediscovered that code was already present.

The feature was added back in 2012, and was briefly a non-optional part of the Xyce raw file header.  In 2014, it was made to be optional, as we wanted (at the time anyway) for the Xyce raw file format to be as close to that of spice3 as possible.   From what I've gathered nutmeg would not read in the Xyce raw file, unless it excluded this "Version:" field.

Anyway, FYI.

Karlsefni2012, 2025-05-08:
One final comment.  I just tested Xyce's rawfiles in the spice3 nutmeg.   

When Xyce's `Version:` field is present, nutmeg rejects the file and exits.  However, if I replace `Version:` with `Command:`,  (like you use in the examples for other simulators) then nutmeg only emits a warning.   But it loads the data and proceeds.   So, it seems that `Command:` is a better choice.    I'll see about changing that.

tvrusso, 2025-05-08:
Just one little comment here.  The "Version:" thing had been added to Xyce in support of IC-CAP integration, which apparently needed it, and it was mistakenly stated in the commit that "Version" was part of the standard (there was even a comment that it was being added for "full compatibility" with SPICE rawfile).   As you note, it was made optional because that addition turned out to make the rawfile incompatible with SPICE3F5's rawfile format, and there was some SPICE-compatible tool or other (in addition to nutmeg) that rejected the files that way.

I suspect that IC-CAP integration is no longer a priority, so it makes a lot of sense to retarget this feature at tools that are actually in current use at Sandia and elsewhere and to emit "Command:" instead.  Or leave it optional for "Version:" to maintain possible IC-CAP compatibility and emit "Command:" unconditionally.