Xyce/Xyce #133: PWL bridges for mixed signal (co-simulation)

kev-cam, 2025-02-06:
It would be nice if someone added my old PWL bridge code (or something similar) -

https://github.com/kev-cam/xyce-top

https://cameron-eda.com/2020/06/03/rolling-your-own-ams-simulator/

If anyone wants to work on "federated simulation", with Xyce as the main framework, let me know.

https://chipletsummit.com/proceeding_files/a0q5f0000044zma/20240206_PreConG_Cameron.PDF

There's stuff in here on doing spectral modeling (with antennas) for RF -

https://docs.google.com/document/d/1PYPo_ZAiOtVXl53TDIBxyPwWoXFiwSpSYHCl3LPexXU/edit?usp=sharing

- that's stuff where you would use AI to create the RF models from the analog behavior (seen in Xyce).

Karlsefni2012, 2025-02-06:
Or this (same project):  https://arxiv.org/pdf/2407.20537

kev-cam, 2025-02-06:
The PWL bridge is just a tweak of the existing PWL to do call backs into external (dynamically loaded) C/C++ to get the actual waveform rather than having a fixed table. I've done it on SPICE3,  GnuCap and Xyce, it's just a few days of effort from scratch.

There's an old "embedded" branch of iVerilog that it works with, I'll do Verilator if there's interest.

I did parallel Verilog simulation with VCS back in ~ '96 (VCS-MT on SMP), and got ~ 4x speed-up, VCS-Cheetah is the second rewrite of that and gets 5x max.

The performance limitations for something like Switchboard (for RTL) are due to the simulation platform, that will be the 5x on SMP, worse networked.

Tasks like analog simulation work better because the ratio of compute-to-communicate is better for mixed-signal simulation, but, as has been pointed out, Xyce does not work well parallel on small simulations where (OpenMPI) communication dominates.

The Switchboard approach is relying on "connecting latency-insensitive interfaces" to avoid the penalties of the network communication, which isn't a general solution, and it's easier to identify with HLS level sims using SystemC-TLM. I tried adding support for "channels" to use instead of "wires" in SystemVerilog to make it easy, but that got rejected (way back), which is why I did this project -

http://parallel.cc

Any of the in-memory RISC-V AI accelerators should be able to speed up circuit simulation without modifying code using this technique -

https://www.youtube.com/watch?v=Bh5axlxIUvM

Parallel simulation throughput is highly dependent on inter-processor latency, that's generally bad with Cloud computing and OpenMPI, but if anyone is keen to build Silicon I have the fix -

https://patents.google.com/patent/US20230336198A1/en

The piece that people usually get wrong with parallel digital simulation is that they communicate resolved values for signals, rather than driver values, and nobody (else) has worked out how to mix driver types on a wire. My proposals for _**Federated simulation**_ are about fixing that stuff in the federation layer, rather than waiting for folks to fix it in the individual simulators - VHDL has been around for 30+ years and has never done it; I'm not expecting the IEEE/Accellera efforts to do anything.