Xyce/Xyce #134: problem to parse m as subckt parameter for multiplier

dwarning, 2025-02-10:
In following circuit Xyce (v7.9) did not recognize subckt parameter "m":

psp103 nch transfer
vd  d 0 dc 0.1
vg  g 0 dc 0.0
vb  b 0 dc 0.0
XPSP103_VA d g 0 b psp_sub w=10u l=1u m=2
.subckt psp_sub d g s b w=10u l=1u m=1
YPSP103_VA m1  d g s b nch l='l' w='w' m='m'
.ends
.step  vb -3 0 0.5
.dc vg 0 1.5 0.02
.print dc format=gnuplot v(g) i(vd)
.model nch psp103_va type=1
.end

Changing m=1 into mult=1 in subckt parameter list and mult=mult in device instantiation everything is fine. Seems a special handling of single letter "m" in parser stage. 

Can someone confirm this problem.

Sorry for duplicating this from google forum.

tvrusso, 2025-02-11:
I don't think there's a general parser problem here.   Since I don't have your PDK and PSP103 verilog installed, I converted your netlist to use Xyce's built-in PSP 103 model:
```
psp103 nch transfer
vd d 0 dc 0.1
vg g 0 dc 0.0
vb b 0 dc 0.0
XPSP103_VA d g 0 b psp_sub w=10u l=1u m=2
.subckt psp_sub d g s b w=10u l=1u m=1
YPSP103_VA m1 d g s b nch l='l' w='w' m='m'
.ends
.step vb -3 0 0.5
.dc vg 0 1.5 0.02
.print dc format=gnuplot v(g) i(vd)
.model nch psp103_va type=1
.end
```
and it ran just fine with no parse errors.

So whatever is going on, it might have something to do with the PSP103_VA model in the PDK.

tvrusso, 2025-02-11:
Yes, the issue is that the PSP103_VA model in the IHP-OpenPDK repo does not define an "M" parameter, only a "MULT" parameter.   So Xyce will throw an error that M isn't a legal instance parameter for that device.  

If you look at the [README.Xyce](https://github.com/Xyce/Xyce/blob/master/utils/ADMS/examples/psp103/README_Xyce) file in Xyce's `utils/ADMS/examples/psp103` directory, I had to manually modify the PSP103 to add "M" as a synonym for "MULT".

The modification is trivial in PSP103_module.include
```
        `IPRco(MULT           ,1.0        ,""         ,0.0         ,inf         ,"Number of devices in parallel")

// XYCE MOD:  Add aliasparam for M to be a synonym for MULT
        aliasparam M=MULT;
```

The first of these lines is in the [PSP103_module.include](https://github.com/IHP-GmbH/IHP-Open-PDK/blob/main/ihp-sg13g2/libs.tech/xyce/adms/PSP103_module.include) file in the IHP-OpenPDK repo, you would just have to add the second.  Since this directory is already a directory of Xyce-specific Verilog-A, that should be no big deal to add.

Karlsefni2012, 2025-02-11:
Thanks @tvrusso for replying to this. 

In some simulators, the simulator framework automatically adds an "M" multiplier parameter to all (or nearly all) device models.   In that case, it isn't necessary for a model developer to manually implement it.   But, Xyce isn't (at least for now) designed this way.  So, it is necessary for each model to directly support it.

The parser does enforce the "M" multiplier for subcircuits.  So, it is involved in that use case, but only on the subcircuit parameter processing.  If the subcircuit contains device models that don't support "M", then it can't apply it properly.

We've set up support for "M" in most of the relevant models that are built into Xyce.  But models being linked in dynamically is a different matter.

dwarning, 2025-02-11:
No, no. Sorry my error test case in first post was not correct. 

This one brings any time 500uA as maximum independent what you set as m in subckt call:

`
psp103 nch transfer
vd  d 0 dc 0.1
vg  g 0 dc 0.0
vb  b 0 dc 0.0
XPSP103_VA d g 0 b psp_sub w=10u l=1u m=4
.subckt psp_sub d g s b w=10u l=1u m=1
YPSP103_VA m1  d g s b nch l=l w=w mult=m
.ends
.step  vb -3 0 0.5
.dc vg 0 1.5 0.02
.print dc format=gnuplot v(g) i(vd)
.model nch psp103_va type=1
.end
` 
The preview format of the code insertion looks terrible, sorry.
Important is, that the call to VA instance is mult=m.

dwarning, 2025-02-11:
Better to attach the file.

[nmos_id_vg_vb_subckt.txt](https://github.com/user-attachments/files/18756264/nmos_id_vg_vb_subckt.txt)

tvrusso, 2025-02-11:
One last example that works, showing that it appears that the special casing of the "m" parameter for subcircuits may be getting somehow confused.
```
psp103 nch transfer
vd  d 0 dc 0.1
vg  g 0 dc 0.0
vb  b 0 dc 0.0
XPSP103_VA d g 0 b psp_sub w=10u l=1u bleem=4
.subckt psp_sub d g s b w=10u l=1u bleem=1
m1  d g s b nch l=l w=w m='bleem'
.ends
.step  vb -3 0 0.5
.dc vg 0 1.5 0.02
.print dc format=gnuplot v(g) i(vd)
.model nch nmos level=103 type=1
.end
```

This is basically the same as Dietmar's netlist, but with "bleem" instead of "m" as the subcircuit parameter.  Using "bleem" for the subcircuit parameter name works fine and the mosfet behaves as 4 in parallel, but using "m" as the subcircuit parameter name breaks and somehow doesn't get the non-default value communicated to the mosfet.

dwarning, 2025-02-11:
Thanks Tom for tracking that.

Yes - as said, any other name for the multiplier works well. And the "automatic" approach with "m" in subckt instantiation behaves well too.

I think we have to change the parametrization in the IHP PDK.

Karlsefni2012, 2025-02-11:
Ok, I understand now.  I think I can see what the problem is, at least in part.  

The parser handling of subckt "m=" doesn't expect there to be other multiplier variables, such as "mult".  Another case that works is one in which, instead of saying "mult=m", the instance line (the m1 line in Tom's examples) uses "m=m".

tvrusso, 2025-02-11:
But in Dietmar's case, the Verilog-A he's using does NOT have an M model parameter, it only has MULT.

Karlsefni2012, 2025-02-11:
Yes, I understand that.   In the N_IO_DeviceBlock code, which applies subcircuit multipliers, it searches the parameters list of a device for a param named "M".  If it finds it, it applies the multiplier.  If it doesn't, it does nothing.

Karlsefni2012, 2025-02-11:
And since the subcircuit multipliers are treated as a "special" parameter, they aren't applied in the conventional way either.

Karlsefni2012, 2025-02-11:
For the subcircuit multipliers, they are multiplicative down the subcircuit stack.  So, the code sets up a separate expression tree to handle them.

Karlsefni2012, 2025-02-11:
Anyway, this is fixable, but I'll have to make sure I don't break any of the other multiplier use cases.  The hardest part about implementing subcircuit multipliers (especially implicit ones) was making sure every use case was correct.    Clearly, I didn't quite think of all of them.

tvrusso, 2025-03-27:
If I'm not mistaken, this got fixed in commit bae2ee and could be closed.

commit 7b309bf7 (2021-02-08): Replace deprecated mem_fun with mem_fn

The old standard function std::mem_fun creates a wrapper function to a
classes' member function given a pointer to that function.

Use of this feature was introduced into the MeasureManager by Dave
Baur in commit ac5bbebf back in 2013.

mem_fun is in fact deprecated in C++11 and will be removed in C++17.
It has been replaced by the more functional and more flexible mem_fn.

This commit simply replaces mem_fun usage with mem_fn.

Closes #134
https://gitlab-ex.sandia.gov/Xyce/Xyce/-/issues/134
closed 2025-03-28 by Karlsefni2012