Xyce/Xyce #137: Xyce fails to configure with trilinos-16.1.0

yurivict, 2025-03-29:
configure fails:

```
checking for Teuchos_SerialDenseMatrix.hpp... no
configure: error: Unable to compile a test program against Trilinos that includes Teuchos_Serial_DenseMatrix.hpp.  It is possible Trilinos was not properly configured or the environment has changed since Trilinos was installed.  See config.log for more information.
```

config.log has these relevant lines:
```
checking for Teuchos_SerialDenseMatrix.hpp... no
configure: error: Unable to compile a test program against Trilinos that includes Teuchos_Serial_DenseMatrix.hpp.  It is possible Trilinos was not properly configured or the environment has changed since Trilinos was installed.  See config.log for more information.
```

It can't find Kokkos_Complex.hpp

Xyce compiles fine with trilinos-16.0.0

Versions: 7.8.0, 7.9.0
clang-19
FreeBSD 14.2

Karlsefni2012, 2025-03-29:
We are in the process of switching from the autotools (like configure) to CMake.   In the near future the autotools build of Xyce will be officially deprecated.

If you are compiling vs a recent version of Trilinos (more recent than the ancient version 12.12.1) you need to use CMake, not autotools.

The build guide at the Xyce website hasn't been updated for CMake yet.   However, you can find instructions for building Xyce with CMake in the Xyce/INSTALL.md file.

yurivict, 2025-03-29:
I know about the cmake transition.
But in the meantime the autotools build should probably also work until you officially switch to cmake.

tvrusso, 2025-04-03:
So, the Xyce team has already deprecated the autotools build given that Trilinos has deliberately removed stuff (the "Makefile.export" system) that helped Xyce's autotools work out the interdependence of Trilinos librariesowever, since you build Trilinos as shared libraries and the Trilinos CMake system makes sure to link them correctly, this is not such a big issue, as you've found in your FreeBSD port.   It would be a showstopper if the Trilinos libraries were NOT built as shared libraries, but as static libraries.

Not only is it really deprecated, but there's nobody left on the team who will work on autotools, all effort in the build system is being spent on the CMake system.

The Xyce team certainly won't fix up the autotools system anymore, but because I, too, prefer autotools, I was able to make Xyce build against Trilinos 16.1.0 with autotools with some very minor tweaks.  I do not use your ports on FreeBSD, because the way you build and install things makes it very difficult to have both serial and parallel Xyce installed simultaneously (you dump Trilinos headers and libraries directly into /usr/local/include and /usr/local/lib as the building guide strongly recommends against, and which, as you have seen, causes trouble because Trilinos dumps a dummy mpi.h into its include directory which can conflict with MPICH).  However, I was able to build Trilinos 16.1.0 and install it elsewhere just fine on FreeBSD 13.5.

Once Trilinos was installed to $TRILINOS_ROOT (in my case, a directory under my home directory), I was forced to modify Xyce's configure.ac to change the `AX_CXX_COMPILE_STDCXX(11,,mandatory)` call to `AX_CXX_COMPILE_STDCXX(17,,mandatory)`.  This was necessary because leaving it as is forced "-std=gnu++11" into the CXX line, even when I had explicitly put "-std=c++17" into CXXFLAGS.  Without doing that, I could get past the error you're reporting just fine, but something else puked (complaining about "conditional" not being in the std namespace).

Finally, I had to add "-I$TRILINOS_ROOT/include/kokkos" to CPPFLAGS when running configure.  That was sufficient to get past the error you're reporting.   The main problem is that some Trilinos headers access each other without specifying subdirectories, and so headers in these subdirectories are not found.  CMake takes care of that for them, but autotools don't.

For the record, my build of trilinos was done by running:
```
pushd ~/Xyce/trilinos
git checkout trilinos-release-16-1-0
popd

cmake -C ~/Xyce/Xyce/cmake/trilinos/trilinos-base.cmake \
~/Xyce/trilinos \
-DBUILD_SHARED_LIBS:BOOL=ON \
-DCMAKE_INSTALL_PREFIX=~/XyceLibs/FreeBSD13_trilinos16.1.0 \
-DCMAKE_Fortran_COMPILER=gfortran13 \
-DAMD_INCLUDE_DIRS=/usr/local/include/suitesparse 
```

and my build of Xyce with:
```
TRILINOS_ROOT=~/XyceLibs/FreeBSD13_trilinos16.1.0
TRILINOS_LIB=~/XyceLibs/FreeBSD13_trilinos16.1.0/lib
TRILINOS_INCLUDE=~/XyceLibs/FreeBSD13_trilinos16.1.0/include
KOKKOS_INCLUDE=~/FreeBSD13_trilinos16.1.0/include/kokkos
../Xyce/configure \
--prefix=~/XyceInstalls/Xyce_Serial_t16.1 \
CXXFLAGS="-O3 -DNDEBUG -pipe -Qunused-arguments -Wl,-rpath,$TRILINOS_LIB" \
CFLAGS="-O3 -DNDEBUG -pipe" \
CPPFLAGS="-I$TRILINOS_INCLUDE -I$KOKKOS_INCLUDE -I/usr/local/include -I/usr/local/include/flex " \
LEX=/usr/local/bin/flex \
LDFLAGS="-L$TRILINOS_LIB -L/usr/local/lib -Wl,-rpath=/usr/local/lib/gcc13" \
CXX="clang++ -std=c++17" \
F77="gfortran13" \
--enable-stokhos \
--enable-amesos2 \
--enable-shared \
--enable-xyce-shareable 
```

This only works IF you have patched Xyce's configure.ac as I mentioned, so it doesn't force "-std=gnu++11" into CXXFLAGS.  That apparently overrides the "-std=c++17" if it's given.

You can see I had to force the kokkos subdirectory into the include search path in CPPFLAGS, and also forced the lib directory into LDFLAGS.  I also had to add an RPATH to the binary (in CXXFLAGS) because I'm not using a system directory for the Trilinos.

Since autotools support is abandoned in Xyce you can't expect that they'll fix it. 

This is about the only thing you can do to make it work --- you must stuff appropriate directories into the compiler and loader flags with CXXFLAGS, CPPFLAGS, and LDFLAGS on the configure line.  It is ugly, but it does in fact work.  I do find that the Xyce it creates crashes on one test case in the Xyce regression suite, but that's another matter for another day --- that test is of error messages, and something screwy is going on with it.

How you adapt this approach to your Xyce port on FreeBSD is another matter, but if you really want to use autotools and Trilinos 16.1, you'll have to do something like that.

yurivict, 2025-04-03:
I've solved this problem by adding these 2 lines to the FreeBSD port:
```
CFLAGS+=        -I${LOCALBASE}/include/kokkos # https://github.com/Xyce/Xyce/issues/137
CXXFLAGS+=      -I${LOCALBASE}/include/kokkos # https://github.com/Xyce/Xyce/issues/137
```

yurivict, 2025-04-03:
But we will switch to cmake, it's just a matter of some time expenditure.

yurivict, 2025-04-03:
Thanks for your efforts to fix this problem.
Since we are satisfied now and Xyce has moved on and won't fix anything there are no action items,
and I'll close it for now.

closed 2025-04-03 by yurivict