Xyce/Xyce #138: Error in level 2 diode VJ and CJ0 calculations?

smindinvern, 2025-03-31:
There appears to be a bug in the temperature-adjusted `VJ` and `CJ0` calculations for the level 2 (PSpice) diode model.  I don't have access to the 1998 PSpice user's guide referenced in the Xyce Reference Guide, but the 2000 version of that document gives the following equations for `VJ(T)` and `CJO(T)`:

    VJ(T) = VJ·T/Tnom - 3·Vt·ln(T/Tnom) - Eg(Tnom)·T/Tnom + Eg(T)
    CJO(T) = CJO·(1 + M·(.0004·(T-Tnom)+(1-VJ(T)/VJ)))

As far as I can tell, these match the equations given in the Antognetti and Massobrio text, so I'm not quite sure why there is a separate code path for the level 2 model?

The functions `Xyce::Device::Diode::Instance::updateTemperature()` and `Xyce::Device::Diode::updateTemperature()` (`src/DeviceModelPKG/OpenModels/N_DEV_Diode.C`) have the following implementation:

```c++
tJctPot = (model_.VJ - egfet1)*fact2 - 3*vt*log(fact2) + egfet;
tJctCap = model_.CJO/(1.0 +
                      model_.M*(4.0e-4*(Temp-TNOM) + (1-tJctPot/model_.VJ)));
```

As far as I can tell, the first line is equivalent to:

    tJctPot = (VJ - Eg(TNOM))*(T/TREF) - 3*Vt*ln(T/TREF) + Eg(T);

The difference from the PSpice user's guide here being the appearance of `T/TREF` rather than `T/TNOM`.  The second line substitutes a division in place of a multiplication, causing junction capacitance to increase when it should decrease and vice versa.

I don't have a background in semiconductor physics or modeling, so if anyone who's more familiar with this area could take a look I'd appreciate it.  For reference, the following patch builds properly and brings the computed results in-line with my expectations.

```
From a66cab3db5b18d8ed9460b392b4b40f49f4db9e2 Mon Sep 17 00:00:00 2001
From: Nickolas Lloyd <smindinvern@users.noreply.github.com>
Date: Mon, 31 Mar 2025 09:41:40 -0400
Subject: [PATCH] Fix level 2 diode temperature-adjusted VJ and CJ0 calculation

---
 src/DeviceModelPKG/OpenModels/N_DEV_Diode.C | 8 ++++----
 1 file changed, 4 insertions(+), 4 deletions(-)

diff --git a/src/DeviceModelPKG/OpenModels/N_DEV_Diode.C b/src/DeviceModelPKG/OpenModels/N_DEV_Diode.C
index 42905926..670ef844 100644
--- a/src/DeviceModelPKG/OpenModels/N_DEV_Diode.C
+++ b/src/DeviceModelPKG/OpenModels/N_DEV_Diode.C
@@ -1531,8 +1531,8 @@ bool Instance::updateTemperature( const double & temp_tmp )
   int level = model_.getLevel();
   if(level == 2)   // this section is PSPICE compatible
   {
-    tJctPot = (model_.VJ - egfet1)*fact2 - 3*vt*log(fact2) + egfet;
-    tJctCap = model_.CJO/(1.0 +
+    tJctPot = (model_.VJ - egfet1)*(Temp/TNOM) - 3*vt*log(Temp/TNOM) + egfet;
+    tJctCap = model_.CJO*(1.0 +
                           model_.M*(4.0e-4*(Temp-TNOM) + (1-tJctPot/model_.VJ)));
   }
 
@@ -2205,8 +2205,8 @@ bool updateTemperature
 
   if(level == 2)   // this section is PSPICE compatible
   {
-    tJctPot = (VJ - egfet1)*fact2 - 3*vt*log(fact2) + egfet;
-    tJctCap = CJO/(1.0 +
+    tJctPot = (VJ - egfet1)*(Temp/TNOM) - 3*vt*log(Temp/TNOM) + egfet;
+    tJctCap = CJO*(1.0 +
                           M*(4.0e-4*(Temp-TNOM) + (1-tJctPot/VJ)));
   }
 
-- 
2.39.5 (Apple Git-154)
```

tvrusso, 2025-03-31:
The tJctCap calculation in Xyce's level 1 diode was taken exactly from the SPICE3F5 diode.  The code in SPICE3F5 is:

```
            here->DIOtJctCap = model->DIOjunctionCap/
                    (1+model->DIOgradingCoeff*
                    (400e-6*(model->DIOnomTemp-REFTEMP)-gmaold) );
```

This matches the similar code in the much older Fortran version, [SPICE-2](https://github.com/ethanmancini/SPICE-2) around line 439 of `spice.f`:
```
      VALUE(LOCV+5)=VALUE(LOCV+5)/(1.0D0+VALUE(LOCV+7)
     1     *(400.0D-6*(TEMP1-REFTMP)-GMAOLD))
```

The code for level 1 in Xyce is:
```
  tJctCap = model_.CJO/
            (1.0+model_.M*(4.0e-4*(model_.TNOM-CONSTREFTEMP) -gmaold));
```
It has been that way since 2001.

ng-spice also has exactly this expression.   

Yes, it does appear to disagree with equation (1-50) of Massobrio and Antognetti on page 30, which has the parenthetical expression multiplying CJO.  However, it is clear that SPICE itself never matched equation (1-50) of that book.

Xyce's level 2 diode was once intended to be "pspice compatible" but never really was, and the developer who wrote it had simply cribbed some equations out of the PSPICE manual and allowed Xyce not to choke on parameters that PSPICE had added to the diode model (and in several cases, simply *ignored* those parameters instead of bombing with an "unrecognized parameter" error).   The level 2 diode was completely revamped several years ago, and very few level==2 conditionals remain (c.f. commit babef87 and those that followed it).  Most of the supposed "pspice compatible" stuff was removed, as it turned out to be the same as the extended ngspice-compatible stuff that was added to the level 1 diode in subsequent commits.

In fact, the single block of level==2 conditional you're noting is the only remaining difference between level 1 and level 2, and it was maintained solely for the benefit of the small handful of Sandia internal users who had netlists that used the level 2.

It is entirely possible that the correct fix here is to remove the level 2 completely.

However, the level=1 code that implements the junction capacitance temperature correction is consistent with every version of SPICE dating back to the early 1980s.

smindinvern, 2025-03-31:
> The code for level 1 in Xyce is:
>
> ```
>   tJctCap = model_.CJO/
>             (1.0+model_.M*(4.0e-4*(model_.TNOM-CONSTREFTEMP) -gmaold));
> ```
>
> It has been that way since 2001.
>
> ng-spice also has exactly [...]

Just for clarity, the complete definition is:

```
  tJctCap = model_.CJO/
            (1.0+model_.M*(4.0e-4*(model_.TNOM-CONSTREFTEMP) -gmaold));

  tJctPot = pbfact+fact2*pbo;

  double gmanew = (tJctPot-pbo)/pbo;

  tJctCap *= 1.0 + model_.M*(4.0e-4*(Temp-CONSTREFTEMP)-gmanew);
```

In the case where `TNOM==CONSTREFTEMP` it looks like this simplifies to `tJctCap = model_.CJO*(1 + model_.M(*4.0e-4*(Temp-CONSTREFTEMP)-tJctPot/model_.VJ+1))`, so to me it seems like the code is in agreement with the book.

> It is entirely possible that the correct fix here is to remove the level 2 completely.

Sounds right to me :+1:

tvrusso, 2025-03-31:
Yeah, the equation (1-50) in the book isn't a direct "junction cap at t = cj0*factor", it's the relation between junction cap at t1 and junctioncap at t2.   So in this case, t1 is TNOM and t2 is the desired temperature.

At any rate, your patch isn't a fix.

Whether the lingering remnants of the level 2 "pspice-compatible" diode in Xyce are worth keeping is a different question.    That old code dates back to 2005 and its developer hasn't been with the Xyce team since around 2006.

There's a reason most of it got ripped out.

smindinvern, 2025-03-31:
> Yeah, the equation (1-50) in the book isn't a direct "junction cap at t = cj0*factor", it's the relation between junction cap at t1 and junctioncap at t2. So in this case, t1 is TNOM and t2 is the desired temperature, and there are a few extra steps to get at the complete correction.

Could you point me at those extra steps?  It's not clear to me what the difference is.

> At any rate, your patch isn't a fix.

That's fine, but I'm confused as to what exactly the error is.  Would you be willing to elaborate?  Are you saying that the current level 2 code is actually correct?

Thanks!

tvrusso, 2025-04-01:
Actually, I'm not really saying anything about the level 2 code.  I think it's probably more wrong than your fix suggests.

I'm more waving my arms about why the level 1 code doesn't look like the Massobrio and Antognetti equation (1-50), and that the correct PSPICE compatibility would probably look more like the level 1 code with some tweaks.

One thing that we found early in the Xyce project is that while all the equations and physics in Massobrio and Antognetti are correct and are guides to how to understand how the various parameters go into the models, it often doesn't help when actually implementing the devices, as the code invariably winds up being more convoluted than the expressions in the book suggest.  And similarly, equations spotted in the user manuals of commercial codes are often as misleading.

Honestly, I'd just recommend ignoring the Xyce level 2 diode, which was sort of a clumsy hack to begin with that has never been properly put through all its paces and rigorously tested.  The *only* remaining difference between level 1 and level 2 are the few lines you've spotted.

The level 1, however, has been tested against other codes, notably after inclusion of the sidewall effects.

smindinvern, 2025-04-01:
Ok, thank you, I think we're actually in agreement, then.  The only reason that I spotted this is that the frontend I'm using (Qucs-S) hard-codes `LEVEL=2` in the generated netlist, but now that I know about it I can work around it.

One point about equation (1-50) vs the code: if you go through and substitute all of the variables for their corresponding values in the definition of `tJctCap`, it seems like what it's doing is first computing $CJO(TNOM) \to CJO(TREF)$, and then computing $CJO(TREF) \to CJO(T)$, which explains the division, then multiplication by similar-looking factors.  Does that seem right?  But then I'm not sure why it wouldn't make sense to just directly do $CJO(TNOM) \to CJO(T)$ each time.  I think "if it ain't broke, don't fix it" definitely applies, but it definitely makes me curious.

Thanks for your help!

tvrusso, 2025-04-01:
Yes, your reasoning is exactly what I came to last night while puzzling it over.

SPICE evaluates the temperature correction to go from TNOM to TREF, then from TREF to t.  CJ0 is the value at TNOM, and to get the TREF value one divides by the factor.  It is not at all clear *why* that would be done.  It must have something to do with the interdependence of the junction potential and junction capacitance through "gmaold" and "gmanew", but I haven't worked it through.   But it's been getting done since at least SPICE2.

It is possible that the PSPICE reference manual simplified its discussion and just says they do the TNOM->T conversion directly, and that the developer who hacked the "pspice compatible" level 2 into Xyce didn't work it through and just copied the level 1 code's first line and tweaked it slightly to look a little like the PSPICE manual.   

I can't tell you how much of a disaster that level 2 hack turned out to be.  Years after its author left the project, a customer complained that it wasn't computing the temperature correction to breakdown voltage correctly, and asked us to fix it.  Turns out the original author had left *all* the code that computed the temperature correction commented out, because he didn't have anything to test it against and couldn't figure out how to make it work with SPICE's iterative technique that makes sure the breakdown voltage and breakdown current are consistent with each other when both are specified by the user.

That isn't even the worst of it, but I'll leave it there.

It is for this reason that I really recommend just not using it.  Most of the documentation about this diode level was removed from the reference guide and there's only passing reference to it now.

The reason Qucs-s probably defaults to the level 2 is that until around 2022, Xyce's level 1 was *strictly* the SPICE3F5 compatible diode with no added features.   *some* of the level 2 code implemented extra parameters that ngspice also implemented, and Qucs-s's diode models probably expect.  

However in 2022, in response to numerous internal requirements to support additional diode features like sidewall effects, I went through the level 1 and level 2 diodes, discovered that *ALL* of the duplicated code for level 2 that implemented some of the extra effects reduced to the level 1 anyway when those parameters were left at defaults, and so there was no point in having all that duplicated code and conditionals to select based on level.   So the code was simplified and all those extra parameters (additional breakdown effects, additional emission coefficients, temperature dependence of breakdown, etc) were simply moved into the level 1, where they have no effect if left at their defaults anyway.

When I did that cleanup, I found that the only bits of the level 2 that could not be made to fall back to level 1 just by not setting the additional parameters were the two remaining bits of conditionalized code that you were concerned about.  So I left them there, and moved on with adding other features that apply to both levels.

So there's really no continued reason for Qucs-s to default the the level 2.

smindinvern, 2025-04-01:
Thanks again for the explanation.  If it would be helpful, I'd be happy to submit a PR to remove the remaining level 2 branches.  Otherwise, I'll just make sure to explicitly use level 1 :-)

tvrusso, 2025-04-01:
Well, as I'm retired from the Xyce project my comments about what should be done with the level 2 no longer have force, and the Xyce team can make up its own mind what to do.  Removing those two blocks of conditional code and references to "pspice compatible diode" from the reference guide are straightforward enough tasks for the current members of the team, should they choose to go that route.

Better would be to make a PR with Qucs-s to suggest they remove the level-2 default on Xyce diodes, since everything they really need for Xyce/ngspice compatibility is now in the level 1, and all level 2 does now is confuse the junction capacitance and junction potential temperature dependence.