Xyce/Xyce #139: Xyce crashes and dumps core on one test in test suite on FreeBSD with Clang 19 or Clang 20

tvrusso, 2025-04-06:
Trying to build Xyce against Trilinos 16.1 on FreeBSD 13.5, I find that if I use either Clang 19 (the default compiler for FreeBSD 13.5) or Clang 20 (installed from the ports system), Xyce crashes on exactly one test in the test suite in serial, `Message/Input/DeviceBlock_extractSourceFields_1`.

The crash is down below `Xyce::Util::Param::getMutableValue<double>`, in a system function (`__cxa_free_exception`).

It is happening as a result of this line of code in that Xyce function:
```
          Report::UserError() << "Cannot convert '" << tmp << "' to double for expression " <<  tag_;
```
Unfortunately, I'm unable to do *any* debugging on it, because with the `-O3` optimization level all the relevant variables I could print to figure out what's wrong are optimized away and I can't look at what's happening at all, and if I use `-O0 -fno-inline` to build the code doesn't crash 😦 .

The same problem does not happen if I build Xyce and Trilinos with Clang 18 (from ports) on the same system in exactly the same way with full optimization.

Since I don't have a solution *or* a path forward to finding one, I'm posting it here mostly as a notice that there may be some problem somewhere that is introduced by these two much more recent Clang compilers than the Xyce team uses.  Maybe it's reproducible on some other system than BSD, perhaps even Mac, but who knows.

For the record, I configure and build Trilinos with:
```
pushd ~/Xyce/trilinos
git checkout trilinos-release-16-1-0
popd

cmake -C ~/Xyce/Xyce/cmake/trilinos/trilinos-base.cmake \
~/Xyce/trilinos \
-DBUILD_SHARED_LIBS:BOOL=ON \
-DCMAKE_INSTALL_PREFIX=/users/russo/XyceLibs/FreeBSD13_trilinos16.1.0_clang20 \
-DCMAKE_C_COMPILER=clang20 \
-DCMAKE_CXX_COMPILER=clang++20 \
-DCMAKE_Fortran_COMPILER=gfortran13 \
-DAMD_INCLUDE_DIRS=/usr/local/include/suitesparse 

cmake --build . -j 8 -t install
```

and Xyce with:
```
pushd ../Xyce
git checkout master
popd

TRILINOS_ROOT=/users/russo/XyceLibs/FreeBSD13_trilinos16.1.0_clang20 
cmake  \
-DCMAKE_INSTALL_PREFIX=/users/russo/XyceCrap/Xyce_Serial_t16.1_clang20 \
-DTrilinos_ROOT=$TRILINOS_ROOT \
-DCMAKE_C_COMPILER=clang20 \
-DCMAKE_CXX_COMPILER=clang++20 \
-DCMAKE_Fortran_COMPILER=gfortran13 \
-DCMAKE_EXE_LINKER_FLAGS="-Wl,-rpath,$TRILINOS_ROOT/lib -lexecinfo" \
-DXyce_REGRESSION_DIR=/users/russo/Xyce/Xyce_Regression \
../Xyce

cmake --build . -j 8 -t install
```
"-lexecinfo" is required because something in Trilinos uses `backtrace` and `backtrace_symbols` without searching for a library containing them and linking against it.  These functions live in libexecinfo on BSD.

The crash stack trace is
```
#0  0x000000085198ac00 in ?? ()
#1  0x0000000846705297 in __cxa_free_exception (thrown_exception=0x851a96dc0)
    at /usr/src/contrib/libcxxrt/exception.cc:644
#2  0x00000000023285e8 in double Xyce::Util::Param::getMutableValue<double>() const ()
#3  0x0000000000aa1c03 in Xyce::Device::SinData::SinData(Xyce::Device::DeviceEntity const&, std::__1::vector<Xyce::Device::Param, std::__1::allocator<Xyce::Device::Param> > const&, Xyce::Device::SolverState const&, Xyce::Device::DeviceOptions const&) ()
#4  0x0000000000d5966a in Xyce::Device::Vsrc::Instance::Instance(Xyce::Device::Configuration const&, Xyce::Device::InstanceBlock const&, Xyce::Device::Vsrc::Model&, Xyce::Device::FactoryBlock const&) ()
```
and of course *this* build didn't have -g so doesn't provide the line numbers.  The line number (obtained from a previous build with "-g" in `Xyce::Util::Param::getMutableValue<double>` is line 617, and it's called from line 316 of `Xyce::Device::SinData::SinData`

I have little expectation that anyone's going to figure out why this crashes here unless it can be reproduced on some other system, but I hope that maybe it can be.

Since this crash happens in only ONE of the many test suite tests that check error messaging regarding incorrect netlist input and everything else passes, it doesn't really cause me a real problem, but it suggests some sort of Heisenbug (whether with Xyce, Clang >18, or FreeBSD I can't tell).

tvrusso, 2025-04-06:
Oh, and the output of the test before the crash is:
```
*****
***** Welcome to the Xyce(TM) Parallel Electronic Simulator
*****
***** This is version Xyce DEVELOPMENT-202504061137-(ReRelease-7.9.0-56-g6fd00574a)-opensource
***** Date: Sun Apr 06 11:48:08 MDT 2025

***** Executing netlist DeviceBlock_extractSourceFields_1.cir

***** Reading and parsing netlist...
***** Setting up topology...

Netlist error in file DeviceBlock_extractSourceFields_1.cir at or near line 5
 Device instance V1D: Cannot convert parameter V0 to a numeric value from fred
Netlist error in file DeviceBlock_extractSourceFields_1.cir at or near line 5
 Device instance V1D: Cannot convert parameter VA to a numeric value from (
Netlist error: Cannot convert 'fred' to double for expression V0
```
(note that the Xyce github repo no longer has any valid tag on the current master branch, all the tags are on the old_master branch, so the git describe output in a fresh clone would be null. I created a "ReRelease-7.9.0" tag in my local repo pointing at the new root of the master tree to make mine come out properly)

tvrusso, 2025-04-06:
Ugh.  I see what the problem is.

`getMutableValue<double>` does not initialize `val`, and when there's an error it throws a delayed error message and returns `val`.

Clang 19 and 20 have, apparently, been set up to throw an exception when uninitialized data is returned this way.  

If I initialize `val=0.0` in N_UTL_Param.C on line 598, the code no longer crashes on that one test.

A quick scan of that file shows that NONE of the implementations of getMutableValue initialize the variable they use to return a value, and so if there is an error they will return uninitialized data, and Clang 19 and 20 will helpfully make the program crash instead of doing something informative.

All the getImmutableValue functions *also* have the same problem.

This crash will only show up when the netlist has an error that causes one of the getMutableValue functions to puke on the parameter value it is passed.   And again, didn't happen with any version of Clang prior to 19.

I'm not going to submit a PR for this, but the fix is simple --- just make every one of the getMutableValue and getImmutableValue functions initialize the variable they'll be returning, so it's valid data even if there's an error processing the input.

rich1149, 2025-04-08:
The code has been updated and passed all regression tests.  You should see the update in the next push by the team to GitHub.

tvrusso, 2025-04-14:
This` was fixed by commit e409af6

closed 2025-04-14 by tvrusso