Xyce/Xyce #140: Model Not Found Error for FinFET Inverter Simulation in Xyce 7.8

gowreddysuresh, 2025-04-08:
I'm currently using Xyce 7.8-opensource for a simulation involving FinFET inverter input rise time extraction. However, I’m encountering model-related errors during simulation.

**This is the code:**
.INCLUDE '/home/vlsi/xyce_7.8/7nm_TT_160803.pm'

.global vdd

.param Vdd=0.7  
.param L=21n  
.param NFINn=2  
.param NFINp=3  
.param FO=1

* Load Capacitance at 10th stage (DUT)
Cload out 0 20e-18

* Inverter Subcircuit
.subckt inverter in out VDD GND NFINp NFINn
Mp out in VDD VDD pmos_rvt l=L nfin=NFINp
Mn out in GND GND nmos_rvt l=L nfin=NFINn
.ends

* Power supply
Vdd vdd 0 DC {Vdd}

* Input pulse (Initial Trise = 0)
Vin in 0 PULSE(0 {Vdd} 1n 1p 1p 10n 20n)

* Inverter Chain (FO4 for 10 Stages)
Xinv1 in n1 vdd 0  {NFINp} {NFINn}  inverter M={FO}
Xinv2 n1 n2 vdd 0  {NFINp} {NFINn}  inverter M={FO**2}
Xinv3 n2 n3 vdd 0  {NFINp} {NFINn}  inverter M={FO**3}
Xinv4 n3 n4 vdd 0  {NFINp} {NFINn}  inverter M={FO**4}
Xinv5 n4 n5 vdd 0  {NFINp} {NFINn}  inverter M={FO**5}
Xinv6 n5 n6 vdd 0  {NFINp} {NFINn}  inverter M={FO**6}
Xinv7 n6 n7 vdd 0  {NFINp} {NFINn}  inverter M={FO**7}
Xinv8 n7 n8 vdd 0  {NFINp} {NFINn}  inverter M={FO**8}

* Transient Analysis
.tran 0.01p 100n

* Measure Rise Time at 8th Stage
.measure tran Trise_4 TRIG v(n4) VAL='0.2*Vdd' RISE=1
+                 TARG v(n4) VAL='0.8*Vdd' RISE=1

* Probes
.print tran v(in) v(out)

.end

**Command Used:**
mpirun -np 1 /home/vlsi/XyceInstall/Serial/bin/Xyce -o results trise.cir

**Error Messages**
Netlist error in file trise.cir at or near line 17
 Model is required for device MP and no valid model card found.

Netlist error in file trise.cir at or near line 18
 Model is required for device MN and no valid model card found.

**system details**
 OS: Ubuntu 22.04
Xyce Version: 7.8-opensource

I have added the model in .txt format below:
[7nm_TT_160803.txt](https://github.com/user-attachments/files/19643889/7nm_TT_160803.txt)

Thanks in advance for your help!

tvrusso, 2025-04-08:
There is no level 72 MOSFET in Xyce.  This is not a valid model card.

tvrusso, 2025-04-08:
Just checked, HSPICE's MOSFET level 72 is the BSIM-CMG.  Your model cards all specify the version 107 for the model.

Xyce supports BSIM-CMG, but not the way HSPICE does, with a single model level and choosing version with the "version" parameter.

Xyce supports BSIM-CMG 107, 108, 110, and 111, but as the level 107, 108, 110, and 111 MOSFETS.   Change the level parameter in all your model cards from 72 to 107 and you should be able to get past this error.

gowreddysuresh, 2025-04-09:
Thank you so much for your quick and helpful response! I really appreciate your support