Xyce/Xyce #147: `utils/XyceDatestamp.sh` is broken when no suitable tags exist

robtaylor, 2025-06-22:
Since the history rewrite of the public branch, there are no reachable tags, so 
```
git describe --dirty
```
fails. Fix is to change to 
```
git describe --dirty --all
```

tvrusso, 2025-06-23:
While your patch will address the problem temporarily, it will be a non-issue when the team releases Xyce 7.10 (which as I understand it will be very soon).  Tagging new releases will fix it going forward without the need for patching the datestamp technique.

As a temporary workaround, one is better off just creating a local tag in ones own clone.

`git tag ReRelease-7.9.0 35198fa8c`

should do the trick, as it will create a tag that will make git describe work.  Alternatively, one can reconnect the old and new histories using the technique in README.md:

```
git replace 35198fa8c origin/old_master
```

tvrusso, 2025-06-23:
And I would highly recommend that the Xyce team actually create and push that tag so that it all works again.

`git tag -a -m 'Retag Xyce Release 7.9.0' ReRelease-7.9.0 35198fa8c`

I had not noticed that this broke git describe before.  I wonder if it used to silently succeed and a recent git update made it break?  I distinctly recall after the history reset having a blank in the `Xyce -v` output, which is why I personally created my "ReRelease-7.9.0" tag and shrugged it off.